VoodooSMBus/VoodooRMI #6: F12 implementation

1Revenger1, 2020-06-11:
This function is likely used for touchscreens, as it mentions Pens.
I haven't seen any sensor actually use this, but it likely is something we should likely implement at some point anyways just in case it isn't just for touchscreens.

Feel free to comment below if you get logging related to a missing F12 function.

dkoluris, 2020-06-26:
2020-06-27 00:11:27.063912+0300 0x129      Default     0x0                  0      0    kernel: (VoodooRMI) SMBus version 3
2020-06-27 00:11:27.063978+0300 0x129      Default     0x0                  0      0    kernel: (VoodooSMBus) Touchpad probe
2020-06-27 00:11:27.064372+0300 0x5b6      Default     0x0                  0      0    kernel: (VoodooRMI) RMI Bus (RELEASE) Starting up!
2020-06-27 00:11:27.101551+0300 0x5b6      Default     0x0                  0      0    kernel: (VoodooRMI) Initializing F34.
2020-06-27 00:11:27.103861+0300 0x5b6      Default     0x0                  0      0    kernel: (VoodooRMI) Initializing F01.
2020-06-27 00:11:27.122752+0300 0x5b6      Default     0x0                  0      0    kernel: (VoodooRMI) Found RMI device, manufacturer: Synaptics, product: TM3408-004, fw id: 2751231
2020-06-27 00:11:27.132212+0300 0x5b6      Default     0x0                  0      0    kernel: (VoodooRMI) Initializing F12.
2020-06-27 00:11:27.132226+0300 0x5b6      Default     0x0                  0      0    kernel: (VoodooRMI) Error: Unknown function: 12 - Continuing to load
2020-06-27 00:11:27.136818+0300 0x5b6      Default     0x0                  0      0    kernel: (VoodooRMI) Initializing F54.
2020-06-27 00:11:27.136831+0300 0x5b6      Default     0x0                  0      0    kernel: (VoodooRMI) F54 not implemented - Debug function
2020-06-27 00:11:27.141421+0300 0x5b6      Default     0x0                  0      0    kernel: (VoodooRMI) Initializing F30.
2020-06-27 00:11:27.145667+0300 0x5b6      Default     0x0                  0      0    kernel: (VoodooRMI) Found Button 1
2020-06-27 00:11:27.150267+0300 0x5b6      Default     0x0                  0      0    kernel: (VoodooRMI) Initializing F55.
2020-06-27 00:11:27.150281+0300 0x5b6      Default     0x0                  0      0    kernel: (VoodooRMI) Error: Unknown function: 55 - Continuing to load
2020-06-27 00:11:28.055722+0300 0x128      Default     0x0                  0      0    kernel: (VoodooSMBus) Debug: No response
2020-06-27 00:11:28.055747+0300 0x128      Default     0x0                  0      0    kernel: (VoodooSMBus) hello packet length fail: -6
2020-06-27 00:11:28.156019+0300 0x128      Default     0x0                  0      0    kernel: (VoodooSMBus) Debug: No response
2020-06-27 00:11:28.156040+0300 0x128      Default     0x0                  0      0    kernel: (VoodooSMBus) hello packet length fail: -6
2020-06-27 00:11:28.257398+0300 0x128      Default     0x0                  0      0    kernel: (VoodooSMBus) Debug: No response
2020-06-27 00:11:28.257421+0300 0x128      Default     0x0                  0      0    kernel: (VoodooSMBus) hello packet length fail: -6
2020-06-27 00:11:28.358765+0300 0x128      Default     0x0                  0      0    kernel: (VoodooSMBus) Debug: No response
2020-06-27 00:11:28.358788+0300 0x128      Default     0x0                  0      0    kernel: (VoodooSMBus) hello packet length fail: -6
2020-06-27 00:11:28.459492+0300 0x128      Default     0x0                  0      0    kernel: (VoodooSMBus) Debug: No response

dkoluris, 2020-06-26:
I have a Sunpoint LP just like the one included in the .plist

Devices are TPD0 (trackpad), TPL1 (touchscreen). TPL1 successfully works with VoodooI2C, but not TPD0.

1Revenger1, 2020-07-04:
@dkoluris latest commit should have working F12 support. Feel free to build and test

1Revenger1, 2020-07-07:
Latest commit has good support now for F12
Release should be out today

closed 2020-07-07 by 1Revenger1