VoodooSMBus/VoodooRMI #9: Trackpoint button does not work

gimdh, 2020-06-20:
**Describe the bug**
Trackpoint button does nothing

**To Reproduce**
Install driver. Corresponding patches are applied.

**Expected behavior**
Three trackpoint button works accordingly.

**Log**
N/A

Please note that my model has trackpoint buttons integrated to trackpad assembly, not keyboard. Thus it could send click signal via SMBus.

1Revenger1, 2020-06-20:
Looking at your other issue, sounds like it's VoodooI2c doing something. Make sure you don't have VoodooI2c loading.

gimdh, 2020-06-20:
@1Revenger1 I never had VoodooI2C lodaed on this machine as VoodooPS2 was the only option to use trackpad until this great driver. And again, I checked thoroughly.

I also acked entire repo, and found no matching lines with that error message, but still removing this kext gets rid of error message.

gimdh, 2020-06-20:
Duplicate of #8 
It seems both problem shares same cause. Closing this issue to focus on one.

1Revenger1, 2020-06-20:
This needs code implemented in F30 to commit buttons to F03.

gimdh, 2020-06-23:
Updated code works great.

closed 2020-06-23 by gimdh