VoodooSMBus/VoodooRMI #10: Success on T480 with Synaptics touchpad. Possible to disable/remove Elan-driver?

benbender, 2020-06-21:
**Describe the bug**
I just wanted to give the project a shot and quite pleased. Nice Work!
It works nicely on a Thinkpad T480 with synaptics touchpad. Both Pad and Stick!
The scrolling on the touchpad feels even much more like a native apple one. Perfect!

One little oddity stuck to my eye when I was checking the logs:

````
2020-06-21 16:58:15.978884+0200 0xf5       Default     0x0                  0      0    kernel: (VoodooSMBus) Debug: No response
[... around 10 repeats of the same message...]
2020-06-21 16:58:16.986326+0200 0xf5       Default     0x0                  0      0    kernel: (VoodooSMBus) Error: Could not initialize ELAN device.
2020-06-21 16:58:59.459198+0200 0x1ce0     Default     0x0                  0      0    kernel: (VoodooSMBus) Debug: Lost arbitration
````

Also there seem to be little delay in the boot process - maybe because of the above error?

Is there any possibility to remove/disable the driver for the Elan-touchpad?

PS: As remark for others who want to try - For the T480 one needs to patch ACPI with this little snippet to make VoodooSMBUS recognize the device: https://github.com/daliansky/OC-little/blob/master/05-%E6%B3%A8%E5%85%A5%E8%AE%BE%E5%A4%87/05-3-SBUS_SMBU%E8%A1%A5%E4%B8%81/SSDT-SBUS.dsl

1Revenger1, 2020-06-21:
You could try removing address 0x15 from the Info.plist under VoodooSMBUSController. That'll remove part of it at least. The elans may still try to attach to the nub for 0x2c as well

1Revenger1, 2020-06-21:
The author does want to seperate the elans stuff from VoodooSMBus into another kext, similar to RMI, but they've been busy lately.

1Revenger1, 2020-06-21:
Oh, I'm stupid. Could completely prevent elans from loading if you get rid of the entry for it in the Info.plist under IOKit Personalities. Sorry for not thinking of this earlier, oops

closed 2020-06-21 by 1Revenger1

benbender, 2020-06-21:
Thanks for your fast response, but I think I will leave it for now and won't fiddle around in the codebase I don't fully understand.
Separation would be great indeed, but for the moment, I'll test the code as it is.

Thank you again.