VoodooSMBus/VoodooRMI #12: Lenovo T490 trackpad not working

yusifsalam, 2020-06-24:
Device: Lenovo T490, OpenCore 0.5.9
Issue: Trackpad is not detecting any input, even though trackpoint is working.

What I did:
1. In Info.plist of VoodooSMBus.kext add my device id (0x9da38086)
2. Copy VoodooSMBus and VoodooRMI kexts to kext folder
3. Add the 4 patches for SMBus to config.plist
4. Disable VoodooPS2 mouse, trackpad, input in config.plist
5. Enable VoodooRMI and input plugin, VoodooSMBus

IOReg is showing 2 nubs at 0x2c and 0x15. Trackpad device is detected but doesn't react to input. Trackpoint and the buttons are working. 
```
❯ kextstat | grep -i voodoo
   65    1 0xffffff7f8354b000 0x14000    0x14000    as.acidanthera.voodoo.driver.PS2Controller (2.1.4) F6B9B0C1-59C1-3CA8-9560-79D727E35BB2 <12 8 6 5 3 1>
   76    1 0xffffff7f8555a000 0x31000    0x31000    de.leo-labs.VoodooSMBus (2.1) 672C9AAE-BC8B-3B92-9387-EB6CD2290386 <60 13 6 5 3>
   79    0 0xffffff7f8559b000 0x1d000    0x1d000    com.1Revenger1.VoodooRMI (0.4) A7DD59E2-EF24-3BCB-8A47-7A865BA245AD <76 60 6 5 3>
   98    0 0xffffff7f83564000 0xd000     0xd000     as.acidanthera.voodoo.driver.PS2Keyboard (2.1.4) AC45F7E0-7AAB-397E-8E50-864E40EB4F4A <65 60 8 6 5 3 1>
```

My EFI folder can be found from [here](https://github.com/yusifsalam/t490-macos/tree/testing).

Is this a bug or am I doing something wrong? Thank you

benbender, 2020-06-24:
VoodooInput seems to be missing. Try to move VoodooInput from VoodooRMI.kext before VoodooPS2Controller in config.plist.

1Revenger1, 2020-06-24:
Yes, VoodooInput does not appear to load. I would double check to make sure OpenCore is not erroring trying to add the kext, and make sure that the entry looks right in your config.plist.

benbender, 2020-06-24:
The order of loading should be the problem. Order matters in OC and voodooinput is loaded after voodoops2 according to the  linked repository.

1Revenger1, 2020-06-24:
VoodooPS2Controller/Keyboard has no dependency on VoodooInput.

In addition, VoodooInput is not a dependency for VoodooRMI needed for loading. VoodooInput attaches based off of a property we set, so VoodooInput can be loaded after everything.

1Revenger1, 2020-06-24:
Closing as this likely isn't an issue with VoodooRMI

1Revenger1, 2020-06-25:
@yusifsalam let's continue our discussion here.

Can you try loading the RMI once your in macOS using `kextload` ?
That should provide better logging data

1Revenger1, 2020-06-25:
What was the error when you tried to load manually? You can add `-vvvvvv` to get more verbose loading. If not, you can check the kernel log as well iirc.

danny7g, 2020-06-25:
"failed to load - (libkern/kext) not loadable (reason unspecified); check the system/kernel logs for errors or try kextutil(8)."

The -vvvv option didn't add any useful info (I assume you meant with kextload, not as a boot-arg). 

[dannys_macbook.lan.ioreg.zip](https://github.com/VoodooSMBus/VoodooRMI/files/4832812/dannys_macbook.lan.ioreg.zip)

1Revenger1, 2020-06-25:
kextutil might work better. If not, I'd have console.app open, and have it filtering for "Kernel"

yusifsalam, 2020-06-25:
@1Revenger1  Sorry about posting in the wrong repo, I thought it would be more polite to post in an open issue : D

```
❯ sudo chown -R 0:0 Documents/Dev/Hackintosh/t490-macos/OC/Kexts/VoodooRMI.kext/
  sudo chmod -R 755 Documents/Dev/Hackintosh/t490-macos/OC/Kexts/VoodooRMI.kext/

~
❯ sudo kextutil Documents/Dev/Hackintosh/t490-macos/OC/Kexts/VoodooRMI.kext/ -vvvv
Defaulting to kernel file '/System/Library/Kernels/kernel'
Kext with invalid signature (-67050) allowed: <OSKext 0x7fc2765082e0 [0x7fff89af1b60]> { URL = "file:///Users/yusif/Documents/Dev/Hackintosh/t490-macos/OC/Kexts/VoodooRMI.kext/", ID = "com.1Revenger1.VoodooRMI" }
/Users/yusif/Documents/Dev/Hackintosh/t490-macos/OC/Kexts/VoodooRMI.kext - dependency 'de.leo-labs.VoodooSMBus' not found.
/Users/yusif/Documents/Dev/Hackintosh/t490-macos/OC/Kexts/VoodooRMI.kext - dependency 'de.leo-labs.VoodooSMBus' not found.
/Users/yusif/Documents/Dev/Hackintosh/t490-macos/OC/Kexts/VoodooRMI.kext - dependency 'de.leo-labs.VoodooSMBus' not found.
Code Signing Failure: code signature is invalid
Dependency Resolution Failures:
    No kexts found for these libraries:
        de.leo-labs.VoodooSMBus

Diagnostics for /Users/yusif/Documents/Dev/Hackintosh/t490-macos/OC/Kexts/VoodooRMI.kext:
Dependency Resolution Failures:
    No kexts found for these libraries:
        de.leo-labs.VoodooSMBus
~
```

So according to this SMBus isn't resolved?

1Revenger1, 2020-06-25:
Heh, id rather not fill up the other repo mostly because that's mostly another Authors work. It's fine though.

I've found that usually I need to do `-d /path/to/VoodooSMBus.kext` to have it find that dependency for some reason.

yusifsalam, 2020-06-25:
```
❯ sudo kextutil Documents/Dev/Hackintosh/t490-macos/OC/Kexts/VoodooRMI.kext/ -vvvv -d Documents/Dev/Hackintosh/t490-macos/OC/Kexts/VoodooSMBus.kext/
Defaulting to kernel file '/System/Library/Kernels/kernel'
Kext with invalid signature (-67050) allowed: <OSKext 0x7f85c0f089d0 [0x7fff89af1b60]> { URL = "file:///Users/yusif/Documents/Dev/Hackintosh/t490-macos/OC/Kexts/VoodooRMI.kext/", ID = "com.1Revenger1.VoodooRMI" }
Kext with invalid signature (-67030) allowed: <OSKext 0x7f85c1924360 [0x7fff89af1b60]> { URL = "file:///Users/yusif/Documents/Dev/Hackintosh/t490-macos/OC/Kexts/VoodooSMBus.kext/", ID = "de.leo-labs.VoodooSMBus" }
Code Signing Failure: code signature is invalid
Warnings:
    Dependency lacks appropriate value for OSBundleRequired and may not be available during early boot:
        de.leo-labs.VoodooSMBus - OSBundleRequired not set

Warnings:
    Dependency lacks appropriate value for OSBundleRequired and may not be available during early boot:
        de.leo-labs.VoodooSMBus - OSBundleRequired not set

/Users/yusif/Documents/Dev/Hackintosh/t490-macos/OC/Kexts/VoodooRMI.kext appears to be loadable (not including linkage for on-disk libraries).
Loading /Users/yusif/Documents/Dev/Hackintosh/t490-macos/OC/Kexts/VoodooRMI.kext.
/Users/yusif/Documents/Dev/Hackintosh/t490-macos/OC/Kexts/VoodooRMI.kext successfully loaded (or already loaded).
Invalid signature -67050 for kext <OSKext 0x7f85c0f089d0 [0x7fff89af1b60]> { URL = "file:///Users/yusif/Documents/Dev/Hackintosh/t490-macos/OC/Kexts/VoodooRMI.kext/", ID = "com.1Revenger1.VoodooRMI" }

~
❯ kextstat|grep -i voodoo
   65    1 0xffffff7f835e1000 0x14000    0x14000    as.acidanthera.voodoo.driver.PS2Controller (2.1.5) 2529B917-EE1A-3F67-88B9-12F21B0F5D0A <12 8 6 5 3 1>
   72    1 0xffffff7f8354b000 0x3b000    0x3b000    de.leo-labs.VoodooSMBus (2.1) 3D978383-5782-3051-BF65-8037C9188562 <60 13 6 5 3>
   77    0 0xffffff7f835b0000 0x25000    0x25000    com.1Revenger1.VoodooRMI (0.5) 23224C87-5EA7-37E2-A28B-C2F9D1B78CD7 <72 60 6 5 3>
   84    0 0xffffff7f835fa000 0xd000     0xd000     as.acidanthera.voodoo.driver.PS2Keyboard (2.1.5) 2BDE73B8-0F0D-3915-8D25-9DE623F7A209 <65 60 8 6 5 3 1>

~
```

1Revenger1, 2020-06-25:
Can you get a log of it loading up? I think you were missing F11 as well iirc looking at the pic from the other issue.

yusifsalam, 2020-06-25:
Sorry I'm a bit ignorant when it comes to debugging, how can I produce that log? I tried to pipe the output to a file but that didn't work.

1Revenger1, 2020-06-25:
Uhh, usually I do `log show --last 1m | grep -i voodoormi > /path/to/log.txt` after loading the kext and waiting a few secs. Catalina has a weird issue where `stream` doesn't work unless you run the command first with `show`, so I just tend to skip using stream unless I'm looking at trackpad or trackstick data in real time.

yusifsalam, 2020-06-25:
```
020-06-25 22:09:06.576035+0300 0x40b1     Default     0x0                  793    0    sudo:    yusif : TTY=ttys000 ; PWD=/Users/yusif ; USER=root ; COMMAND=/usr/bin/kextutil Documents/Dev/Hackintosh/t490-macos/OC/Kexts/VoodooRMI.kext/ -d Documents/Dev/Hackintosh/t490-macos/OC/Kexts/VoodooSMBus.kext/ -vvvvvv
2020-06-25 22:09:08.700018+0300 0x4168     Default     0x0                  0      0    kernel: (VoodooRMI) SMBus version 3
2020-06-25 22:09:08.700749+0300 0x419f     Default     0x0                  0      0    kernel: (VoodooRMI) RMI Bus (DEBUG) Starting up!
2020-06-25 22:09:08.708269+0300 0x419f     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: rmi_probe_interrupts: Counting IRQs.
2020-06-25 22:09:08.743673+0300 0x419f     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: IRQ Count: 9
2020-06-25 22:09:08.743749+0300 0x419f     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: rmi_init_functions: Creating functions.
2020-06-25 22:09:08.746093+0300 0x419f     Default     0x0                  0      0    kernel: (VoodooRMI) Initializing F34.
2020-06-25 22:09:08.748467+0300 0x419f     Default     0x0                  0      0    kernel: (VoodooRMI) Initializing F01.
2020-06-25 22:09:08.768414+0300 0x419f     Default     0x0                  0      0    kernel: (VoodooRMI) Found RMI device, manufacturer: Synaptics, product: TM3471-010, fw id: 2867793
2020-06-25 22:09:08.779108+0300 0x419f     Default     0x0                  0      0    kernel: (VoodooRMI) Initializing F12.
2020-06-25 22:09:08.779141+0300 0x419f     Default     0x0                  0      0    kernel: (VoodooRMI) Error: Unknown function: 12 - Continuing to load
2020-06-25 22:09:08.784063+0300 0x419f     Default     0x0                  0      0    kernel: (VoodooRMI) Initializing F54.
2020-06-25 22:09:08.784094+0300 0x419f     Default     0x0                  0      0    kernel: (VoodooRMI) F54 not implemented - Debug function
2020-06-25 22:09:08.789022+0300 0x419f     Default     0x0                  0      0    kernel: (VoodooRMI) Initializing F3A.
2020-06-25 22:09:08.789055+0300 0x419f     Default     0x0                  0      0    kernel: (VoodooRMI) Error: Unknown function: 3A - Continuing to load
2020-06-25 22:09:08.791405+0300 0x419f     Default     0x0                  0      0    kernel: (VoodooRMI) Initializing F03.
2020-06-25 22:09:08.798521+0300 0x419f     Default     0x0                  0      0    kernel: (VoodooRMI) Initializing F55.
2020-06-25 22:09:08.798553+0300 0x419f     Default     0x0                  0      0    kernel: (VoodooRMI) Error: Unknown function: 55 - Continuing to load
2020-06-25 22:09:08.827624+0300 0x419f     Default     0x0                  0      0    kernel: (VoodooRMI) Start finished
2020-06-25 22:09:10.705917+0300 0x4169     Default     0x0                  0      0    kernel: (VoodooRMI) Error: Failed to read SMBus version. Code: 0xFFFFFFFA
```

1Revenger1, 2020-06-25:
Wack, I think I'll need to do some research. No clue what function 3A or 55 are.

Function 12 is likely why your trackpad doesn't work though, and is linked to issue #6

1Revenger1, 2020-06-25:
Oh, F55 is for getting number of tx/rx electrodes apparently, and is used for F54 debugging data.
So, only unknown one is 3A

yusifsalam, 2020-06-25:
That's a shame, guess I'll have to wait for the implementation of F12. I don't know C++ to port it myself but I'd be available if you need testers.

After a quick googling round it doesn't seem like linux implements the 3A function, so that could be a doozy.

1Revenger1, 2020-06-25:
My guess is that F3A isn't important for functioning, though I am super curious as to what it is. None of the documentation I have mentions it

1Revenger1, 2020-06-26:
I looked into implementing F12, and it doesn't look to bad aside from needing to somehow replicate some of the bitops in linux. I doubt I'll get around to implementing it myself in the next couple days, though would support someone else implementing it if they wanted to. I don't have a trackpad which uses F12, so I have no way to test.

1Revenger1, 2020-07-04:
F12 should be implemented. Feel free to build and test it yourself

benbender, 2020-07-04:
Very nice. Thank you will build, test and report back today! :)

benbender, 2020-07-04:
I've build latest VoodooSMBUS from your branch, linked it inside VodooRMBus (otherwise it could not find VoodooSMBusDeviceNub.hpp) and needed to fix a small error. Sadly it didn't load:

````
Defaulting to kernel file '/System/Library/Kernels/kernel'
Kext with invalid signature (-67050) allowed: <OSKext 0x7ffb7cf048b0 [0x7fff90a49b60]> { URL = "file:///Users/bb/Desktop/VoodooRMI/VoodooRMI.kext/", ID = "com.1Revenger1.VoodooRMI" }
Kext with invalid signature (-67030) allowed: <OSKext 0x7ffb7cf091d0 [0x7fff90a49b60]> { URL = "file:///Users/bb/Desktop/VoodooRMI/VoodooSMBus.kext/", ID = "de.leo-labs.VoodooSMBus" }
Code Signing Failure: code signature is invalid
/Users/bb/Desktop/VoodooRMI/VoodooRMI.kext appears to be loadable (not including linkage for on-disk libraries).
Loading /Users/bb/Desktop/VoodooRMI/VoodooRMI.kext.
Disabling KextAudit: SIP is off
KextAudit initialized: audit=F
(kernel) kxld[com.1Revenger1.VoodooRMI]: The following symbols were defined more than once:
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZN6RMIBus10gMetaClassE: 0xffffff7f9b22fb98 - 0xffffff7f9b2058f8
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZN6RMIBus10superClassE: 0xffffff7f9b22d310 - 0xffffff7f9b201f08
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZN6RMIBus16handleHostNotifyEv: 0xffffff7f9b225ce2 - 0xffffff7f9b1f7aa0
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZN6RMIBus4initEP12OSDictionary: 0xffffff7f9b225ade - 0xffffff7f9b1f7948
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZN6RMIBus4stopEP9IOService: 0xffffff7f9b226042 - 0xffffff7f9b1f7ab4
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZN6RMIBus5probeEP9IOServicePi: 0xffffff7f9b225b94 - 0xffffff7f9b1f795a
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZN6RMIBus5startEP9IOService: 0xffffff7f9b225c44 - 0xffffff7f9b1f79e4
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZN6RMIBus9MetaClassC1Ev: 0xffffff7f9b22579a - 0xffffff7f9b1f77b6
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZN6RMIBus9MetaClassC2Ev: 0xffffff7f9b225866 - 0xffffff7f9b1f7876
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZN6RMIBus9metaClassE: 0xffffff7f9b22d308 - 0xffffff7f9b201f00
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZN6RMIBusC1EPK11OSMetaClass: 0xffffff7f9b2257fc - 0xffffff7f9b1f7812
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZN6RMIBusC1Ev: 0xffffff7f9b2258e0 - 0xffffff7f9b1f78e8
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZN6RMIBusC2EPK11OSMetaClass: 0xffffff7f9b2257d6 - 0xffffff7f9b1f77f2
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZN6RMIBusC2Ev: 0xffffff7f9b225916 - 0xffffff7f9b1f7918
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZN6RMIBusD0Ev: 0xffffff7f9b225836 - 0xffffff7f9b1f7846
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZN6RMIBusD1Ev: 0xffffff7f9b22582c - 0xffffff7f9b1f783c
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZN6RMIBusD2Ev: 0xffffff7f9b225822 - 0xffffff7f9b1f7832
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZNK6RMIBus12getMetaClassEv: 0xffffff7f9b225858 - 0xffffff7f9b1f7868
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZNK6RMIBus9MetaClass5allocEv: 0xffffff7f9b225898 - 0xffffff7f9b1f78a8
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZTV6RMIBus: 0xffffff7f9b22d318 - 0xffffff7f9b201f10
(kernel) kxld[com.1Revenger1.VoodooRMI]: 	__ZTVN6RMIBus9MetaClassE: 0xffffff7f9b22db80 - 0xffffff7f9b202780
(kernel) Can't load kext com.1Revenger1.VoodooRMI - link failed.
(kernel) Failed to load executable for kext com.1Revenger1.VoodooRMI.
(kernel) Kext com.1Revenger1.VoodooRMI failed to load (0xdc008016).
(kernel) Failed to load kext com.1Revenger1.VoodooRMI (error 0xdc008016).
Failed to load /Users/bb/Desktop/VoodooRMI/VoodooRMI.kext - (libkern/kext) link error.
Failed to load /Users/bb/Desktop/VoodooRMI/VoodooRMI.kext - (libkern/kext) link error.
Check library declarations for your kext with kextlibs(8).
````

Did I miss something or is there an error I didn't catch?

Diff from my checkout:
````
diff --git a/VoodooRMI.xcodeproj/project.pbxproj b/VoodooRMI.xcodeproj/project.pbxproj
index 82a614a..f1395f9 100644
--- a/VoodooRMI.xcodeproj/project.pbxproj
+++ b/VoodooRMI.xcodeproj/project.pbxproj
@@ -39,6 +39,29 @@
 /* End PBXBuildFile section */

 /* Begin PBXFileReference section */
+		21EC34B124B069D8006B0ECB /* helpers.cpp */ = {isa = PBXFileReference; lastKnownFileType = sourcecode.cpp.cpp; path = helpers.cpp; sourceTree = "<group>"; };
+		21EC34B224B069D8006B0ECB /* VoodooSMBusDeviceNub.hpp */ = {isa = PBXFileReference; lastKnownFileType = sourcecode.cpp.h; path = VoodooSMBusDeviceNub.hpp; sourceTree = "<group>"; };
+		21EC34B324B069D8006B0ECB /* VoodooSMBusIntelLpssI2C.hpp */ = {isa = PBXFileReference; lastKnownFileType = sourcecode.cpp.h; path = VoodooSMBusIntelLpssI2C.hpp; sourceTree = "<group>"; };
+		21EC34B424B069D8006B0ECB /* rmi_smbus.hpp */ = {isa = PBXFileReference; lastKnownFileType = sourcecode.cpp.h; path = rmi_smbus.hpp; sourceTree = "<group>"; };
+		21EC34B524B069D8006B0ECB /* VoodooSMBusSlaveDeviceDriver.hpp */ = {isa = PBXFileReference; lastKnownFileType = sourcecode.cpp.h; path = VoodooSMBusSlaveDeviceDriver.hpp; sourceTree = "<group>"; };
+		21EC34B624B069D8006B0ECB /* i2c_smbus.h */ = {isa = PBXFileReference; lastKnownFileType = sourcecode.c.h; path = i2c_smbus.h; sourceTree = "<group>"; };
+		21EC34B724B069D8006B0ECB /* TrackpointDevice.hpp */ = {isa = PBXFileReference; lastKnownFileType = sourcecode.cpp.h; path = TrackpointDevice.hpp; sourceTree = "<group>"; };
[... cut ...]
 			);
 			sourceTree = "<group>";
 		};
diff --git a/VoodooRMI/Transports/RMISMBus.cpp b/VoodooRMI/Transports/RMISMBus.cpp
index 607e8dd..070ade5 100644
--- a/VoodooRMI/Transports/RMISMBus.cpp
+++ b/VoodooRMI/Transports/RMISMBus.cpp
@@ -35,7 +35,7 @@ RMISMBus *RMISMBus::probe(IOService *provider, SInt32 *score)
         return NULL;
     }

-    device_nub->wakeupController();
+    /* device_nub->wakeupController(); */
     device_nub->setSlaveDeviceFlags(I2C_CLIENT_HOST_NOTIFY);

     do {
````

1Revenger1, 2020-07-04:
Hmm, that commit appears to be almost a month old. Or atleast, doesn't look right at all.

benbender, 2020-07-04:
I found my error: I used the branch of VoodooSMBUS in your repo because I overlooked that the fixes for opencore where merged into mainline. 

Duplicated symbols where resolved using mainline, but VoodooInput (loaded as plugin from inside the fresh VoodooRMI-build) doesn't seem to load without error. VoodooSMBUS attaches in IOReg, but VoodooRMI is not. 

If I unload VoodooInput, VoodooRMBUS & VoodooSMBUS and reload them, they are loading fine without error, but I don't have the Trackpad-Prefpane or any functionality from the Touchpad/Trackpoint.

Do you have any idea where to look/what to do to help?

benbender, 2020-07-04:
Update:
VoodooRMI-Master + VoodooSMBUS from the release-package are working fine:

````
❯ kextstat |grep -i voodoo
   69    1 0xffffff7f845df000 0x14000    0x14000    as.acidanthera.voodoo.driver.PS2Controller (2.1.5) 2529B917-EE1A-3F67-88B9-12F21B0F5D0A <12 8 6 5 3 1>
   78    1 0xffffff7f848d8000 0x31000    0x31000    de.leo-labs.VoodooSMBus (2.1) 672C9AAE-BC8B-3B92-9387-EB6CD2290386 <63 13 6 5 3>
   80    0 0xffffff7f84919000 0x22000    0x22000    com.1Revenger1.VoodooRMI (0.5) 6E322A9B-83DC-3CF6-AF88-34BD377F7553 <78 63 6 5 3>
  102    0 0xffffff7f845f8000 0xd000     0xd000     as.acidanthera.voodoo.driver.PS2Keyboard (2.1.5) 2BDE73B8-0F0D-3915-8D25-9DE623F7A209 <69 63 8 6 5 3 1>
  114    0 0xffffff7f845c7000 0x18000    0x18000    me.kishorprins.VoodooInput (1.0.6) AC8ACC4A-1D2B-3CA1-8ECC-9B19093940CF <63 6 5 3>
````

yusifsalam, 2020-07-04:
Latest commit of RMI and SMBus from the release of RMI is still not loading VoodooInput on startup. So I unloaded the RMI kext and loaded it manually with kextutil and it's saying F12 isn't implemented. 
```
2020-07-04 13:20:41.701230+0300 0x3c99     Default     0x7e65               986    0    AdvancedBlocking: (libswiftFoundation.dylib) AG: The extension received a message (getAdvancedBlockingData) from a script injected into (https://github.com/VoodooSMBus/VoodooRMI/issues/12) with userInfo ([:])
2020-07-04 13:21:24.878879+0300 0x3e56     Default     0x0                  997    0    sudo:    yusif : TTY=ttys000 ; PWD=/Users/yusif ; USER=root ; COMMAND=/usr/bin/kextutil Documents/Dev/Hackintosh/t490-macos/OC/Kexts/VoodooRMI.kext/ -vvvv -d Documents/Dev/Hackintosh/t490-macos/OC/Kexts/VoodooSMBus.kext/
2020-07-04 13:21:26.227072+0300 0x3e7a     Default     0x0                  0      0    kernel: (VoodooRMI) SMBus version 3
2020-07-04 13:21:26.228133+0300 0x3e91     Default     0x0                  0      0    kernel: (VoodooRMI) RMI Bus (DEBUG) Starting up!
2020-07-04 13:21:26.235684+0300 0x3e91     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: rmi_probe_interrupts: Counting IRQs.
2020-07-04 13:21:26.271114+0300 0x3e91     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: IRQ Count: 9
2020-07-04 13:21:26.271176+0300 0x3e91     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: rmi_init_functions: Creating functions.
2020-07-04 13:21:26.273791+0300 0x3e91     Default     0x0                  0      0    kernel: (VoodooRMI) Initializing F34.
2020-07-04 13:21:26.273806+0300 0x3e91     Default     0x0                  0      0    kernel: (VoodooRMI) Error: Unknown function: 34 - Continuing to load
2020-07-04 13:21:26.276401+0300 0x3e91     Default     0x0                  0      0    kernel: (VoodooRMI) Initializing F01.
2020-07-04 13:21:26.297631+0300 0x3e91     Default     0x0                  0      0    kernel: (VoodooRMI) Found RMI device, manufacturer: Synaptics, product: TM3471-010, fw id: 2867793
2020-07-04 13:21:26.306838+0300 0x3e91     Default     0x0                  0      0    kernel: (VoodooRMI) Initializing F12.
2020-07-04 13:21:26.306852+0300 0x3e91     Default     0x0                  0      0    kernel: (VoodooRMI) F12 not implemented (Issue VoodooRMI/#6)
2020-07-04 13:21:26.312075+0300 0x3e91     Default     0x0                  0      0    kernel: (VoodooRMI) Initializing F54.
2020-07-04 13:21:26.312088+0300 0x3e91     Default     0x0                  0      0    kernel: (VoodooRMI) F54 not implemented
2020-07-04 13:21:26.317296+0300 0x3e91     Default     0x0                  0      0    kernel: (VoodooRMI) Initializing F3A.
2020-07-04 13:21:26.317310+0300 0x3e91     Default     0x0                  0      0    kernel: (VoodooRMI) F3A not implemented
2020-07-04 13:21:26.319903+0300 0x3e91     Default     0x0                  0      0    kernel: (VoodooRMI) Initializing F03.
2020-07-04 13:21:26.327251+0300 0x3e91     Default     0x0                  0      0    kernel: (VoodooRMI) Initializing F55.
2020-07-04 13:21:26.327265+0300 0x3e91     Default     0x0                  0      0    kernel: (VoodooRMI) F55 not implemented
2020-07-04 13:21:26.356475+0300 0x3e91     Default     0x0                  0      0    kernel: (VoodooRMI) Start finished
2020-07-04 13:21:28.735001+0300 0x3e7b     Default     0x0                  0      0    kernel: (VoodooRMI) Error: Failed to read SMBus version. Code: 0xFFFFFFFA
```

1Revenger1, 2020-07-04:
Oops forgot to remove this line and the next https://github.com/VoodooSMBus/VoodooRMI/blob/master/VoodooRMI/RMIBus.cpp#L243 {line 243 and line 244). Mind just removing those and trying again?

yusifsalam, 2020-07-04:
Getting a kernel panic.
[image]

1Revenger1, 2020-07-04:
latest commit should fix that, sorry

yusifsalam, 2020-07-04:
hmm it seems like rmibus doesn't attach properly now.
voodooinput isn't loading still and logging for voodoormi creates an empty file. 

[voodoormi.ioreg.zip](https://github.com/VoodooSMBus/VoodooRMI/files/4873808/voodoormi.ioreg.zip)

1Revenger1, 2020-07-05:
I made a dumb mistake and forgot to read some registers from the device. See if the latest commit fixed it. You may need to load it manually again to get the logs, idk why macOS eats the logs when the driver is injected.

yusifsalam, 2020-07-05:
Everything loads up fine now! The touchpad still doesn't move the cursor but at least I can see that my input is detected ```2020-07-05 10:53:02.194162+0300 0x2b25     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: F12 Packet ```

benbender, 2020-07-05:
Seems to be working fine on T480 also. Multiple reboots without hickup on init. Nice one! Thanks!

yusifsalam, 2020-07-05:
Ok now I'm experiencing weird behavior: 
- trackpoint works as expected but the buttons don't react to single clicks, even though click and drag somehow works
- touchpad detects input but does not react to it. is it supposed to be sending events to the OS?
- RMI bus does not attach 100% of the time on system startup. Sometimes it fails and neither trackpoint nor touchpad work

1Revenger1, 2020-07-05:
I just pushed another commit which should hopefully fix the issues with F12

Also, I'm not sure about the buttons tbh. Are the buttons sending PS2 packets if you look in the log?  They'd have mentions of F03

yusifsalam, 2020-07-05:
The touchpad is alive with the latest commit, thanks!

Some issues persist however:
- short strokes are fine but for longer strokes the touchpad just stops reacting to input, for example if I draw circles on the touchpad with one finger it stops moving the cursor almost right away or at best I can draw one full circle before it stops reacting
- trackpoint single click button is ignored, this is the log message
```
kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: 8
kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: 0
kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: 0
```
I am fairly certain it was working fine when I opened this issue and my setup is almost identical, so perhaps some bug introduced in the last 11ish days?
BetterTouchTool's live preview can detect all the gestures fairly accurately, they're just not translated to system events for some reason. Here's a screencast of BTT's live view (at the start and end I moved the cursor with my mouse, that's why the input wasn't detected but the cursor still moved). As you can see I continue to move my finger from left to right on the touchpad but the cursor stops moving. 
[image]

Btw, slightly off-topic but would you consider opening a gitter chatroom or similar for this sort of thing where we go back and forth a lot, which imo does not necessarily work that well within the constraints of GitHub issues.

1Revenger1, 2020-07-05:
The voodooSMBus repo has a gitter so I think that would be preferable.

As for the cursor stopping - yeah I'm not sure why it happens. I think I messed something up abstracting the logic for the multi touch. I have issues with it as well with my trackpad.

1Revenger1, 2020-07-07:
@yusifsalam what kext are you using for battery status?

I had switched over to ACPIBatteryManager for Big Sur, and had thought that I had changed something in my own code messing stuff up. I moved over to SMCBatteryManager and it appears that the random stopping doesn't happen anymore

yusifsalam, 2020-07-07:
I'm using the SMCBatteryManager with OpenCore on Catalina.

1Revenger1, 2020-07-07:
https://gitter.im/VoodooSMBus/dev?utm_source=share-link&utm_medium=link&utm_campaign=share-link This should work.

1Revenger1, 2020-07-07:
Btw, trackpoint buttons should be fixed in last commit. I attempted to fix the issue with the trackpad as well, though I really haven't been able to reproduce any issues since I moved back to SMCBatteryManager

closed 2020-07-07 by 1Revenger1

gimdh, 2020-07-14:
According to what I've read, ACPIBatteryManager polls battery status. However, Laptop manufacturers cut corners and put obnoxious spin wait code to the function ACPIBatteryManager uses in order to make sure battery controller is initialised. Either of deleting sleep() or avoiding that code by SMCBatteryManager solves problem.

So I assume yusifsalam's issue is another one. I have similar problem where my pointer and physical finger traces do not quite match, but did not made issue report as I'm not sure if driver is the problem or Lenovo cheaped out on touchpad and this is what I get.

Do you think your touchpad is comparable with MacBooks' or MT2? I have another laptop with excellent I2C precision touchpad and it does feel legit with VoodooI2C. Fingers are basically directly mapped to cursor movement.
However my Thinkpad precision touchpad (Windows spec shows I2C and USB are only available as device protocol, yet laptop sees a precision touchpad. So.. I guess SMBus counts as I2C?) feels somewhat off with VoodooRMI.