VoodooSMBus/VoodooRMI #13: [Feature Request] Better Control of the Trackstick

SukkaW, 2020-06-24:
The default value for `TrackstickMultiplier` and `TrackstickScrollMultiplier` are far too high.

Although I have set `TrackstickScrollMultiplierX` & `TrackstickScrollMultiplierY` down to `1`, the scroll speed is still too fast.

Is it possible to get scroll speed even more slower?

1Revenger1, 2020-06-24:
Does changing the multipliers actually change the speed?

I don't mind allowing a bigger range of values, but you already can go down to 20% of the original value which seems very slow. I'm suspecting something isn't working right.

SukkaW, 2020-06-24:
@1Revenger1 Maybe something is broken. But most thing is working perfectly (even after waking from sleep).

I have also changed the scroll speed inside system preferences (Mouse) but no help. I assume the touchpad doesn't attach to MOU (unlike VoodooPS2).

1Revenger1, 2020-06-24:
https://github.com/VoodooSMBus/VoodooRMI/commit/0c861dc2ab5b5249248195ce23125084afb9d49b Should be better with this commit. Setting multiplier to 1 now makes my trackstick (basically) unmovable though lol.

closed 2020-06-24 by 1Revenger1

SukkaW, 2020-06-25:
@1Revenger1 LGTM! I will be able to adjust the speed more concisely then.