VoodooSMBus/VoodooRMI #16: No trackpoint

Rohan200220, 2020-07-05:
Hey, I followed all the instructions and my trackpad and gestures are working but there is no trackpoint.

1Revenger1, 2020-07-05:
I need logs, and a picture of IORegistryExplorer would be appreciated.

Will open once more information is provided

closed 2020-07-05 by 1Revenger1

Rohan200220, 2020-07-07:
[image]

No result is coming when trying to obtain logs.

Rohan200220, 2020-07-07:
Also I am using the latest version of VoodooRMI that was posted 2 hours ago.

1Revenger1, 2020-07-07:
Your IOReg suggests that there is no trackpoint. Can you boot linux and run `lsdev`? Send the output here

Rohan200220, 2020-07-07:
I don’t have linux installed, only windows

1Revenger1, 2020-07-07:
Without a log or more info - there is nothing I can do.
You may try loading it manually within windows using `kextload -d VoodooSMBus.kext VoodooRMI.kext`
May have better luck getting a log from using `dmesg` as well

Rohan200220, 2020-07-08:
Using dmesg gives me a lot of things, I just wrote sudo dmesg in terminal, Would that be helpful? Because I really want to use this trackpad kext as its gestures are working amazing

1Revenger1, 2020-07-08:
Sure, I guess you can give the entire dmesg log
`dmesg > ~/Desktop/Log.txt`

Rohan200220, 2020-07-08:
[log.txt](https://github.com/VoodooSMBus/VoodooRMI/files/4893259/log.txt)
Here it is.

1Revenger1, 2020-07-08:
Hmm, it seems very truncated and incomplete. There's no useful info in it other than the fact that the trackpad is working.

Rohan200220, 2020-07-08:
So what should i do now? Any other way of obtaining logs? Would the bootlog work that is obtained from clover configurator?

1Revenger1, 2020-07-08:
Can try loading it manually once within macOS using `sudo kextload`

Rohan200220, 2020-07-08:
Its giving me this error- failed to load - (libkern/kext) authentication failure (file ownership/permissions); check the system/kernel logs for errors or try kextutil(8).

1Revenger1, 2020-07-08:
When loading kexts:
SIP needs to at least be partially disabled to allow unsigned kexts to load
The kext needs to be owned by root (`sudo chown -R root:wheel VoodooRMI.kext`) - You may need to do the same thing with VoodooSMBus
Remember to load using `sudo kextload -d VoodooSMBus.kext VoodooRMI.kext`

Rohan200220, 2020-07-08:
So I disabled SIP from recovery, and ran this command

log show --last 1h | grep -I voodoormi

And it is giving me some failing logs.
[logs.txt](https://github.com/VoodooSMBus/VoodooRMI/files/4893376/logs.txt)

And now My trackpad is not working and its not showing in System Preferences either.
Also I think I edited my device id Correctly in info.plist
[image]

1Revenger1, 2020-07-08:
Are VoodooSMBus.kext/VoodooRMI.kext under /Users/rohan/ ? Its saying that it can't find the kexts.
`Unable to determine realpath for /Users/rohan/VoodooRMI.kext - failing`
You need to be in the same folder as the kexts, or specify the path for them

Rohan200220, 2020-07-08:
I also Tried this command-sudo kextload -d VoodooSMBus.kext VoodooRMI.kext
 but its giving me this

No kernel extensions specified; name kernel extension bundles
    following options, or use -bundle-id (-b).
usage: kextload [options] [--] [kext] ...

use kextload -help for an explanation of each option