VoodooSMBus/VoodooRMI #20: Touchpad dragging

marcolp98, 2020-07-08:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log**
Run `log show --last boot | grep -i voodoo > ~/Desktop/RMILog.txt` when using the DEBUG version of the driver. If you attach a log using the RELEASE version, we may ask you to use the DEBUG version and redo the log. This will provides logs for VoodooRMI from when the device was booted up.

1Revenger1, 2020-07-08:
How are you trying to drag?
macOS doesn't have double tap to drag enabled by default - so you have to enable it, or use a button.

marcolp98, 2020-07-08:
Woooops, i forgot to enable dragging in settings, my bad

btw thank you and who is working on this, it's just a pre-release and everything is working perfectly.

closed 2020-07-08 by marcolp98

1Revenger1, 2020-07-08:
Yep - main thing that's holding it from a 1.0 release are some couple outstanding panics that happen sometimes for a small subset of people, and the building system. Though, I think both are resolved now, or close to it.