VoodooSMBus/VoodooRMI #22: Make Trackpoint Scroll feels smooth

SukkaW, 2020-07-12:
Currently Trackpoint scroll (with middle button) feels like mouse wheel: things scroll like line by line.

Is it possible to make trackpoint scroll in a smooth way (like two fingers gesture on touchpad)?

1Revenger1, 2020-07-12:
I'm not entirely sure how to do that unfortunately. Would be nice to have though. An interim solution could be to use smoothScroll https://www.smoothscroll.net/mac/

SukkaW, 2020-07-13:
@1Revenger1 Well, VoodooPS2 supports smooth scroll. I am not sure whether it is a feature of PS2 protocol or not.

1Revenger1, 2020-07-13:
It isn't a feature of PS2. The trackstick data reported is the same between the two buses.

I took a quick looks and found these properties, so I can take a look in a bit
https://github.com/acidanthera/VoodooPS2/blob/master/VoodooPS2Trackpad/VoodooPS2SynapticsTouchPad.cpp#L528

gimdh, 2020-08-03:
https://github.com/VoodooSMBus/VoodooTrackpoint/pull/6
Last PR works fine with mine.

SukkaW, 2020-08-04:
> [VoodooSMBus/VoodooTrackpoint#6](https://github.com/VoodooSMBus/VoodooTrackpoint/pull/6)
> Last PR works fine with mine.

I will have a test of your branch to see if it works.

SukkaW, 2020-08-04:
> [VoodooSMBus/VoodooTrackpoint#6](https://github.com/VoodooSMBus/VoodooTrackpoint/pull/6)
> Last PR works fine with mine.

@gimdh I have a custom build and it works. And https://github.com/VoodooSMBus/VoodooRMI/issues/13 has been solved as well (The trackstick speed back to normal instead of too fast).

gimdh, 2020-08-04:
I suspect high mouse scroll sensitivity in macOS was causing #13. Scroll is too slow for me, too, so we might revert patch done for #13.

1Revenger1, 2020-08-04:
I'll need to try it personally at some point, currently messing with Big Sur.

Cool that it fixed quite a bit though. I'll check out the sensitivity when I take a look at it. I don't see much point in decreasing the denominator (currently 20) but we could increase the default value maybe.

1Revenger1, 2020-08-05:
Haha, ok I might lower the denominator. I've currently got the multiplier at 40 (40/20 = 2x multiplier) and it's still pretty slow. This otherwise is smoother.

closed 2020-08-26 by 1Revenger1

gimdh, 2020-09-03:
I have just found out Trackpoint scroll is not smooth in terminal. Maybe something more should be done.

1Revenger1, 2020-09-03:
Afaik it isn't smooth even with a trackpad. 
Terminal isn't like other programs - it has rows and columns it adheres to afaik. I can test tomorrow but iirc trackpad and trackpoint behaved basically the same.

SukkaW, 2020-09-03:
> Afaik it isn't smooth even with a trackpad.

Confirmed.

> I have just found out Trackpoint scroll is not smooth in terminal. Maybe something more should be done.

It should be considered as the desired behavior.

gimdh, 2020-09-03:
I compared magic trackpad 2 with trackpoint and thought MT2 to be much more smoother than trackpoint but yes, Macbook Air behaves same, and touchpad uses MT2 simulation via VoodooInput so they should be identical. My bad.

I think faster polling rate might made difference.

1Revenger1, 2020-09-03:
Hmm, could be polling rate. My trackpoint doesn't react to the polling rate command so it could be pretty far off. Wouldn't mind if someone compared different rates vs power consumption so we could choose the optimal one

gimdh, 2020-09-04:
@1Revenger1  It was way more simple problem. With Trackpoint scroll, multiple lines scroll at a time. However, with touchpad, only a single line scrolls at a time, but much more frequently. That's why I thought touchpad is smoother.
Skipping multiple lines at a time is generally bad for locating line, so I hope this to be fixed if possible. I wonder VoodooPS2 is any better.

PR #43 (2020-08-26): Smooth scroll for trackpoint and update Dependencies
Closes #35
Closes #22

Waiting on @gimdh 's PR to be accepted in VoodooTrackpoint.
commit af72b4ee (2020-08-26): Smooth scroll for trackpoint and update Dependencies (#43)

* Increase speeds, seperate trackpoint timeout to it's own variable
Closes #35
Closes #22

* Add info.plist

* Update dependencies

* Revert to 2:1 PS2 mode