VoodooSMBus/VoodooRMI #23: T450s Big Sur can't load the VoodooRMI.kext or VoodooSMBus.kext

findyy99, 2020-07-13:
-  when I didi what the readme said, I still can't make the trackpad of T450s work. When I use the kextstat, I found that the kext wasn't loaded. 

-  I have use the debug kext, the boot log has nothing about voodoo, so the txt file is bank.

-  this is the kext.

[image]

-  this is the config.plist.

[image]

- the txt generated has nothing.
[image]

- should I disable SIP or do something to make the kext loaded ?

1Revenger1, 2020-07-13:
What Bootloader?
If it's OpenCore, are there any logs from opencore?
Kernel Collection or Prelinked Kernel?
Where is the kext? Is it injected or under Library/Extensions or other place?

I've tested it with Big Sur using Prelinked Kernel and it 100% worked. I'm not planning on testing more though with Big Sur.

findyy99, 2020-07-13:
@1Revenger1 

- sorry, It's OpenCore. version 0.6.0

- I will upload the OpenCore log later

-  Prelinked Kernel

- The kext is in EFI/OC/kext

findyy99, 2020-07-13:
This is the OpenCore log!
[opencore-2020-07-13-031013.txt](https://github.com/VoodooSMBus/VoodooRMI/files/4910413/opencore-2020-07-13-031013.txt)

1Revenger1, 2020-07-13:
Does it work under Catalina?
I'd also check IORegistryExplorer and make sure something isn't attached to SBUS other than VoodooSMBus

findyy99, 2020-07-13:
@1Revenger1 I will try to boot my Catalina on my USB drive, and give a feedback later! Thank you!

findyy99, 2020-07-13:
@1Revenger1 Sorry, it seems my problem. Catalina still doesn't work. Can you please help me with my EFI? 
[OC.zip](https://github.com/VoodooSMBus/VoodooRMI/files/4910583/OC.zip)

1Revenger1, 2020-07-13:
Again, check SBUS under IORegistryExplorer

findyy99, 2020-07-13:
- unfortunately, I think there is nothing about SBUS.

[image]

1Revenger1, 2020-07-13:
It may be a different name, should start with S though. Basically are looking for the SMBus PCI device. Basically you want to make sure VoodooSMBus is attached, not AppleSMBusPCI

findyy99, 2020-07-13:
- I don't know why the AppleSMBusPCI.kext was loaded. So the  VoodooSMBus is not attached.
[image]

1Revenger1, 2020-07-13:
Hmm, so we haven't had anyone test this yet - my system doesn't need it. But can you try adding the two lines shown below under VoodooSMBus.kext/Contents/Info.plist? Should be done actually as below, right before the end of the plist

https://github.com/VoodooSMBus/VoodooSMBus/pull/41/files

findyy99, 2020-07-13:
@1Revenger1 Anyway, Thank you so much, I will have a try! 👍

1Revenger1, 2020-07-13:
Yep. If you don't mind, can you send a DSDT as well? The above link should fix it, but I'm curious. AppleSMBusPCI will load on apple devices and is another kext for the i801 SMBus chip found on Intel chipsets. On some systems, it'll attach I'm guessing because some properties exist in ACPI that I'd like to double check. The above link is a much cleaner solution then any ACPI solution though.

findyy99, 2020-07-13:
Sure! And I have tried add the above two lines to the plist. Doesn't work, this is the DSDT I am using. I really appreciate you taking the time to help me！
[DSDT.zip](https://github.com/VoodooSMBus/VoodooRMI/files/4910890/DSDT.zip)

1Revenger1, 2020-07-13:
Were the two lines added right before
```
</dict>
</plist>
```
At the very end of the file?

And thanks for the DSDT, I'll take a look tomorrow. You don't have an SSDT for SBUS/MCHC also right? That would cause this as well

findyy99, 2020-07-13:
- I add it to the end of` VoodooSMBus.kext/Contents/Info.plist`.Just like this

[image]

- No, I did have the `SSDT-MCHC.aml and SSDT-SMBU.aml`. This is the log of APCI.
[image]

findyy99, 2020-07-13:
I have found the SMBU in the IORegistryExplorer. But not VoodooSMBus, It's AppleSMBusPCI.

[image]

1Revenger1, 2020-07-13:
Oh, this laptop is from like 2015. Likely needed the edit above, but you now also need to add the device id to the Info.plist. there should be instructions for adding device id in the README under the dev branch (not master!) in the VoodooSMBus repo.

findyy99, 2020-07-13:
Sorry, I just realized maybe I should go to VoodooSMBus repository looking for some information. Thank you!😄😄👍👍👍

findyy99, 2020-07-13:
amazing！I add my own device-id and vendor-id everything works fine!  Thank you again! 👍👍

gimdh, 2020-07-13:
Can you check if VoodooInput kext is loaded properly? If you have Trackpoint, but no touchpad, then it is likely that VoodooInput not loading is the issue.

findyy99, 2020-07-13:
@gimdh Thanks! I figure out it. The Voodooinput has been loaded. Just because my  
Acpibatterymanager or SMCBattery is not loaded. By the way, it works good on ThinkPad T450S!

1Revenger1, 2020-07-13:
Cool! Do you mind adding the device Id in a PR maybe for VoodooSMBus? If you can't figure it out, I can put a PR in for you.

closed 2020-07-13 by 1Revenger1

findyy99, 2020-07-14:
Sorry , I am not familiar with how to use Github,  Maybe you can put a PR!