VoodooSMBus/VoodooRMI #24: Driver sometimes does not reload after sleep.

gimdh, 2020-07-13:
**Describe the bug**
After wake from deep sleep, sometimes touchpad and Trackpoint do not work.

**To Reproduce**
Put device in deep sleep, then wake.

**Expected behavior**
Touchpad and Trackpoint should work properly.

**Log**
I was using Release target kext. I'll switch to Debug one and try to reproduce bug.

1Revenger1, 2020-07-13:
Hmm, I was having issues with this earlier on, though I haven't been able to reproduce lately. Could try editing VoodooSMBus's Info.plist to remove address 0x15 (which is 21).

gimdh, 2020-07-14:
Reproducing bug is quite hard for me, too. It happened 2~3 times during 2 month period. For now, it seems to happen only when device goes into deep sleep(Power is cut completely, I guess?).
After I can reproduce bug consistently, I'll try suggested fix.

gimdh, 2020-07-15:
It seems latest code somhow fixes the issue. I was not able to reproduce bug even after numerous sleep/wake attempts. Temporarily closing issue until I can get detailed debug log.

closed 2020-07-15 by gimdh