VoodooSMBus/VoodooRMI #30: dependency 'de.leo-labs.VoodooSMBus' not found

the-eric-kwok, 2020-07-24:
**Describe the bug**
Unable to load the kext with OpenCore, so I tried to load it with kextload after OS booted.
Load VoodooRMI after VoodooSMBus and VoodooInput.

**Log**
```
2020-07-24 17:38:06.486412+0800  localhost sudo[963]:     eric : TTY=ttys004 ; PWD=/Volumes/Data/eric/Downloads/VoodooRMI-0.5-Debug ; USER=root ; COMMAND=/sbin/kextload VoodooSMBus.kext
2020-07-24 17:38:06.518160+0800  localhost kextd[116]: (Security) [com.apple.securityd:security_exception] MacOS error: -67030
2020-07-24 17:38:06.518236+0800  localhost kextd[116]: (Security) [com.apple.securityd:security_exception] MacOS error: -67030
2020-07-24 17:38:06.522370+0800  localhost kextd[116]: (Security) [com.apple.securityd:security_exception] MacOS error: -67030
2020-07-24 17:38:06.522673+0800  localhost kextd[116]: (IOKit) [com.apple.kext:kextlog] Kext with invalid signature (-67030) allowed: <OSKext 0x6000006b8fc0 [0x7fff8be0cb60]> { URL = "file:///Volumes/Data/eric/Downloads/VoodooRMI-0.5-Debug/VoodooSMBus.kext/", ID = "de.leo-labs.VoodooSMBus" }
2020-07-24 17:38:06.533999+0800  localhost kextd[116]: (Security) Created Activity ID: 0x9a95, Description: SecTrustEvaluateIfNecessary
2020-07-24 17:38:06.542340+0800  localhost kextd[116]: (Security) [com.apple.securityd:security_exception] MacOS error: -67030
2020-07-24 17:38:06.543287+0800  localhost kextd[116]: (Security) [com.apple.securityd:security_exception] MacOS error: -67030
2020-07-24 17:38:06.543349+0800  localhost kextd[116]: (Security) [com.apple.securityd:security_exception] MacOS error: -67030
2020-07-24 17:38:06.546063+0800  localhost kextd[116]: (Security) [com.apple.securityd:security_exception] MacOS error: -67030
2020-07-24 17:38:06.547032+0800  localhost kextd[116]: (Security) [com.apple.securityd:security_exception] MacOS error: -67030
2020-07-24 17:38:06.547094+0800  localhost kextd[116]: (Security) [com.apple.securityd:security_exception] MacOS error: -67030
2020-07-24 17:38:06.547169+0800  localhost kextd[116]: (IOKit) [com.apple.kext:kextlog] Invalid signature -67030 for kext <OSKext 0x6000006b8fc0 [0x7fff8be0cb60]> { URL = "file:///Volumes/Data/eric/Downloads/VoodooRMI-0.5-Debug/VoodooSMBus.kext/", ID = "de.leo-labs.VoodooSMBus" }
2020-07-24 17:38:06.547934+0800  localhost kextd[116]: (Security) [com.apple.securityd:security_exception] MacOS error: -67030
2020-07-24 17:38:06.548102+0800  localhost kextd[116]: (Security) [com.apple.securityd:security_exception] MacOS error: -67030
2020-07-24 17:38:08.345308+0800  localhost sudo[968]:     eric : TTY=ttys004 ; PWD=/Volumes/Data/eric/Downloads/VoodooRMI-0.5-Debug ; USER=root ; COMMAND=/sbin/kextload VoodooRMI.kext
2020-07-24 17:38:08.387962+0800  localhost kextd[116]: (IOKit) [com.apple.kext:kextlog] Kext with invalid signature (-67050) allowed: <OSKext 0x6000006afe40 [0x7fff8be0cb60]> { URL = "file:///Volumes/Data/eric/Downloads/VoodooRMI-0.5-Debug/VoodooRMI.kext/", ID = "com.1Revenger1.VoodooRMI" }
2020-07-24 17:38:08.389708+0800  localhost kextd[116]: (IOKit) [com.apple.kext:kextlog] /Volumes/Data/eric/Downloads/VoodooRMI-0.5-Debug/VoodooRMI.kext - dependency 'de.leo-labs.VoodooSMBus' not found.
2020-07-24 17:38:08.390202+0800  localhost kextd[116]: (IOKit) [com.apple.kext:kextlog] com.1Revenger1.VoodooRMI's dependencies failed security checks; failing.
```
```
$ sudo kextstat| grep -i voodoo                                 
   69    1 0xffffff7f84913000 0x14000    0x14000    as.acidanthera.voodoo.driver.PS2Controller (2.1.5) 2529B917-EE1A-3F67-88B9-12F21B0F5D0A <12 8 6 5 3 1>
  103    0 0xffffff7f8492c000 0xd000     0xd000     as.acidanthera.voodoo.driver.PS2Keyboard (2.1.5) 2BDE73B8-0F0D-3915-8D25-9DE623F7A209 <69 61 8 6 5 3 1>
  174    0 0xffffff7f8498b000 0x2d000    0x2d000    de.leo-labs.VoodooSMBus (2.1) 672C9AAE-BC8B-3B92-9387-EB6CD2290386 <61 13 6 5 3>
  175    0 0xffffff7f84901000 0x12000    0x12000    me.kishorprins.VoodooInput (1.0.6) 4CCC7C09-5BB8-355E-B104-2C81D8A29A91 <61 6 5 3>
```

1Revenger1, 2020-07-24:
When using kextload/kextutil, you usually need to add VoodooSMBus as an argument. `sudo kextload -d VokdooSMBus.kext VoodooRMI.kext`

the-eric-kwok, 2020-07-24:
Now it loads successfully. But the trackpad is still not working.
```
2020-07-24 21:13:32.336858+0800  localhost sudo[1188]:     eric : TTY=ttys004 ; PWD=/Volumes/Data/eric/Downloads/VoodooRMI-0.5-Debug ; USER=root ; COMMAND=/sbin/kextload -d VoodooSMBus.kext VoodooRMI.kext
2020-07-24 21:13:32.399727+0800  localhost kextd[116]: (IOKit) [com.apple.kext:kextlog] Kext with invalid signature (-67050) allowed: <OSKext 0x6000006ad1a0 [0x7fff8be0cb60]> { URL = "file:///Volumes/Data/eric/Downloads/VoodooRMI-0.5-Debug/VoodooRMI.kext/", ID = "com.1Revenger1.VoodooRMI" }
2020-07-24 21:13:32.425506+0800  localhost kextd[116]: (Security) Created Activity ID: 0x106e4, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.448449+0800  localhost kextd[116]: (Security) Created Activity ID: 0x106e5, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.457748+0800  localhost kextd[116]: (Security) Created Activity ID: 0x106e6, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.470755+0800  localhost kextd[116]: (Security) Created Activity ID: 0x106e7, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.484894+0800  localhost kextd[116]: (Security) Created Activity ID: 0x106e8, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.497526+0800  localhost kextd[116]: (Security) Created Activity ID: 0x106e9, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.520380+0800  localhost kextd[116]: (Security) Created Activity ID: 0x106ea, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.537124+0800  localhost kextd[116]: (Security) Created Activity ID: 0x106eb, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.537377+0800  localhost kextd[116]: (Security) Created Activity ID: 0x106ec, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.537474+0800  localhost kextd[116]: (Security) Created Activity ID: 0x106ed, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.537480+0800  localhost kextd[116]: (Security) Created Activity ID: 0x106ee, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.537532+0800  localhost kextd[116]: (Security) Created Activity ID: 0x106ef, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.537626+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d40, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.537744+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d41, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.537833+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d42, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.542415+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d43, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.557704+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d44, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.557744+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d45, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.575277+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d46, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.589351+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d47, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.607846+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d48, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.623777+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d49, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.624123+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d4a, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.624381+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d4b, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.624388+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d4c, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.624466+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d4d, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.624470+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d4e, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.624478+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d4f, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.624486+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d60, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.637081+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d61, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.637089+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d62, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.637428+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d63, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.637531+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d64, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.637558+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d65, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.637605+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d66, Description: SecTrustEvaluateIfNecessary
2020-07-24 21:13:32.642893+0800  localhost kextd[116]: (Security) Created Activity ID: 0x10d67, Description: SecTrustEvaluateIfNecessary
[... cut ...]
  103    0 0xffffff7f8492c000 0xd000     0xd000     as.acidanthera.voodoo.driver.PS2Keyboard (2.1.5) 2BDE73B8-0F0D-3915-8D25-9DE623F7A209 <69 61 8 6 5 3 1>
  174    1 0xffffff7f8498b000 0x2d000    0x2d000    de.leo-labs.VoodooSMBus (2.1) 672C9AAE-BC8B-3B92-9387-EB6CD2290386 <61 13 6 5 3>
  175    0 0xffffff7f84901000 0x12000    0x12000    me.kishorprins.VoodooInput (1.0.6) 4CCC7C09-5BB8-355E-B104-2C81D8A29A91 <61 6 5 3>
  176    0 0xffffff7f849b8000 0x28000    0x28000    com.1Revenger1.VoodooRMI (0.5) 7409967A-87FB-3A4B-8724-833A8F81F7F8 <174 61 6 5 3>
```

the-eric-kwok, 2020-07-24:
I'm using OpenCore, kexts are installed in EFI, there is nothing about 'voodoo' in my boot log.
```
$ log show --last boot --style syslog | grep -i voodoo
$                                                     
```
Here is my config:

Inject:
[image]

Block:
[image]

Here is SBUS in ioReg (Seems to be not loaded)
[image]

[image]

the-eric-kwok, 2020-07-24:
Works perfectly after add device-id and vendor-id into Info.plist!

The magic number is:
```
0xa1238086
```

For those guys whose kext don't load:
[VoodooSMBus does not load](https://github.com/VoodooSMBus/VoodooSMBus/blob/dev/README.md#voodoosmbus-does-not-load)

the-eric-kwok, 2020-07-24:
Close due to PR already exists [https://github.com/VoodooSMBus/VoodooSMBus/pull/35](https://github.com/VoodooSMBus/VoodooSMBus/pull/35)

closed 2020-07-24 by the-eric-kwok

1Revenger1, 2020-07-24:
Good to hear!