VoodooSMBus/VoodooRMI #33: Synaptic PS/2 don't work

LocTran016, 2020-07-28:
Excuse me. Can you help me with this, please. Although I’ve tried to used the VoodooRMI.kext, VoodooSMBus.kext (from here: https://github.com/VoodooSMBus/VoodooRMI) but my synaptic trackpad didn’t work (Even I’ve tried disable VoodooPS2Mouse,Trackpad and then VoodooInput inside VoodooPS2 as wel, it can’t movel). I had used the troubleshoot page, too. As you see, there aren’t any SMBus-related SSDT and I’ve modified the VoodooSMBus.kext, too. Here is my EFI as well if you want. Thanks a lot for reading my message.
P/S: I also tried to add these patch from here: https://github.com/leo-labs/macOS-ThinkPad-T480s/blob/master/EFI/CLOVER/config.example.plist#L500-L535 but my mouse still can’t move
[EFI.zip](https://github.com/VoodooSMBus/VoodooRMI/files/4989768/EFI.zip)

[image]

1Revenger1, 2020-07-28:
What bootloader? If it's OC, make sure that the order of kext loading is VoodooSMBus.kext, then VoodooRMI.kext, then finally RMISMBus.kext. Make sure that VoodooInput from RMI is being injected as well.
If that doesn't work, can you show what your SMBus looks like in IORegistryExplorer, and get the output from `kextstat` in terminal.

LocTran016, 2020-07-28:
@1Revenger1 Maybe the issues is because ProperTree put them in wrong order. One more thing before trying your advice, can RMISMBus.kext be after Whatevergreen like this
[image]

1Revenger1, 2020-07-28:
Yeah that's fine. The main important thing is that VoodooRMI and VoodooSMBus are before RMISMBus. Doesn't really matter what the order is between the two or if anything is before it/mixed in.

LocTran016, 2020-07-29:
Well, Although I've set everything like what you say, it can't work, ask for what my SMBus looks like in IORegistryExplorer, and get the output from kextstat in terminal, I'll try to get them to you as soon as possible because I don't have any usb mouse and I can't use my built-in mouse so sorry a lot for that. I'll try to send them to you soon.
P/S: Here is my SMBUS hardward ID if you want:
[image]
[EFI.zip](https://github.com/VoodooSMBus/VoodooRMI/files/4992517/EFI.zip)

LocTran016, 2020-07-29:
@1Revenger1 ,By the way, my bootloader is OpenCore and I don't know why but if I use the VoodooPS2.kext from Acidanthera my mouse won't work but if i use Rehabman's VoodooPS2 from here: https://github.com/RehabMan/OS-X-Voodoo-PS2-Controller, my trackpad can work with 1 finger and some 2 fingers features (not all of them, scrolling is OK but I can zoom in/out). And here is my full specs by AIDA64 Engineer (a *.txt and *htm in the zip)
[Report.zip](https://github.com/VoodooSMBus/VoodooRMI/files/4992654/Report.zip)
[Report.txt](https://github.com/VoodooSMBus/VoodooRMI/files/4992638/Report.txt)

LocTran016, 2020-07-29:
For the others things I will try my best for them

1Revenger1, 2020-07-29:
I haven't gotten this to run succesfully on any Ivy Bridge laptops - I'm not entirely sure why. It may be a different version of RMI? Not entirely sure on that, especially since RMI4 goes back to around ~2012. You may have better luck loading it in manually once within the OS just using the keyboard.

```
sudo chown -R root:wheel *.kext
sudo kextutil -vvvv -d VoodooSMBus.kext -d VoodooRMI.kext VoodooRMI.kext/Contents/PlugIns/VoodooSMBus.kext
```

Then getting a log by doing

```
log show --last boot | grep -i voodoo > ~/Desktop/log.txt
```

LocTran016, 2020-07-29:
Do I need to change -d VoodooSMBus.kext -d VoodooRMI.kext VoodooRMI.kext/Contents/PlugIns/VoodooSMBus.kext to my kext direction (by dragging the kext inside the terminal) or just copy and paste your code into the terminal, then. Thanks

1Revenger1, 2020-07-29:
The paths need to be accurate, if you are in the same directory as them then the above command should work. But if you aren't, then yes you need to make sure the path is correct to them all.

LocTran016, 2020-07-29:
There is a problem that you don't have VoodooRMI.kext/Contents/PlugIns/VoodooSMBus.kext so I think that I should change it into this:
sudo chown -R root:wheel *.kext
sudo kextutil -vvvv -d VoodooSMBus.kext -d VoodooRMI.kext VoodooRMI.kext/Contents/PlugIns/RMISMBus.kext

LocTran016, 2020-07-29:
@1Revenger1 Well, I've tried both but couldn't work. Anyway, all the necessary file is here, sorry if is there anything is miss, just tell me and I'll give it to you.
[Fixing trackpad.zip](https://github.com/VoodooSMBus/VoodooRMI/files/4995789/Fixing.trackpad.zip)

gimdh, 2020-07-29:
https://www.rockbox.org/wiki/pub/Main/SansaFuzePlus/Synaptics_RMI3_Specification.pdf

I guess older touchpads with RMI3 use F10.

gimdh, 2020-07-29:
@LocTran016 You have to cd into the downloaded and unzipped VoodooRMI folder. Also, make sure to use Debug kext for detailed log information.

1Revenger1, 2020-07-29:
It may be that it is RMI3, I haven't seen any way to detect which RMI version it is. It does look programmed very similarly to RMI4 though.

LocTran016, 2020-07-29:
@gimdh Well, I've run like this: 
cd ~/Downloads/VoodooRMI-1.0-Debug
sudo chown -R root:wheel *.kext
sudo kextutil -vvvv -d VoodooSMBus.kext -d VoodooRMI.kext VoodooRMI.kext/Contents/PlugIns/RMISMBus.kext
And this is what it shown in the terminal (the log is empty)
[Newest.zip](https://github.com/VoodooSMBus/VoodooRMI/files/4996143/Newest.zip)

LocTran016, 2020-07-29:
I'm using catalina, do you think I need to mount the System in Read-Write

LocTran016, 2020-07-29:
SIP is disabled

gimdh, 2020-07-29:
Log should be located under Downloaded kext folder.
However, it seems you need to remove kext from OC bootloader first. (Just delete two Kext files.)
You can load kext after boot, and in this way you can test kext without reboot.

BTW, you may want to edit post for minor extra informations rather than posting new comment, as issue board is somewhat different from chat.

1Revenger1, 2020-07-29:
So, what I've noticed is that testing with another person is that on Ivy Bridge, when we ask for the SMBus version, it gets stuck and doesn't get any sort of response, so no attempts are logged to try and get the version. It may be that some sort of timeout may be needed.

Rohan200220, 2020-08-01:
I am also on an ivy bridge laptop, Inspiron 14z 5423 and my trackpoint is also not working properly. I edited the VoodooSMBus info.plist file and entered my vender id and device id. All the gestures are working but the trackpoint is very laggy. Btw I am using Clover.

LocTran016, 2020-08-01:
@Rohan200220 , I've tried that but didn't work, though

Rohan200220, 2020-08-01:
@LocTran016 Sometimes it works for me, sometimes it doesn't. When I restart my laptop the trackpad doesn't even respond.

1Revenger1, 2020-08-27:
Does someone mind testing out these new versions of VoodooRMI and VoodooSMBus?
https://files.gitter.im/5e938b02d73408ce4fe02f4f/H2AL/VoodooRMI.kext.zip
https://files.gitter.im/5e938b02d73408ce4fe02f4f/66yp/VoodooSMBus.kext.zip

If you are injecting these kexts rather than loading it manually within macOS, better way to get logs it to add `msgbuf=1048576` to your boot args, and use `sudo dmesg` to get logs, as `log show --last boot` is pretty atrocious for early boot logs.

Rohan200220, 2020-08-28:
My trackpad is still not responding with the above kexts, and I am using clover

1Revenger1, 2020-08-28:
I can't do anything without logs. Please reread my last message.

Rohan200220, 2020-08-28:
[Log.txt](https://github.com/VoodooSMBus/VoodooRMI/files/5139332/Log.txt)

Sorry, here it is

1Revenger1, 2020-08-28:
I'd double check which VoodooSMBus kext your using. It looks like it's trying to probe for an Elans trackpad, when all of the Elans stuff should've been taken out in the above VoodooSMBus download.

Rohan200220, 2020-08-28:
I used voodoosmbus from the above link, and how to take out elen stuff from voodoosmbus?

1Revenger1, 2020-08-28:
There isn't any Elans stuff in the above link. If you check the Info.plist for the kext you have injected, it should be version `3.0`, and not have any Elans stuff within it. Additionally, should have a place to put different addresses.
This is currently what I see
[image]

Rohan200220, 2020-08-28:
Um, sorry for being a noob :p, but what do i have to do?

1Revenger1, 2020-08-28:
Double check that you have the right Kext. Make sure that you don't have VoodooSMBus in kextcache, Library/Extensions, or System/Library/Extensions. If that all looks alright, double check that the Info.plist says it's version 3.0. You can find it by right clicking the kext in macOS and pressing "Show Package Contents". If your on windows, the kext already appears a folder so just look in there.

Rohan200220, 2020-08-28:
I don't have VoodooSMBus in S/L/E or in L/E, where is kextcache location?
[image]

It says version 3.0

1Revenger1, 2020-08-28:
Double check for duplicates I guess - the kext cache isn't stored in a place you can mess with. Easiest way I guess to be sure it's gone is to invalidate it using `sudo kextcache -i /`

Rohan200220, 2020-08-28:
> sudo kextcache -i /. It'll complain about kexts that shouldn't be in L/E or S/L/E as well.

just got this-
KernelCache ID: C13F4697AEBE1E8A99946DD865E9237F

Rohan200220, 2020-08-28:
I never installed any kexts in L/E or in S/L/E so I know no wrong kexts would be there. I just placed them in my EFI folder

1Revenger1, 2020-08-28:
Hmm, I guess you could try getting a log again

LocTran016, 2020-08-28:
With the newest version of VoodooRMI and VoodooSMBus, the freezing is fixed as you said before but through System Preference, it doesn't detect my trackpad as a trackpad (I can't even move my mouse so I have to use a USB mouse). I still using the same laptop. I will update the EFI as soon as possible
Here is my log:
[Terminal Saved Output.txt](https://github.com/VoodooSMBus/VoodooRMI/files/5141537/Terminal.Saved.Output.txt)

1Revenger1, 2020-08-28:
Your log doesn't show VoodooSMBus loading at all. Are you sure something else isn't attaching to SBUS like Apple's own SMBus kext?

1Revenger1, 2020-08-28:
What does IORegistryExplorer show? I didn't see any traces of VoodooSMBus in the logs.

1Revenger1, 2020-08-29:
Can you please post new responses rather than editing the one messages? I don't get notifications.

It shows VoodooSMBusIntelLpssI2C as attaching to `SBUS`, make sure you add the device ID to VoodooSMBus, not VoodooSMBusIntelLpssI2C.

LocTran016, 2020-08-31:
As you have already said before, the freezing is fixed but I can't use my mouse (I don't have the ability to move the built-in mouse). Here is everything my config.plist, the kext and terminal log as well as the ioreg file. Through the terminal (`sudo dmesg`), there are some RMI configuration and the kexts are loaded as well

[image]

[image]

[image]

[Troubleshooting.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5149878/Troubleshooting.zip)

proektus123, 2020-08-31:
I have tested it on my elitebook  9470m 
VoodooSMBus .kext and VoodooRMI.kext does not loads..  no info in kextstat. I have place these kexts in opencore kexts folder.

LocTran016, 2020-08-31:
> I have tested it on my elitebook 9470m
> VoodooSMBus .kext and VoodooRMI.kext does not loads.. no info in kextstat. I have place these kexts in opencore kexts folder.

You should update (editing the comment) your specs. Did you try to clean snap shot, check that no VoodooPS2 mouse and trackpad in the config.plist, add the device id and vendor id (in SMBus troubleshooting)?

1Revenger1, 2020-09-01:
Hmm, that's weird that it doesn't respond.

Oh, that's weird - VoodooRMI is trying to boot up before rooting occurs? I wonder if it's booting up early or if it's inaccurate logging. It may be worth testing to see if delaying when VoodooRMI starts booting up would help. Easiest way is probably just loading it manually from within macOS to see if anything changes. If you can test this, csrutil needs to allow unsigned kexts to load. Assuming VoodooSMBus, VoodooRMI are in the same folder, you can run `sudo kextutil -vvvv -d VoodooSMBus.kext -d VoodooRMI.kext VoodooRMI.kext/Contents/PlugIns/RMISMBus.kext`. The kexts do need to be owned by root (and I wouldn't recommend modifying the kexts in OC/Kexts), which you can make happen by running `sudo chown -R root:wheel *.kext` (this will make every kext in that directory be owned by Root)

LocTran016, 2020-09-01:
I don't know if this may help:
I'm using this [Touchpad Driver](https://www.dell.com/support/home/en-vn/drivers/driversdetails?driverid=9nnk9&oscode=wt64a&productcode=inspiron-14-3421) for my Dell Insiron 3421 (OS: Window 10 LTSB 2016), and it seem like a lot of laptops sharing the same driver with mine, even the XPS 12 9Q33 which is using the 4th gen intel cpu. I don't have the enough knowledge to be sure that using the same driver mean the same device but I don't know if could take a look at this. Hope this will help you in some way

1Revenger1, 2020-09-01:
[VoodooRMI.kext 2.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5158267/VoodooRMI.kext.2.zip)
This tries to read Address 0x200, which should give the major/minor RMI version if it's an RMI3 device. Do you mind trying this version? It has no other modifications from master.

LocTran016, 2020-09-02:
The mouse doesn't work, I have tried the lastest VoodooRMI.kext but didn't help (I still keep the same VoodooSMBus)
[Troubleshooting.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5159690/Troubleshooting.zip)

1Revenger1, 2020-09-02:
This should properly probe for RMI3 (Switches to Page 0x02 and probes address 0x00)
I also fixed the mapping table possibly not being zeroed at within RMISMBus, but I doubt that was the issue plaguing you.

I'm not expecting this to work for you, but it should hopefully give more helpful debug info
[VoodooRMI.kext 4.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5164950/VoodooRMI.kext.4.zip)

LocTran016, 2020-09-03:
With the latest version of VoodooRMI.kext (I haven't change the VoodooSMBus.kext), my trackpad work perfectly, the launchpad work reliably although there is a small problem that when I first try to click, I have to  tap with one finger twice but from the second time, it work reliably (this also apply to other gestures as well)
[Troubleshooting.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5165766/Troubleshooting.zip)

1Revenger1, 2020-09-03:
Huh. So maybe it was not zeroing the mapping table out at boot. I'll upload the changes later after I've taken a look at the logs. Thanks for testing.

On Monday I'll push out a new release with a release binary

@Rohan200220 any luck getting this to work on your trackpad?

LocTran016, 2020-09-03:
Do you think that I should try the VoodooSMBus as well because the trackpad work perfectly?

1Revenger1, 2020-09-03:
Na, functionality doesn't change at all

1Revenger1, 2020-09-03:
Oh, have you testing this over a few boots to make sure that we weren't being lucky for some reason?

LocTran016, 2020-09-03:
No, I haven't done that yet

LocTran016, 2020-09-03:
I have tried to load the SSDT-SBUS-MCHC (follow dortania guide) and the trackpad doesn't work at all so I delete that SSDT, enable 2 patch to prevent apple smbus kext attach to SMBUS controller (in VoodooSMBus instruction) as well as reset NVRAM but it seem doesn't help at all. Do you think this is a problem with VoodooRMI?
[Troubleshooting.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5168320/Troubleshooting.zip)

Rohan200220, 2020-09-03:
I can't get my trackpad to work with any of the above voodoormi kexts.
I am uploading the log of the VoodooRMI 2 kext. I am using the first VoodooSMBUS kext
[log 2.txt](https://github.com/VoodooSMBus/VoodooRMI/files/5171614/log.2.txt)

Rohan200220, 2020-09-03:
I am using clover, do I need to add any patch in my config.plist for this to work?

1Revenger1, 2020-09-03:
@Rohan200220 you are getting some linking errors. What macOS version is this? I built these test kexts in Catalina 10.15.6, so if your running Mojave I may need to try and build it for an older version.

@LocTran016 huh, the log looks exactly the same as before.
SSDT-SBUS-MCHC will mess up VoodooSMBus so I'd avoid it.
Looks like it isn't able to communicate with the sensor again? That's so weird - I'll need to think on it. It appears to be aan issue which isn't always there which suggests I'm doing something wrong with memory, or something else is happening. Are you able to load/unload this repeatedly within macOS? It'd be good to see if it consistently breaks or doesn't.

Rohan200220, 2020-09-03:
Yes i am running Mojave 10.14.6

1Revenger1, 2020-09-03:
[VoodooRMI.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5171650/VoodooRMI.zip)
This is from Github Actions, which is set to build in High Sierra. See if this version works. When getting logs, use the debug version

[45 more replies]

PR #59 (2020-09-03): Edit log macro to include new line and kext identifier for dmesg
Closes #58
Closes #33 (Turns out I wasn't zeroing out memory and it was causing issues, oops! Suprised it didn't appear sooner)
closed 2020-11-26 by 1Revenger1