VoodooSMBus/VoodooRMI #35: [Feature Request] Palm detection for Trackpoint

SukkaW, 2020-08-01:
VoodooRMI already supports `DisableWhileTypingTimeout`. But it is also common that trackpad having accidental input when using trackpoint.

VoodooSMBus has a configuration `DisableWhileTrackpoint` but it has been removed from VoodooRMI bundled one. So I assume there is no Palm detection for trackpoint at this moment.

1Revenger1, 2020-08-01:
It uses the same value as DisableWhileTypingTimeout. It is there though.

Though, I would like to see about better detecting palm brushes without relying on any sort of timeout. I haven't thought of anything yet though.

gimdh, 2020-08-01:
Personally, disabling Touchpad during trackpoint is good enough - No timeout yet dual input is prohibited.

SukkaW, 2020-08-01:
@gimdh Considering this: It is very common to stop moving trackpoint while move the thumb toward "ThinkNav" for clicking.

1Revenger1, 2020-08-03:
Do you want a seperately controlled value then for the trackpoint disable? It currently just uses the value for DisableWhileTypingTimeout.

gimdh, 2020-08-03:
Lenovo driver lets you configure touchpad deadzone where inputs initiated in the zone are ignored. Similar feature might help.

1Revenger1, 2020-08-03:
A feature like that may be better suited in VoodooInput itself, or possibly in a different touch library. Though, I doubt that'd be hard to implement other than for allowing the swiping gesture for notifications.

gimdh, 2020-08-03:
Setting two rectangular or triangular deadzones from upper left and right corner could be a workaround.

I felt Apple's trackpad driver already has some palm rejection logic built in. (My VoodooI2C touchpad works great yet VoodooI2C has no workaround for palm rejection AFAIK.) Issue with trackpoint is it is not a touchpad and thus cannot stand on Apple's shoulder. So adding custom code to VoodooRMI seems good to me.

PR #43 (2020-08-26): Smooth scroll for trackpoint and update Dependencies
Closes #35
Closes #22

Waiting on @gimdh 's PR to be accepted in VoodooTrackpoint.
commit af72b4ee (2020-08-26): Smooth scroll for trackpoint and update Dependencies (#43)

* Increase speeds, seperate trackpoint timeout to it's own variable
Closes #35
Closes #22

* Add info.plist

* Update dependencies

* Revert to 2:1 PS2 mode
closed 2020-08-26 by 1Revenger1