VoodooSMBus/VoodooRMI #39: Minor] Terminology cleanup

gimdh, 2020-08-06:
I would like to make following changes

* That tiny pointer thingy we all love
HP, Dell, and others use 'Pointer stick'/'Pointing stick'/'Point stick' as a general term, while Lenovo uses trademarked 'TrackPoint'. Wikipedia uses 'Pointing stick'. 

I think codebase should settle with one of popular terms instead of trackstick.

* Touchpad / Trackpad
Touchpad is general term for flat touch input device. Trackpad is Apple's trademarked counterpart.  I think 'touchpad' fits most cases and can avoid confusion with genuine Apple ones.

1Revenger1, 2020-08-06:
Seems sane. I'm most used to Trackpoint, so we may use that (unless you'd prefer one of the other terms). If we use Trackpoint, then maybe it'd make sense to use trackpad? So they both begin with Track? There is a distinction between clickpads and trackpads as well which I think should be made clear as well.

gimdh, 2020-08-06:
I'm using Thinkpad so I prefer Trackpoint too. I just wanted to open possibilities since this driver should work with other laptops with pointing stick, but TrackPoint seems famous enough.

Clickpad is Synaptics' touchpad brand which can be pressed anywhere AFAIK and technically Trackpad can also be clicked anywhere so.. it's kind of mess already. Therefore I really didn't care about this, but I am going to cleanup trackstick anyways and if there's anything should be done, I want to do it at the same time.

1Revenger1, 2020-08-06:
"Click anywhere" Yeah I can't click down the top 30% of my trackpad due to the hinge design

It is a real mess. I would atleast like to make it clear what devices support the force touch emulation, since it's only technically "clickpads" which I have supported right now where you can click down on it. If that distinction can be made clear, then I don't have an issue just switching to trackpad.0

Maybe we need a dictionary or glossary... :P

gimdh, 2020-08-06:
Top 30% is reserved for superheroes. 
I think clickpad is way to go even if it's not terribly accurate. I don't mind both trackpad and touchpad too, but at least want to have consistency. Maybe trackpad is better option since VoodooRMI is a driver for macOS.

1Revenger1, 2020-08-06:
My preference is trackpad, but I don't care that much. Sounds good to me though.

gimdh, 2020-09-03:
It's getting delayed. I think I messed up xcode. I cannot load   kext, even the one from master.

I put VoodooRMI and VoodooSMBus in clover/kext , but  I can't get it to work.
It worked just fine when I worked on trackpoint scroll.

xcodebuild results certificate issue, although it claims build to be successful. 

I'll try another machine and grab pre-built kext there.

1Revenger1, 2020-09-03:
GitHub actions publishes the kext artifact - so you could try going to the actions page for one of your commits and grabbing it. You can grab VoodooSMBus from issue #33 as well, I believe that's up to date.

gimdh, 2020-09-03:
Oh, I have forgotten about CI. I'll give it a shot.

gimdh, 2020-09-04:
Xcode certification error was not an 'error' error, as internet says. It was a different issue(#62).

PR #57 (2020-09-04): Term cleanup, touch up readme
 Deals #39. Terms replaced are as follows. 

> Touchpad -> Trackpad
> Trackstick -> Trackpoint

1Rev ~ Closes #39
closed 2020-09-04 by 1Revenger1