VoodooSMBus/VoodooRMI #40: Click-&-drag lag

simprecicchiani, 2020-08-06:
Hi, thanks for the effort on making this awesome project. I'm using it on a T460s a so far it's pretty much perfect. 
The only issue I found is when click and dragging on text or windows, the pointer remain stuck at the starting point until drag is released, making it unusable to select a portion of text.
When click is made with physical button or with 3-finger drag (set in accessibility settings) the behaviour is correct.

Thanks again

simprecicchiani, 2020-08-06:
A quick demo

[image]

gimdh, 2020-08-06:
I have marginally different T470s and cannot reproduce issue.
Does it work fine with 1-finger (cursor movement) input? Try moving cursor continuously and see if cursor stops.

simprecicchiani, 2020-08-06:
The problem is with the same 1 finger click and drag. When clicking with one and dragging with an other it's fine.

gimdh, 2020-08-06:
Ah, my question was silly. Do you have an issue while pointing(no click)? If it does, I had somewhat similar issue.

simprecicchiani, 2020-08-06:
No, I didn't experience any other issue a part from the click & drag with 1 finger. I'm comfortable with other methods of selecting and moving so I'm totally fine with the kext in general.

gimdh, 2020-08-06:
Could you check log with Console if input was registered correctly?
If you are using release kext, replace it with debug kext, then filter for F11.

1Revenger1, 2020-08-06:
Is your finger particularly flat while doing this? That looks like your triggering force touch. You can disable force touch or increase the Z value to trigger it in the Info.plist for VoodooRMI.

gimdh, 2020-08-06:
I can easily reproduce this with Force touch turned on in prefpane.

simprecicchiani, 2020-08-07:
> Is your finger particularly flat while doing this? That looks like your triggering force touch. You can disable force touch or increase the Z value to trigger it in the Info.plist for VoodooRMI.

That was exactly the cause, thanks! Although force touch is disabled in preference panel, I guess I'll play around with `Info.plist` settings.

closed 2020-08-07 by simprecicchiani

1Revenger1, 2020-08-07:
Yeah, VoodooRMI locks the finger in place if it thinks your doing a force touch, because otherwise it's nearly impossible to use it lol.

simprecicchiani, 2020-09-13:
Sorry to bother you here. I just wanted you to know that this driver is fricking unbelievable! TBH I would have never thought trackpad gestures could become that good on Hackintosh. I'm driving it for a month now and everything is snappy and buttery smooth. Thanks once again for the amazing work you've done.