VoodooSMBus/VoodooRMI #48: Trackpoint Middle Button Behavior

1299651405, 2020-08-16:
The middle Trackpoint button on my X270 emits a mouse click on both the press and release, i.e. middle clicking a link on Safari or Firefox will open two new tabs. This also affects scrolling, if you start scrolling over a link the link will open in a new tab as well as scrolling the page.

I am using the latest release on a Thinkpad X270.

1Revenger1, 2020-08-16:
Hrm, I thought I fixed that a while ago.

Do you mind providing a log as the template said to provide?
You may want to use `log stream | grep -i voodoormi > ~/Desktop/voodoormi.txt`, and then use the middle button to try and scroll, before ending that log command.

1299651405, 2020-08-16:
Sorry, here is `log show --last boot` 
```
2020-08-16 06:13:35.524031+0100 0xde       Default     0x0                  0      0    kernel: (VoodooSMBus) Debug: Publishing nub for slave device at address 0x2c
2020-08-16 06:13:35.535612+0100 0xde       Default     0x0                  0      0    kernel: (VoodooSMBus) Debug: Publishing nub for slave device at address 0x15
2020-08-16 06:13:35.587226+0100 0xe5       Default     0x0                  0      0    kernel: (VoodooSMBus) Debug: No response
2020-08-16 06:13:35.685429+0100 0x11a      Default     0x0                  0      0    kernel: (VoodooPS2Controller) ApplePS2Controller: Notification consumer published: ApplePS2Controller
2020-08-16 06:13:35.696328+0100 0x11b      Default     0x0                  0      0    kernel: (VoodooPS2Controller) ApplePS2Controller: Notification consumer published: ApplePS2Keyboard
2020-08-16 06:13:41.976557+0100 0x5e4      Default     0x0                  0      0    kernel: (VoodooSMBus) Debug: No response
2020-08-16 06:13:41.980785+0100 0x5e4      Default     0x0                  0      0    kernel: (VoodooRMI) Error: Unable to read IRQ
2020-08-16 06:13:42.118114+0100 0x62a      Default     0x0                  0      0    kernel: (VoodooSMBus) Debug: Lost arbitration
2020-08-16 06:13:42.578099+0100 0x6de      Default     0x0                  0      0    kernel: (VoodooSMBus) Debug: Lost arbitration
2020-08-16 06:18:11.301005+0100 0x43e2     Default     0x0                  0      0    kernel: (VoodooSMBus) Debug: Lost arbitration
2020-08-16 06:20:14.351193+0100 0x5634     Default     0x0                  0      0    kernel: (VoodooSMBus) Debug: Lost arbitration
2020-08-16 06:25:04.525661+0100 0x8ba2     Default     0x0                  0      0    kernel: (VoodooSMBus) Debug: Lost arbitration
2020-08-16 06:28:00.462235+0100 0xad09     Default     0x0                  0      0    kernel: (VoodooSMBus) Debug: No response
2020-08-16 06:28:00.462251+0100 0xad09     Default     0x0                  0      0    kernel: (VoodooRMI) Error: Could not read F11 attention data: -6
2020-08-16 06:28:02.143113+0100 0xada5     Default     0x0                  0      0    kernel: (VoodooSMBus) Debug: Lost arbitration
```
Looks like I need to patch my IRQ?

No output in `log stream | grep -I voodoo`

1Revenger1, 2020-08-16:
Try using the debug version of VoodooRMI, it gives a *lot* more info. Most of the debug info isn't needed and just wastes CPU cycles so it's left out of the release build.
No need to patch IRQ, it just wasn't able to read it once. If it continued working, it likely isn't an issue. This is the IRQ register within the sensor itself, not IRQ like with HPET/TMR, etc.

1299651405, 2020-08-16:
Using the debug version seems to reduce the probability of two clicks being sent for one press, it now seems to happen 50% of the time, but not in a truly random fashion, it seems to get 'stuck' on either sending one click or two, I'm not sure what triggers this yet.

When it is working properly, i.e only opening a link when I release the middle button it will still open a link when I start to scroll over a link. Like this:

Button Press -> nothing
Move Trackpoint -> Opens link AND begins scrolling
Button Release -> Ends Scroll (won't open a link if scroll ends over one)

I've also notices this sometimes happens with the primary buttons, but very rarely, it might send a double left click for a single press 1% of the time.

`log stream` output during scroll
```
2020-08-16 15:04:50.121332+0100 0xfb04     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: 28
2020-08-16 15:04:50.121357+0100 0xfb04     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: 0
2020-08-16 15:04:50.121365+0100 0xfb04     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: fe
2020-08-16 15:04:50.121413+0100 0xfb04     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: Sending event 3825174530 to F11/F12: 0
2020-08-16 15:04:50.121424+0100 0xfb04     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: Dx: 0 Dy : 1, Buttons: 0
2020-08-16 15:04:50.122089+0100 0xfb04     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: Key 1 is Up
2020-08-16 15:04:50.122098+0100 0xfb04     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: Key 2 is Up
2020-08-16 15:04:50.122103+0100 0xfb04     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: Key 3 is Down
2020-08-16 15:04:50.122128+0100 0xfb04     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: Sending trackpoint button to F03: 4
2020-08-16 15:04:50.151842+0100 0xfb08     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: 28
2020-08-16 15:04:50.151867+0100 0xfb08     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: 0
2020-08-16 15:04:50.151876+0100 0xfb08     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: ff
2020-08-16 15:04:50.151943+0100 0xfb08     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: Dx: 0 Dy : 0, Buttons: 0
2020-08-16 15:04:50.165134+0100 0xfb0a     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: 28
2020-08-16 15:04:50.165145+0100 0xfb0a     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: 0
2020-08-16 15:04:50.165148+0100 0xfb0a     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: ff
2020-08-16 15:04:50.165165+0100 0xfb0a     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: Dx: 0 Dy : 0, Buttons: 0
2020-08-16 15:04:50.202730+0100 0xfb0c     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: 28
2020-08-16 15:04:50.202757+0100 0xfb0c     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: 0
2020-08-16 15:04:50.202761+0100 0xfb0c     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: fe
2020-08-16 15:04:50.202834+0100 0xfb0c     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: Sending event 3825174530 to F11/F12: 0
2020-08-16 15:04:50.202841+0100 0xfb0c     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: Dx: 0 Dy : 1, Buttons: 0
2020-08-16 15:04:50.217502+0100 0xfb0e     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: 28
2020-08-16 15:04:50.217516+0100 0xfb0e     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: 0
2020-08-16 15:04:50.217518+0100 0xfb0e     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: fe
2020-08-16 15:04:50.217555+0100 0xfb0e     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: Sending event 3825174530 to F11/F12: 0
2020-08-16 15:04:50.217561+0100 0xfb0e     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: Dx: 0 Dy : 1, Buttons: 0
2020-08-16 15:04:50.224439+0100 0xfb0f     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: 38
2020-08-16 15:04:50.224450+0100 0xfb0f     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: ff
2020-08-16 15:04:50.224452+0100 0xfb0f     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: F03 - Recieved data over PS2: fe
2020-08-16 15:04:50.224490+0100 0xfb0f     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: Sending event 3825174530 to F11/F12: 0
2020-08-16 15:04:50.224495+0100 0xfb0f     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: Dx: 0 Dy : 1, Buttons: 0
[... cut ...]
2020-08-16 15:04:50.637638+0100 0xfb1e     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: Key 1 is Up
2020-08-16 15:04:50.637661+0100 0xfb1e     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: Key 2 is Up
2020-08-16 15:04:50.637665+0100 0xfb1e     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: Key 3 is Up
2020-08-16 15:04:50.637678+0100 0xfb1e     Default     0x0                  0      0    kernel: (VoodooRMI) Debug: Sending trackpoint button to F03: 0
```

1299651405, 2020-08-16:
I've been looking at the `log stream` outputs a lot, trying to catch something odd, but it always seems to agree with my inputs, no extra button presses anywhere, time between button presses is consistent, no delay etc. 
Could it be something further down the input stack?

1Revenger1, 2020-08-17:
Oh, I see what the issue is - generally trackstick buttons are sent over the PS2 passthrough, but sometimes they are instead sent through a different function. Mine functions as the former, so I unfortunately wasn't able to catch it. When the buttons are sent to the function which handles the PS2 passthrough, it also sends a [pointer event w/ buttons](https://github.com/VoodooSMBus/VoodooRMI/blob/master/VoodooRMI/Functions/F03.cpp#L266) to macOS which by passes logic I had in place to avoid it...oops.

1Revenger1, 2020-08-17:
Do you mind trying this version out? Should be fixed.
[VoodooRMI.kext.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5086529/VoodooRMI.kext.zip)

1299651405, 2020-08-17:
Working perfectly. Thank you for your efforts.

PR #50 (2020-08-17): Fix F30->F03 bypassing logic in handlePacket
Closes #48
closed 2020-08-17 by 1Revenger1