VoodooSMBus/VoodooRMI #62: VoodooSMBus Dependency resolution failure

gimdh, 2020-09-04:
**Describe the bug**
When trying to load RMISMbus.kext following error occurs.

**To Reproduce**
Build with xcodebuild and try loading with kextutil.

**Expected behavior**
Loads properly

**Log**
```Dependency Resolution Failures: 
    Kexts found for these libraries lack valid OSBundleCompatibleVersion: 
        de.leo-labs.VoodooSMBus```

gimdh, 2020-09-04:
I wonder if I used wrong VoodooSMBus. I tried latest release from
VoodooSMBus/VoodooSMBus and this repo.

gimdh, 2020-09-04:
In **VoodooRMI/VoodooRMI/Transports/SMBus/Info.plist**, dependency is defined as follows
```
<key>OSBundleLibraries</key>
	<dict>
		<key>de.leo-labs.VoodooSMBus</key>
		<string>3.0</string>
```

However, VoodooSMBus does not have `OSBundleCompatibleVersion` key in info.plist. I added lines below, and now it loads properly.

```
        <key>OSBundleCompatibleVersion</key>
        <string>1.0</string>
```

1Revenger1, 2020-09-04:
The dev branch of VoodooSMBus has the correct version, which is what this is based on. I did indeed bump up the version, and I should have a release on Monday with the new voodooSMBus in it.

gimdh, 2020-09-05:
I'll keep in mind that I should use dev branch. Thanks!

closed 2020-09-05 by gimdh