VoodooSMBus/VoodooRMI #67: Are Touchscreen <SYNA2393> and Touchpad <SYNA1D31> supported?

profzei, 2020-09-23:
Hi,
I'm trying using your project for the two devices in the object.
* I attach a .zip file with my compressed EFI.
* I disabled `SSDT-SBUS.aml`
* `SSDT-TPXX.aml` controls `TPD0` (which is Touchscreen-> `SYNA2393`) and `TPL1` (which is Touchpad-> `SYNA1D31`) using VoodooI2C and VoodooI2CHID with GPI0 pinning (they were working but with the [following issue]( https://github.com/profzei/Matebook-X-Pro-2018/issues/46), so it was suggested to give a try to your project!)
* in my `config.plist` I disabled all calls to VoodooI2C and VoodooI2CHID and enabled (but I'm not sure whether the loading order is right and whether I need to load also Trackpoint) calls to VoodooRMI and VoodooSMBus

Could you give it a look, please?
Thanks in advance! @profzei 
[MateBookXPro-OC-v1.6.0-beta.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5266624/MateBookXPro-OC-v1.6.0-beta.zip)

profzei, 2020-09-23:
Hi, This is the associated IOReg dump:
[MBXP.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5268753/MBXP.zip)

1Revenger1, 2020-09-23:
With the PCIClassMatch, I'd just leave it alone. It should match to any SMBus controller already as it is, and I need to update the docs to reflect that, oops.

Tbh, I've never seen an SMBus touchscreen. It may be that your trackpad is only able to be seen over I2C as well, rather than SMBus. It's sort of rare in my experience for it to appear on both busses. VoodooRMI is compatible with I2C devices though, if you want to try enabling VoodooI2C again and loading RMII2C.kext. You might want to try using the version from #65 if you come across an issue similar to #64 though where it resets randomly. I'm not sure how well it works for touchscreens though, and you may want to use VoodooI2CHID for that. Otherwise I have no clue why it'd randomly stop working after sleep, and you may benefit from opening an issue for VoodooI2C and providing logs assuming that VoodooI2CHID behaves better than trying to use VoodooRMI.

gimdh, 2020-09-25:
I closed an isssue since I was not able to reproduce it, but I had similar problem with VoodooRMI; No touchpad after sleep. If that's the issue, I would recommend to report debug log to VoodooI2C while trying VoodooRMI.

profzei, 2020-09-25:
@gimdh :I have already reported this issue to VoodooI2C a few months ago... but without a definitive success...

@1Revenger1 : How can I attach my TPL1 device (Touchpad) to VoodooRMI and  my TPD0 device (touchscreen) to VoodooI2CHID

Now I'm trying the configuration:
- at first loading VoodooI2C
- then loading VoodooRMI (->RMII2C)
... sequence is right?
With this loading sequence my touchscreen is not working... 

...need more digging for "touchpad after wake-up"...

1Revenger1, 2020-09-25:
In the Info.plist for RMII2C.kext, you may need to add the names of your trackpad.

zhen-zen, 2020-09-26:
> In the Info.plist for RMII2C.kext, you may need to add the names of your trackpad.

The matching entry was replaced to `PNP0C50` for all compatible input. If you want to specify a single device, just replace it with `SYNAXXXX`(your id). However, the touchscreen should be supported as well.

The issue after sleep seems to be I2C controller problem regarding default clock speed timing. On my laptop, I have to add following parameter to sensor hub (I2C3) to get it initialized correctly. And that also fixed the touchscreen (I2C0) issue after sleep (patch not required for this one). The related log is like `slave address not acknowledged (7bit mode)` (or `lost arbitration`? I forgot that).

```
    External (_SB_.PCI0.I2C3, DeviceObj)

    Scope (_SB.PCI0.I2C3)
    {
        Name (SSCN, Package (0x03)
        {
            0x01B0, 
            0x01FB, 
            0x1E
        })
        Name (FMCN, Package (0x03)
        {
            0x48, 
            0xA0, 
            0x1E
        })
    }
```

I got this set of values from coreboot, and there are other sets such as 
```
Name (SSCN, Package () { 0x210, 0x280, 0x1e })
Name (FMCN, Package () { 0x80, 0xa0, 0x1e })
```
And `FMCN` could also be:
```
Name (FMCN, Package () { 0x101, 0x12C, 0x62 })
```

profzei, 2020-09-28:
@zhen-zen Thank you for your hints!

This is the result of command `sudo dmesg | grep -I "Voodoo"`                          

```
VoodooI2CSensorHubEnabler::INT33D0 Found Intel ACPI Sensor Hub Enabler
VoodooGPIOSunrisePointLP::Loading GPIO Data for SunrisePoint-LP
VoodooGPIOSunrisePointLP::VoodooGPIO Init!
VoodooGPIOSunrisePointLP::VoodooGPIO Initializing Community 0
VoodooGPIOSunrisePointLP::VoodooGPIO Initializing Community 1
VoodooGPIOSunrisePointLP::VoodooGPIO Initializing Community 2
VoodooGPIOSunrisePointLP::GPIO Controller is already awake! Not reinitializing.
VoodooI2CSensorHubEnabler::INT33D0 Enabled Sensor Hub
VoodooI2CPCIController::pci8086,9d61 Starting I2C controller
VoodooI2CPCIController::pci8086,9d61 Set PCI power state D0
VoodooI2CPCIController::pci8086,9d61 Publishing nub
line: 5478 VoodooI2CPCIController::pci8086,9d60 Starting I2C controller
VoodooI2CControllerDriver::pci8086,9d61 Probing controller
VoodooI2CControllerDriver::pci8086,9d61 Found valid Synopsys component, continuing with initialisation
VoodooI2CPCIController::pci8086,9d60 Set PCI power state D0
VoodooI2CControllerNub::pci8086,9d61 SSCN not implemented in ACPI tables
VoodooI2CPCIController::pci8086,9d60 Publishing nub
VoodooI2CControllerNub::pci8086,9d61 FMCN not implemented in ACPI tables
VoodooI2CControllerDriver::pci8086,9d61 Warning: Error getting bus config, using defaults where necessary
VoodooI2CControllerDriver::pci8086,9d60 Probing controller
VoodooI2CControllerDriver::pci8086,9d61 Publishing device nubs
VoodooI2CControllerDriver::pci8086,9d60 Found valid Synopsys component, continuing with initialisation
VoodooI2CControllerNub::pci8086,9d60 SSCN not implemented in ACPI tables
VoodooI2CControllerDriver::pci8086,9d61 Found I2C device: SYNA1D31
VoodooI2CControllerNub::pci8086,9d60 FMCN not implemented in ACPI tables
VoodooI2CControllerDriver::pci8086,9d60 Warning: Error getting bus config, using defaults where necessary
VoodooI2CControllerDriver::pci8086,9d60 Publishing device nubs
VoodooI2CControllerDriver::pci8086,9d60 Found I2C device: SYNA2393
VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOSunrisePointLP
VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOSunrisePointLP
VoodooI2CHIDDevice:0x1000003dd start
VoodooGPIOSunrisePointLP::Registering hardware pin 0x55 for GPIO IRQ pin 0x55
VoodooGPIOSunrisePointLP::Successfully registered hardware pin 0x55 for GPIO IRQ pin 0x55
VoodooI2CHIDDevice:0x1000003dc start
VoodooGPIOSunrisePointLP::Registering hardware pin 0x91 for GPIO IRQ pin 0x91
VoodooGPIOSunrisePointLP::Successfully registered hardware pin 0x91 for GPIO IRQ pin 0x91
VoodooI2CHIDDevice::SYNA1D31 Device initiated reset accomplished
VoodooI2CHIDDevice::SYNA2393 Device initiated reset accomplished
VoodooI2CHIDDevice:0x1000003dc creating interfaces
VoodooInputSimulatorDevice:0x1000003fe start
VoodooI2CHIDDevice:0x1000003dd creating interfaces
VoodooInputSimulatorDevice:0x10000040a start
VoodooI2CHIDDevice:0x1000003dc open by VoodooI2CTouchscreenHIDEventDriver 0x1000003f8 (0x0)
VoodooInputActuatorDevice:0x10000040e start
VoodooInputSimulatorDevice:0x1000003fe creating interfaces
VoodooInputActuatorDevice:0x10000040e creating interfaces
VoodooInputActuatorDevice:0x10000040e open by AppleActuatorHIDEventDriver 0x100000413 (0x0)
VoodooInputSimulatorDevice:0x1000003fe open by AppleMultitouchTrackpadHIDEventDriver 0x100000414 (0x0)
VoodooI2CHIDDevice:0x1000003dd open by VoodooI2CPrecisionTouchpadHIDEventDriver 0x100000401 (0x0)
VoodooI2CPrecisionTouchpadHIDEventDriver::SYNA1D31 Putting device into Precision Touchpad Mode
VoodooInputActuatorDevice:0x10000041c start
VoodooInputSimulatorDevice:0x10000040a creating interfaces
VoodooInputActuatorDevice:0x10000041c creating interfaces
VoodooInputSimulatorDevice:0x10000040a open by AppleMultitouchTrackpadHIDEventDriver 0x100000421 (0x0)
VoodooInputActuatorDevice:0x10000041c open by AppleActuatorHIDEventDriver 0x10000042a (0x0)
ApplePS2Controller: Notification consumer published: VoodooI2CPrecisionTouchpadHIDEventDriver
VoodooInputSimulatorDevice:0x10000040a new user client
VoodooInputSimulatorDevice:0x10000040a open by IOHIDLibUserClient 0x100000513 (0x0)
schedule_before_userret(1516): 4/4 tasks scheduledVoodooInputSimulatorDevice:0x1000003fe new user client
VoodooInputSimulatorDevice:0x1000003fe open by IOHIDLibUserClient 0x100000514 (0x0)
VoodooI2CHIDDevice:0x1000003dd new user client
VoodooI2CHIDDevice:0x1000003dd open by IOHIDLibUserClient 0x100000515 (0x0)
VoodooI2CPrecisionTouchpadHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
VoodooI2CTouchscreenHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
Sandbox: apsd(99) deny(1) mach-lookup com.apple.imagent.apsSandbox: fmfd(393) deny(1) ipc-posix-shm-write-create com.apple.AppleDatabaseChangedSandbox: fmfd(393) deny(1) file-write-data /Users/riccardo/Library/Preferences/com.apple.icloud.fmfd.notbackedup.plistSandbox: 1 duplicate report for fmfd deny(1) file-write-data /Users/riccardo/Library/Preferences/com.apple.icloud.fmfd.notbackedup.plistSandbox: mds(88) System Policy: allow file-read-metadata /private/var/folders/p4/bzpjjdxn2bs334sbq_1scy0m0000gn/0/com.apple.ScreenTimeAgent/Store/RMAdminStore-Local.sqliteSandbox: mds(88) System Policy: allow file-read-data /private/var/folders/p4/bzpjjdxn2bs334sbq_1scy0m0000gn/0/com.apple.ScreenTimeAgent/Store/RMAdminStore-Local.sqliteSandbox: mds(88) System Policy: allow file-read-metadata /private/var/folders/p4/bzpjjdxn2bs334sbq_1scy0m0000gn/0/com.apple.Scre [... cut ...]
VoodooGPIOSunrisePointLP::Registering hardware pin 0x55 for GPIO IRQ pin 0x55
VoodooGPIOSunrisePointLP::Successfully registered hardware pin 0x55 for GPIO IRQ pin 0x55
VoodooI2CHIDDevice::SYNA1D31 Woke up
```

As you can see I found out `SSCN` and `FMCN` related issue!
But now I need more digging:
- you stated that you found your values **from coreboot**... what's coreboot? a Linux distro? How can I use it for finding **my values**?
- do you think the values you suggested for `SSCN` and `FMCN` can also be applied to my `pci8086,9d61` and `pci8086,9d60`?

benbender, 2020-09-30:
@profzei coreboot is a opensource-bios-implementation. See f.e. https://github.com/coreboot/coreboot/search?q=FMCN

zhen-zen, 2020-10-01:
> @zhen-zen Thank you for your hints!
> 
> This is the result of command `sudo dmesg | grep -I "Voodoo"`
> 
> ```
> VoodooI2CSensorHubEnabler::INT33D0 Found Intel ACPI Sensor Hub Enabler
> VoodooGPIOSunrisePointLP::Loading GPIO Data for SunrisePoint-LP
> VoodooGPIOSunrisePointLP::VoodooGPIO Init!
> VoodooGPIOSunrisePointLP::VoodooGPIO Initializing Community 0
> VoodooGPIOSunrisePointLP::VoodooGPIO Initializing Community 1
> VoodooGPIOSunrisePointLP::VoodooGPIO Initializing Community 2
> VoodooGPIOSunrisePointLP::GPIO Controller is already awake! Not reinitializing.
> VoodooI2CSensorHubEnabler::INT33D0 Enabled Sensor Hub
> VoodooI2CPCIController::pci8086,9d61 Starting I2C controller
> VoodooI2CPCIController::pci8086,9d61 Set PCI power state D0
> VoodooI2CPCIController::pci8086,9d61 Publishing nub
> line: 5478 VoodooI2CPCIController::pci8086,9d60 Starting I2C controller
> VoodooI2CControllerDriver::pci8086,9d61 Probing controller
> VoodooI2CControllerDriver::pci8086,9d61 Found valid Synopsys component, continuing with initialisation
> VoodooI2CPCIController::pci8086,9d60 Set PCI power state D0
> VoodooI2CControllerNub::pci8086,9d61 SSCN not implemented in ACPI tables
> VoodooI2CPCIController::pci8086,9d60 Publishing nub
> VoodooI2CControllerNub::pci8086,9d61 FMCN not implemented in ACPI tables
> VoodooI2CControllerDriver::pci8086,9d61 Warning: Error getting bus config, using defaults where necessary
> VoodooI2CControllerDriver::pci8086,9d60 Probing controller
> VoodooI2CControllerDriver::pci8086,9d61 Publishing device nubs
> VoodooI2CControllerDriver::pci8086,9d60 Found valid Synopsys component, continuing with initialisation
> VoodooI2CControllerNub::pci8086,9d60 SSCN not implemented in ACPI tables
> VoodooI2CControllerDriver::pci8086,9d61 Found I2C device: SYNA1D31
> VoodooI2CControllerNub::pci8086,9d60 FMCN not implemented in ACPI tables
> VoodooI2CControllerDriver::pci8086,9d60 Warning: Error getting bus config, using defaults where necessary
> VoodooI2CControllerDriver::pci8086,9d60 Publishing device nubs
> VoodooI2CControllerDriver::pci8086,9d60 Found I2C device: SYNA2393
> VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOSunrisePointLP
> VoodooI2CDeviceNub::Got GPIO Controller! VoodooGPIOSunrisePointLP
> VoodooI2CHIDDevice:0x1000003dd start
> VoodooGPIOSunrisePointLP::Registering hardware pin 0x55 for GPIO IRQ pin 0x55
> VoodooGPIOSunrisePointLP::Successfully registered hardware pin 0x55 for GPIO IRQ pin 0x55
> VoodooI2CHIDDevice:0x1000003dc start
> VoodooGPIOSunrisePointLP::Registering hardware pin 0x91 for GPIO IRQ pin 0x91
> VoodooGPIOSunrisePointLP::Successfully registered hardware pin 0x91 for GPIO IRQ pin 0x91
> VoodooI2CHIDDevice::SYNA1D31 Device initiated reset accomplished
> VoodooI2CHIDDevice::SYNA2393 Device initiated reset accomplished
> VoodooI2CHIDDevice:0x1000003dc creating interfaces
> VoodooInputSimulatorDevice:0x1000003fe start
> VoodooI2CHIDDevice:0x1000003dd creating interfaces
> VoodooInputSimulatorDevice:0x10000040a start
> VoodooI2CHIDDevice:0x1000003dc open by VoodooI2CTouchscreenHIDEventDriver 0x1000003f8 (0x0)
> VoodooInputActuatorDevice:0x10000040e start
> VoodooInputSimulatorDevice:0x1000003fe creating interfaces
> VoodooInputActuatorDevice:0x10000040e creating interfaces
> VoodooInputActuatorDevice:0x10000040e open by AppleActuatorHIDEventDriver 0x100000413 (0x0)
> VoodooInputSimulatorDevice:0x1000003fe open by AppleMultitouchTrackpadHIDEventDriver 0x100000414 (0x0)
> VoodooI2CHIDDevice:0x1000003dd open by VoodooI2CPrecisionTouchpadHIDEventDriver 0x100000401 (0x0)
> VoodooI2CPrecisionTouchpadHIDEventDriver::SYNA1D31 Putting device into Precision Touchpad Mode
> VoodooInputActuatorDevice:0x10000041c start
> VoodooInputSimulatorDevice:0x10000040a creating interfaces
> VoodooInputActuatorDevice:0x10000041c creating interfaces
> VoodooInputSimulatorDevice:0x10000040a open by AppleMultitouchTrackpadHIDEventDriver 0x100000421 (0x0)
> VoodooInputActuatorDevice:0x10000041c open by AppleActuatorHIDEventDriver 0x10000042a (0x0)
> ApplePS2Controller: Notification consumer published: VoodooI2CPrecisionTouchpadHIDEventDriver
> VoodooInputSimulatorDevice:0x10000040a new user client
> VoodooInputSimulatorDevice:0x10000040a open by IOHIDLibUserClient 0x100000513 (0x0)
> schedule_before_userret(1516): 4/4 tasks scheduledVoodooInputSimulatorDevice:0x1000003fe new user client
> VoodooInputSimulatorDevice:0x1000003fe open by IOHIDLibUserClient 0x100000514 (0x0)
> VoodooI2CHIDDevice:0x1000003dd new user client
> VoodooI2CHIDDevice:0x1000003dd open by IOHIDLibUserClient 0x100000515 (0x0)
> VoodooI2CPrecisionTouchpadHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
> VoodooI2CTouchscreenHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
> Sandbox: apsd(99) deny(1) mach-lookup com.apple.imagent.apsSandbox: fmfd(393) deny(1) ipc-posix-shm-write-create com.apple.AppleDatabaseChangedSandbox: fmfd(393) deny(1) file-write-data /Users/riccardo/Library/Preferences/com.apple.icloud.fmfd.notbackedup.plistSandbox: 1 duplicate report for fmfd deny(1) file-write-data /Users/riccardo/Library/Preferences/com.apple.icloud.fmfd.notbackedup.plistSandbox: mds(88) System Policy: allow file-read-metadata /private/var/folders/p4/bzpjjdxn2bs334sbq_1scy0m0000gn/0/com.apple.ScreenTimeAgent/Store/RMAdminStore-Local.sqliteSandbox: mds(88) System Policy: allow file-read-data /private/var/folders/p4/bzpjjdxn2bs334sbq_1scy0m0000gn/0/com.apple.ScreenTimeAgent/Store/RMAdminStore-Local.sqliteSandbox: mds(8 [... cut ...]
> **Update**: I do not have any `I2C3` device in my `DSDT.aml` file... nor any `SHUB` device... do you thing your values need to be applied to `I2C0` and `I2C1` devices or otherwise to `TPD0` and `TPL1` devices?
> [DSDT.aml.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5295313/DSDT.aml.zip)

Missing `SSCN` and `FMCN` is not an issue, there's a default value set for them. Actually hardcoded parameters like `SSCN` is deprecated. You can use them as keywords in coreboot mailing list and find some explanation like [this](https://mail.coreboot.org/pipermail/coreboot-gerrit/2016-July/046924.html). This commit also has the answer to your second question.

profzei, 2020-10-02:
@benbender & @zhen-zen : thank you very much for your reply and your help!
At the moment I added `SSCN` and `FMCN` only to `I2C0` and `I2C1`... since `TPD0` (touchscreen) belongs to Device `I2C0` and `TPL1` (touchpad) belongs to Device `I2C1`.

I added the same following code to both `I2CX`:
```
Scope (_SB.PCI0.I2C0) //touchscreen
    {
        Name (SSCN, Package (0x03)
        {
            0x01B0, 
            0x01FB, 
            0x1E
        })
        Name (FMCN, Package (0x03)
        {
            0x48, 
            0xA0, 
            0x1E
        })
    }
```
and 
```
Scope (_SB.PCI0.I2C1) // touchpad
    {
        Name (SSCN, Package (0x03)
        {
            0x01B0, 
            0x01FB, 
            0x1E
        })
        Name (FMCN, Package (0x03)
        {
            0x48, 
            0xA0, 
            0x1E
        })
    }
```
but, probably I need to change `SSCN` and `FMCN` for `I2C1` since they are different from those provided for the device `I2C1` in the linked document... @zhen-zen Do you agree? **At the moment** I do not have any issues with my (not perfect...) choice... but probably it's better also to try the specific values for `I2C1`...

zhen-zen, 2020-10-04:
So the custom `SSCN`/`FMCN` fixed the issue? And I think it's not necessary to add them to touchscreen which is not involved.

Also I wonder if your touchscreen works with VoodooRMI, or how does that compared to VoodooI2CHID?

profzei, 2020-10-04:
> So the custom SSCN/FMCN fixed the issue? And I think it's not necessary to add them to touchscreen which is not involved.

@zhen-zen : what I can say at the moment is that the custom SSCN/FMCN **seems** to fix the issue... I went through 20 sleep & wake-up phases without any issues... I am also waiting for the response of another user **@wiregen** who is helping me in this check phase.
What I really do not understand is that using 
```
Scope (_SB.PCI0.I2C1) // touchpad
    {
        Name (SSCN, Package (0x03)
        {
            0x01B0, 
            0x01FB, 
            0x1E
        })
        Name (FMCN, Package (0x03)
        {
            0x48, 
            0xA0, 
            0x1E
        })
    }
```
which is specific for `I2C1` i.e. for my touchpad device `TPL1` then `TPL1` is affected by the above mentioned issue, while using the values specific for `I2C0` instead the issue seems to be fixed...

> Also I wonder if your touchscreen works with VoodooRMI, or how does that compared to VoodooI2CHID?

@zhen-zen : the touchscreen is working with `VoodooRMI` but the feeling is better with `VoodooI2CHID`... with `VoodooRMI`, for example, I need to press a little bit more the screen for making gestures effective... the gestures are a little bit less smooth...

1Revenger1, 2020-10-04:
It may be that the values reported by VoodooRMI for the touchscreen are different, so it rejects some fingers thinking they are palms/accidental touches. Best way to tell would be a log from the debug version.

zhen-zen, 2020-10-09:
> @zhen-zen : what I can say at the moment is that the custom SSCN/FMCN **seems** to fix the issue... I went through 20 sleep & wake-up phases without any issues... I am also waiting for the response [...]

Because controllers might have different speed configurations. And I would like to ask you to revert the `TPXX` hack after upgrading to VoodooI2C 2.5.2 .

1Revenger1, 2020-10-10:
Closing since this was a configuration issue most likely

closed 2020-11-25 by 1Revenger1