VoodooSMBus/VoodooRMI #69: glitchy behaviour on thinkpad x1 extreme

antgeth, 2020-09-27:
i just attempted to install this on my x1 hoping it would solve the click-and-drag problems i had with PS2.  i _believe_ i followed the instructions properly.

however the touchpad behaviour is now very erratic. it seems to time out constantly, and response is intermittent.  the log has about a dozen messages like this over the course of 15 minutes.
`
2020-09-27 16:30:01.845267-0700 0x9ec3     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
`

sitting in my clover kext folder are VoodooRMI and VoodooSMBus as supplied here, and the latest version of VoodooPS2Controller minus all of the kexts but VoodooPS2Keyboard.  did i make some obvious error?  i hope i can get to the bottom of this as the touchpad is one of the last outstanding, frustrating, issues i have with this hack.

1Revenger1, 2020-09-27:
Huh, this was developed mostly for my X1 extreme so that's really weird that it's behaving erratically. What other kexts do you have and ACPI modifications? I'm using OpenCore 0.6.1 as well, which may make a difference?
Only ~12 error messages is probably fine tbh, it sends hundreds of packets and one dropped packet isn't a big deal.

gimdh, 2020-10-08:
Try without VoodooTrackpoint to see if it works any better. Mine also had many -6 error codes, and this might be related with #8.

1Revenger1, 2020-10-09:
Closing this as Auther hasn't responded for a couple weeks, and laggy/glitchy trackpad info can be found in #63

closed 2020-10-09 by 1Revenger1