VoodooSMBus/VoodooRMI #71: support 2nd Generation Sandbridge?

ryulove3, 2020-10-03:
1. HP 8570w (3rd Ivy bridge). SBUS@1F,3 (1e228086) 
OS -- Mojave
Upper 3button & trackstick not working
bottom 3button working (middle button wired)
Touchpad working (all gestures.. good)

2. HP 850G1 (4th Haswall), SBUS@1F,3 (9c228086) 
OS -- Catalina
(trackstick not test)
bottom 2 button & Touchpad working (all gestures..so good)

3. HP 8460p (2nd sandbridge) - OS : Sierra
not connected SBUS. blank.
What should i do?

1Revenger1, 2020-10-03:
I build these on High Sierra with a minimum target of 10.13 - it may work if you update to High Sierrra, or later today I can get the OS target down to a minimum of 10.11 (Planned to do it earlier but forgot). VoodooSMBus would need modifications as well though.

1Revenger1, 2020-10-04:
This should work with Sierra as well now - I'll be making a release tomorrow for VoodooRMI, and I'll include a version of VoodooSMBus which should work with Sierra as well.

ryulove3, 2020-10-07:
still do not work even in version 1.2. 
SBUS .. blank
The system is unstable on "hd3000 + High Sierra or higher --> not supoorted nativly". I can't upgrade OS

(Thank you for your effort).

1Revenger1, 2020-10-07:
[VoodooSMBus.kext.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5343458/VoodooSMBus.kext.zip)
Mind trying this version and see if VoodooSMBus loads?

ryulove3, 2020-10-08:
Still blank..

1Revenger1, 2020-10-08:
is it in `kextstat`? (`kextstat | grep -i VoodooSMBus`). Mind adding `msgbuf=1048576` to your boot args as well, and running `sudo dmesg | grep -i VSMB > ~/Desktop/log.txt` and providing the log as well?

ryulove3, 2020-10-08:
I'll test your seggestion.
Now, Installing mojave only for testing. I'll compare it.

ryulove3, 2020-10-08:
The os version seems to be irrelevant. 
Both mojave& sierra are not connected. 
I can't get any debug messages. kext does not load(inject& install). 
RehabMan's driver is the only one that works (synapstic)

gimdh, 2020-10-08:
You can try loading kext online after boot with kextutil.

1Revenger1, 2020-10-09:
I'm realizing that I had some of the dependency versions wrong, which might've prevented Mojave and older - mind trying these?
[VoodooSMBus.kext.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5357674/VoodooSMBus.kext.zip)
[VoodooRMI.kext.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5357676/VoodooRMI.kext.zip)

ryulove3, 2020-11-14:
with Clover r5126, HP 850 G1 Catalina 10.15.7 ---> big sur. 
everything's good. 
but can't load 'voodooinput'. 
I removed voodooinput v1.0.7 and add  v1.0.8(acidanthera), Touchpad is working.
I don't know Why ..

------------
HP 8570w (Mojave & catalina) .
When scrolling or moving cursor, somtimes click or stop scrolling. 
I thought '8570w's touchpad is bad'. but no problems with Rehabman's kext(at least moving cursor or scrolling). so smoothing
What is the reason for this difference?

1Revenger1, 2020-11-17:
VoodooRMI should come with VoodooInput already - it might be out of date though since it's from almost a month ago. I can update it for the release at some point.

I have no clue why the 8570 has issues though - VoodooInput for some reason has issues with some systems and I have no idea why.

ryulove3, 2020-11-17:
The trackpad has been working for two days, but Today, it stops working for no reason. Audio doesn't work either. I think it's probably because clover is unstable. On the 8570w, the RMI is not available due to a lot of misclicks and poping.

1Revenger1, 2020-11-17:
For the 850G1, I'd try IRQ fixes - especially if Audio isn't working.

Can you get some finger data from the 8570w? Some logs from within macOS should be fine. `log stream | grep -i vrmi > ~/Desktop/finger_data.txt` in terminal should work. You will need to run `log show --last 5m | grep -i vrmi` first though due to macOS bugs.

ryulove3, 2020-11-17:
850 G1 ( sound solved , nvram brokened due to opencore )
VoodooRMI
catalina10.15.7 -- working good. perfect! (clover 5122)
Big Sur (clover 5126) -- loaded but not working. sometimes working
For two days, I was in heaven.
Today.. Where I am?
-----------------
8570w, 
All gestures is working
(except Swipe left from the right edge with two fingers to show Notification Centre<<--working but not easy) 
 It is not bad Compared to 850g1(VoodooRMI), hp x360(VoodooI2C), or magic trackpad 2.

it is a small problem what not smooth scrolling and stops in the middle.
It is the most big problem ''''unintended one finger tab, two finger tab''''
 In the web browser, Finder, everywhere. 
It is rarely available.
-----------------------
I'll check finger data as quickly as possible.

ryulove3, 2020-11-18:
fingdata 

2020-11-18 12:17:35.089391+0900 0x5379     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:17:57.771868+0900 0x5935     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:17:58.058476+0900 0x594d     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:17:58.079350+0900 0x594f     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:17:59.916644+0900 0x59ef     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:01.571341+0900 0x5a4f     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:01.603012+0900 0x5a53     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:01.612865+0900 0x5a54     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:02.281383+0900 0x5a85     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:02.637862+0900 0x5a9a     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:03.336940+0900 0x5ad2     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:03.552640+0900 0x5ae7     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:03.758568+0900 0x5af9     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:06.033082+0900 0x5ba5     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:37.088132+0900 0x5eff     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:37.150288+0900 0x5f04     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:37.458529+0900 0x5f1d     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:37.540592+0900 0x5f25     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:37.754561+0900 0x5f3b     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:49.811047+0900 0x5f8b     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:55.298050+0900 0x6077     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:55.514255+0900 0x608b     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:55.699355+0900 0x609e     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:55.720158+0900 0x60a0     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:55.864712+0900 0x60ab     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:18:56.058956+0900 0x60b9     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6

ryulove3, 2020-11-18:
seewh0@seewh0s-8570w ~ % log show --last 5m | grep -i vrmi
2020-11-18 12:26:28.972145+0900 0xa8bd     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:26:29.536451+0900 0xa8cc     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:26:48.275611+0900 0xab37     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:27:18.861433+0900 0xb07c     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:28:26.339385+0900 0xbd1a     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:02.625393+0900 0xc352     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:02.646282+0900 0xc354     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:02.667331+0900 0xc358     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:02.852950+0900 0xc366     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:02.873679+0900 0xc368     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:02.997071+0900 0xc372     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:03.017871+0900 0xc374     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:03.038629+0900 0xc376     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:03.335048+0900 0xc38c     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:03.712745+0900 0xc3ac     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:05.124690+0900 0xc435     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:05.941379+0900 0xc487     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:06.166408+0900 0xc49f     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:06.685989+0900 0xc4d4     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:09.379336+0900 0xc57f     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:09.399972+0900 0xc581     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:09.409785+0900 0xc582     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:09.472289+0900 0xc589     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:09.553589+0900 0xc590     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:09.862764+0900 0xc5a8     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:09.980628+0900 0xc5b2     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:10.279330+0900 0xc5b8     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:10.311041+0900 0xc5bb     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:10.331863+0900 0xc5bd     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:10.352581+0900 0xc5bf     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:10.558089+0900 0xc5ce     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:10.578872+0900 0xc5d0     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:24.800226+0900 0xc81e     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:24.831883+0900 0xc821     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
[... cut ...]
2020-11-18 12:29:32.393823+0900 0xca8b     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:32.425479+0900 0xca8f     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:32.833232+0900 0xcab4     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6
2020-11-18 12:29:41.593434+0900 0xcb35     Default     0x0                  0      0    kernel: (kernel) VRMI - Error: Could not read F11 attention data: -6

1Revenger1, 2020-11-26:
Moved to #84

PR #72 (2020-10-04): Initial MacKernelSDK version
Closes #71
closed 2020-11-26 by 1Revenger1