VoodooSMBus/VoodooRMI #73: kernel panic when laptop goes to sleep

Jamesxxx1997, 2020-10-09:
**Describe the bug**
**when laptop goes to sleep , sometimes it will wake up automatically and becomes very hot , with screen doesn't light up , suscepted kernel panic**.
I am a thinkpad x1 yoga 2018 user , now using voodoormi to drive touchpad & voodooi2c+voodooi2chid to drive usb touchscreen , but finding out kernel panic when laptop goes to sleep , which doesn't seems to happen when I used voodoops2 to drive touchpad & voodooi2c+voodooi2chid to drive usb touchscreen ; here is [my error-reporting file](https://drive.google.com/file/d/1Z6oMQdDPtOgq-cP2UG6qLSH_H4qPg_eW/view?usp=sharing) in voodooi2c-documentation request format, thanks! 
Although I have add debug=0x100 and keepsysm=1 in clover boot argument , I don't (know) find out the kernel panic message ><

**To Reproduce**

**Expected behavior**
Sleep with normal behavior.

**Log**
the log is no recorded after the kernel panic , I will upload another log after boot when there is another kernel panic when laptop sleeping.
[log 2.txt](https://github.com/VoodooSMBus/VoodooRMI/files/5352977/log.2.txt)

1Revenger1, 2020-10-09:
When you reboot, you should be getting a pop up after logging in which states that the mac restarted for some reason. You should be able to click on Send/Show Report and it'll show the panic report. I think you can get the panic to in the folder where panics are usually saved.

zhen-zen, 2020-10-09:
cc @KirisameMarisaa Did you have similar issue before?

KirisameMarisaa, 2020-10-10:
> cc @KirisameMarisaa Did you have similar issue before?

No.

zhen-zen, 2020-10-10:
@KirisameMarisaa Thanks! Also feel free to join YogaSMC gitter chat.

@Jamesxxx1997 I think you still should cleanup your config or adapt to other's config. I have no idea how your DSDT is patched, some expected ACPI device even appear in a wrong location.

[image]

closed 2020-11-26 by 1Revenger1