VoodooSMBus/VoodooRMI #77: Implement F3A

1Revenger1, 2020-11-02:
F3A was [recently added to the Linux kernel](https://github.com/torvalds/linux/commit/9e4c596bfd004f447a652205163234dfd4aafa69#diff-2bf67870c583b060301909f1ea29f82f9d149ab77917b19d50ce2d269463a611) so we can actually implement it now

bobert5, 2020-11-05:
That's great news. I look forward to trying it out.

1Revenger1, 2020-11-16:
@bobert5 would you be willing to test it out? I don't have a trackpad which uses F3A so I am unable to test.

bobert5, 2020-11-21:
Sure, I'll give it a spin.  I haven't used that laptop (x1 yoga 4) much recently, so my hackintosh install is a bit outdated.  Should I update OC & kexts before testing, or does it not matter?

1Revenger1, 2020-11-21:
Doesn't matter much afaik. You can get a download through github actions if you want (green checkmark then little menu on the right iirc)

bobert5, 2020-11-21:
Same problems as before, unfortunately. Scrolling is difficult. Dragging even more so.  According to the BetterTouchTool live view, it seems to be better at rejecting my thumb, but it's not translating to any tangible improvement in clicking or dragging. A whole lot of unregistered thumb clicks.  
I'm a little rusty - what should I be looking for in the logs? I'm not seeing any errors or anything different than before.

1Revenger1, 2020-11-21:
uhh, can you just send me logs?
It's not registering physical clicks very consistently?

bobert5, 2020-11-21:
Took me a while to gather logs solely using the touchpad.
Yeah it misses at least half of the clicks.  Both thumb-clicks and one-finger-clicks.  Performance is basically the same as it was before.
[logs.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5577101/logs.zip)

1Revenger1, 2020-11-21:
When you say thumb and one finger clicks - what do you mean by that? It shouldn't matter what fingers you use for the buttons right? Or are they capacitive or something weird.

I think I see why dragging and scrolling is difficult - it looks like you lay your fingers down particularly flat which is triggering the palm rejection system I wrote. It allows a maximum delta of 2 between the WX and WY, which may be a bit more sensitive for trackpads which tend to report large WX/WY values (mine reports very small values). If your comfortable compiling, you could try increasing the value here or just removing that check completely: https://github.com/VoodooSMBus/VoodooRMI/blob/master/VoodooRMI/RMI_2D_Sensor.cpp#L169

bobert5, 2020-11-21:
Oh my mistake haha. I had downloaded it from here https://github.com/VoodooSMBus/VoodooRMI/runs/1410734861
I think I found the right one now.

bobert5, 2020-11-21:
> "I think I see why dragging and scrolling is difficult - it looks like you lay your fingers down particularly flat which is triggering the palm rejection system I wrote."

Interesting that my fingers should register as flat.  I never checked until now, but my hand is positioned in more of a raised arch/claw, so only the very tips of my fingers are on the touchpad.
Is the purpose of palm-rejection to stop the touchpad from triggering while you're typing?  Is that a common problem?  I can't say I've ever had that issue on any laptop I've ever used.  Particularly not on this laptop, where the trackpoint buttons already act as a barrier between the touchpad and keyboard.

Anyway, now two-finger right-clicks aren't registering and something's weird with dragging.  I can drag a window by it's header/title bar, but it only moves when I release my fingers, rather than continuously following the cursor as my fingers are pressed down.  Same with selecting text.  Also, I can't resize windows by dragging in the corners or edges. 

I don't know if it's evident in the logs, but overall something's not right with the way macOS is interpreting touchpad/cursor input on this laptop.  It just seems much jerkier, bogged-down, and inaccurate than it does in Linux. And it's not even in the same league as the magic trackpad 2 on my hack desktop.  I suspect the problems begin further upstream, either somewhere else in my config, or with voodooI2c/voodooinput, because like I said - the touchpad isn't great with voodooI2CHID either.  So take my input with a grain of salt.

[logs2.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5577212/logs2.zip)

PR #79 (2020-11-25): F3A
Closes #77
closed 2020-11-25 by 1Revenger1