VoodooSMBus/VoodooRMI #78: RMISMBus.kext - Invalid parameter

Litarvan, 2020-11-11:
**Describe the bug**

When only enabling VoodooPS2 (Controller, Mouse, Keyboard, Trackpad) I had my trackpad working perfectly except the buttons which were doing nothing (nor the touch click).

I saw that I had a SMBus Synaptic device, so I installed VoodooRMI + VoodooSMBus + RMISMBus, and disabled VoodooPS2 Mouse and Trackpad.

But I can't get my trackpad to work at all anymore, and when I look at OpenCore logs, I see that it can't load RMISMBus.kext but doesn't say why ('Invalid parameter', seems like a generic error).

Since I'm trying to boot Catalina Recovery, and does not have a physical mouse, I do not have any system logs, but I attached my zipped EFI and the OpenCore logs.

[efi.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5521592/efi.zip)
[opencore-2020-11-11-033018.txt](https://github.com/VoodooSMBus/VoodooRMI/files/5521593/opencore-2020-11-11-033018.txt)

1Revenger1, 2020-11-11:
Use VoodooSMBus from the VoodooRMI release ZIP - the release zip of VoodooSMBus from their repo is old and doesn't account for changes in the `dev` branch over there. Sorry for the confusion

Litarvan, 2020-11-11:
Works perfectly, thanks a lot !

closed 2020-11-11 by Litarvan

nettoyoussef, 2021-04-07:
I have the exact same issue described above. However, I am using all the kext files from the zip release (particularly the debug version) from this repo. 

What could be causing this trouble?

```
147:676 00:656 OC: Prelinked injection VoodooRMI.kext\Contents\PlugIns\RMISMBus.kext () - Invalid Parameter
```
Below I attached my complete log file and my config.plist.
Any help would be greatly appreciated.

[config.plist.zip](https://github.com/VoodooSMBus/VoodooRMI/files/6274112/config.plist.zip)

[opencore-2021-04-07-150251.txt](https://github.com/VoodooSMBus/VoodooRMI/files/6274094/opencore-2021-04-07-150251.txt)