VoodooSMBus/VoodooRMI #83: Devices to not respond to reading PDT Tables

1Revenger1, 2020-11-26:
Some devices do not respond when trying to read the PDT tables, producing an output similar to:
```
VRMI - Info: SMBus version 2
VRMI - Debug: Config TrackpointMultiplier loaded: a -> a
VRMI - Debug: Config TrackpointScrollMultiplierX loaded: a -> 1e
VRMI - Debug: Config TrackpointScrollMultiplierY loaded: a -> 1e
VRMI - Debug: Config TrackpointDeadzone loaded: 1 -> 1
VRMI - Debug: Config DisableWhileTypingTimeout loaded: 1f4 -> 64
VRMI - Debug: Config DisableWhileTrackpointTimeout loaded: 1f4 -> 64
VRMI - Debug: Config ForceTouchMinPressure loaded: 50 -> 5a
VRMI - Debug: Config ForceTouchEmulation loaded: 1 -> 1
VRMI - Debug: Config MinYDiffThumbDetection loaded: c8 -> c8
VRMI - Debug: Updating Configuration
VRMI - Info: RMI Bus (DEBUG) Starting up!
VRMI - Error: Read PDT entry at 0x00e9 failed, code: -6
VRMI - Error: RMI initial reset failed! Continuing in spite of this
VRMI - Error: Could not read PDT properties from 0x00ef (code -6). Assuming 0x00.
VRMI - Debug: rmi_probe_interrupts: Counting IRQs
VRMI - Error: Read PDT entry at 0x00e9 failed, code: -6
VRMI - Error: IRQ counting failed with code -6
VRMI - Error: Could not probe
VRMI - Error: Could not probe
VRMI - Debug: Freeing function list
```

They respond to the version being read.

I haven't seen why this occurs - it generally occurs on older devices (Ivy Bridge), but also seen it on devices as late as Broadwell.

Related Issues:
* https://github.com/VoodooSMBus/VoodooRMI/issues/33

I've tried rigging VoodooPS2Trackpad to reset the trackpad then notify VoodooRMI (where VRMI will then load) but a similar output was gotten.

gimdh, 2020-12-01:
[image]

It might be an one of those requires 'product-specific documentation'. Dump of entire register should help if possible.

1Revenger1, 2020-12-01:
Well, the PDT is in the register - so you should be able to read it. The issue is that the trackpad isn't responding *at all* (-6 is no response iirc).
That said - the trackpad iirc does respond correctly when loading the driver from within macOS. It only has issues when being injected. Will need to double check chat history and other issues though.

martinlhzy, 2020-12-08:
Hello!

I had the same problem with my HP ZBook 14 G2 (a rebranded EliteBook 840 G2). The RMI driver only loaded when it was injected from the macOS or the OpenCore was booted from a pen drive.

I believed it was because the boot order is different on a pen drive, but that wasn’t the case. In fact every time when I opened the diagnostic menu on my laptop, and then booted to OpenCore the driver loaded flawlessly.

The solution is to turn off the fast boot and set it to UEFI Native (not UEFI with CSM). On HP machines it isn’t evident, because if you turn off the CSM weird graphics glitches will occurs until you either turn off and turn on your display or set the resolution in OpenCore to 480p.

I think it’s because without these two the system stand up too quickly and the trackpad doesn’t wake up in time.

1Revenger1, 2020-12-08:
I wonder if there is some additional setup done over PS2 that we might be missing if fast boot being disabled+no CSM is being used. I've tried hardcoding a 10 second delay before trying to bring up the trackpad.

1Revenger1, 2020-12-18:
[Reset_Before_init.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5714154/Reset_Before_init.zip)
For those having this issue, do you mind trying the above version of VoodooPS2/VoodooRMI/VoodooSMBus? These include some small fixes - but more importantly uses VoodooPS2Trackpad to reset the trackpad before initializing VoodooRMI.

Make sure that VoodooPS2Trackpad is enabled in your config.plist if using OpenCore.

This is not meant to be used super long term, but if this does work I will make a more permanent solution that may or may not involve a PR into VoodooPS2 (unless I write a custom PS2 implementation myself)

martinlhzy, 2020-12-20:
Sorry for the late answer. I tested it and now it works flawlessly, even with Fast Boot and CSM enabled. Even the trackpointer worked for the first start what I've never seen before, even if it disappeared later. I tried it multiple times and it constantly loads up correctly.

LocTran016, 2020-12-22:
For me, VoodooRMI and VoodooSMBus doesn't load and there is no log. I attached the `kextstat command` log and opencore log
[opencore-2020-12-22-042637.txt](https://github.com/VoodooSMBus/VoodooRMI/files/5727824/opencore-2020-12-22-042637.txt)
[log.txt](https://github.com/VoodooSMBus/VoodooRMI/files/5727823/log.txt)

[image]

1Revenger1, 2020-12-22:
Need to fix VoodooSMBus attaching first, do you have a log with VSMB as the search term?
@LocTran016

LocTran016, 2020-12-22:
Oh, sorry to tell you that I'm using MacOS 10.14.6

1Revenger1, 2020-12-22:
@LocTran016 do you mind trying with this version of VoodooSMBus? (Keep everything else the same)
[VoodooSMBus.kext 2.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5727981/VoodooSMBus.kext.2.zip)

LocTran016, 2020-12-22:
Doesn't help. My OpenCore log is here:
[opencore-2020-12-22-061929.txt](https://github.com/VoodooSMBus/VoodooRMI/files/5728184/opencore-2020-12-22-061929.txt)

1Revenger1, 2020-12-22:
I had not tested that version, sorry about that - there turned out to be some linking errors. Tested it in Mojave 10.14.6 as well
[Uploading VoodooSMBus.kext 2.zip…]()

LocTran016, 2020-12-22:
> I had not tested that version, sorry about that - there turned out to be some linking errors. Tested it in Mojave 10.14.6 as well
> 
> [Uploading VoodooSMBus.kext 2.zip…]()
> 
> 

Are you uploading sth because the link seem to be died.

1Revenger1, 2020-12-22:
[VoodooSMBus.kext.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5729163/VoodooSMBus.kext.zip)
I may have been a bit quick on the comment button, lol

LocTran016, 2020-12-22:
For the first time I boot, it work (including 4 fingers gesture). I haven't tried the second time. Here is my log
[log.txt](https://github.com/VoodooSMBus/VoodooRMI/files/5729253/log.txt)

LocTran016, 2020-12-22:
Until thirst time, the trackpad still working (4 fingers). Only the physical mouses don't work (those at the bottom of the trackpad, don't know how to describe it, sorry).
[log.txt](https://github.com/VoodooSMBus/VoodooRMI/files/5730415/log.txt)

1Revenger1, 2020-12-22:
This should be a more final version which is much less hacked together. This version is a fully enabled version of VoodooPS2Trackpad, which gets shutdown by VoodooRMI when it starts running. This may lengthen boot times, though IntelBluetoothFirmware was still way longer for me :(

This version also should start without waiting for VoodooPS2Trackpad if it isn't injected, in which case boot times shouldn't be lengthened in this case.

@LocTran016 I also fixed VoodooTrackpoint

When using this, test with VoodooPS2Trackpad should be Enabled, same as with previous tests.

[VoodooRMI_PS2_Reset.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5732463/VoodooRMI_PS2_Reset.zip)

LocTran016, 2020-12-23:
This is so awesome!!! The mouse work very well (including trackpad and physical buttons at the bottom of trackpad)
[log.txt](https://github.com/VoodooSMBus/VoodooRMI/files/5733157/log.txt)

1Revenger1, 2020-12-23:
@Rohan200220 you able to confirm that the last zip works for you?

closed 2020-12-24 by 1Revenger1

Rohan200220, 2020-12-24:
Hey, sorry for the late response, I tried the zip yesterday, but it’s working as it was last time. The mouse pointer is barely usable, I tried it both with voodoops2mouse enabled and disabled.

1Revenger1, 2020-12-24:
oh, sounds like #84 then? Sorry to bug you, thanks for verifying it works at least

PR #90 (2020-12-24): Wait for VoodooPS2 if present
Closes #83