VoodooSMBus/VoodooRMI #84: Delayed Packets

1Revenger1, 2020-11-26:
While using the trackpad, some people have come across an issue where the trackpad will randomly stop working for a second or two, before macOS acts upon whatever actions occurred during that time. This only occurs on some devices.

The only time this has happened to me is when using ACPIBatteryManager.kext, and I rarely get it after sleeping under Catalina. This may be related to some scheduling issues and VoodooInput, or other weird interaction between VoodooInput and VoodooRMI? I've seen people with these issues have similar issues with Acidanthera's VoodooPS2 while are fine with Rehabman's VoodooPS2. I have no clue why this would happen though - and don't have a good way to figure out why this happens.

Related Issues:
* #71
* #33

gimdh, 2020-12-01:
I think you mean [this](https://github.com/VoodooSMBus/VoodooRMI/issues/12#issuecomment-658084702).

Basically entire ACPI system halts whenever ACPIBatterManager polls battery information.

1Revenger1, 2020-12-01:
I've seen it on laptops not using ACPIBatteryManager as well - I'm not entirely sure what it could be. The trackpad on my device feels like my MT2 in terms of smoothness/mapping/etc.

gimdh, 2020-12-01:
I think SMCBatteryManager might also not be able to avoid spin wait. VoodooI2C suffers similar issue AFAIK. Have you ever seen a laptop with this issue even without any battery manager kext?

FreeJHack, 2020-12-28:
One or two seconds is a long time, however, I had a similar issue and I found the fix.
It was related to ACPIBatteryManager kext and some newer ACPI.
Take a look at [this](https://github.com/VoodooI2C/VoodooI2C/issues/179), maybe can help...