VoodooSMBus/VoodooRMI #85: "No trackpad found" despite everything working

mbirth, 2020-11-26:
**Describe the bug**
I've made my trackpad (Synaptics SynPS/2) in my hp Spectre x360 13-4104ng work using VoodooRMI via SMBus. However, the macOS Settings still show "No trackpad found".

**To Reproduce**
Steps to reproduce the behavior:
1. Setup VoodooRMI
2. Go to macOS System Preferences --> Trackpad
3. See "No trackpad found" message

**Expected behavior**
Trackpad settings

**Log**
Since it seems to work, what logs do you need exactly?

My hardware: https://linux-hardware.org/index.php?probe=d258888474

My config: https://github.com/mbirth/hp-spectre-x360-4104ng-hackintosh

1Revenger1, 2020-11-26:
Have you patches your battery yet? Working battery is needed for the trackpad to appear in system preferences as stupid as it sounds

mbirth, 2020-11-26:
Okay, that explains it. Didn't find the willpower to patch battery yet. 😉

Thanks a lot.

closed 2020-11-26 by mbirth