VoodooSMBus/VoodooRMI #86: I had a similar problem like #83 and #33 but I'm on Broadwell

microtruong, 2020-11-30:
[image]

microtruong, 2020-11-30:
I'm on Clover 5122 on Catalina 10.15.7. I have the latest VoodooPS2, VoodooRMI, and VoodooSMbus kexts installed and yet the trackpad doesn't work.

ryulove3, 2020-12-01:
My test result (hp 850 g1)
on catalina 10.15.7
1. clover 5119 --> no problems
2. opencore 0.6.3 --> no problems

on big sur 11.01
1. clover 5126 or 5127 --> attached 'voodooSMBUS' but cant load 'voodooRMI'
2. opencore 0.6.3.     ---> no problems
-----------------------
I think that It's problems on clover 5122& over
If on catalina, use 'clover 5119& below'
if on big sur, use 'opencore'

1Revenger1, 2020-12-21:
This is basically #83 
Can you please post your success using [these versions of kexts](https://github.com/VoodooSMBus/VoodooRMI/issues/83#issuecomment-747912493) within that issue?

closed 2020-12-21 by 1Revenger1