VoodooSMBus/VoodooRMI #87: Could Trackpoint be optimized like Windows 10?

lucas-www, 2020-12-01:
First of all, Thank you very much for the excellent project. 
When I used Trackpoint in Windows 10 OS, I found that the Trackpoint is so smooth than any other OS distribution.
So if we can optimize it in OS X, thanks again.

gimdh, 2020-12-01:
I guess that's due to better acceleration curve in Windows. However, I am not entirely sure if we can change acceleration curve in per device manner in MacOS if at all.
Custom curve could be applied by VoodooRMI to compensate MacOS's default curve. However, finding best curve seems to require extensive research.

1Revenger1, 2020-12-01:
I think packets get sent less often for the mouse? Not entirely sure - I've noticed a slight bit of lag on my side as well but don't know of any ways to fix it. It is basically like a PS2 mouse so if we could update the reporting rate that might help?

PaulEau73, 2022-08-05:
> I think packets get sent less often for the mouse? Not entirely sure - I've noticed a slight bit of lag on my side as well but don't know of any ways to fix it. It is basically like a PS2 mouse so if we could update the reporting rate that might help?

I don't know about that. But I was trying VoodooPS2Controller and I also noticed that the trackpoint was not smooth, unless I press a physical trackpad button while using it. By doing that it's perfectly smooth. I think the interruptions must be more numerous when pressing the physical buttons ?