VoodooSMBus/VoodooRMI #89: Click to sensitive on Big Sur Lenovo T440

jarno83, 2020-12-08:
Hi, I have a problem on my T440. Touchpad works well with VoodooRMI but the click is to sensitive. The problem is that if I move the cursor across the display I accidentally click on items and so on.
If I set it to Firm in the settings it's still the same. 
Can I change some settings for that in plist? 

Thank you.
[image]

1Revenger1, 2020-12-21:
I don't believe that is anything I can fix with VoodooRMI, unless some touch data is getting reported wrong. Please provide a snippet of touch data from the trackpad specifically highlighting this issue in the form of a log file.

jarno83, 2020-12-21:
Hi, thanks for reply.
This is what I got from logs from catalina install :

VRMI - Info: SMBus version 2
VRMI - Info: RMI Bus (RELEASE) Starting up!
VRMI - Info: Initializing F34.
VRMI - Info: F34 not implemented
VRMI - Info: Initializing F01.
VRMI - Info: Found RMI4 device, manufacturer: Synaptics, product: TM3053-004, fw id: 1741117
VRMI - Info: Initializing F03.
VRMI - Info: Initializing F11.
VRMI - Info: Initializing F54.
VRMI - Info: F54 not implemented
VRMI - Info: Initializing F30.
VRMI - Info: Finish PS2 init

Thanks.

1Revenger1, 2020-12-22:
This needs data from VoodooInput as well unfortunately, can you use `Voodoo` rather than `VRMI` in the command?

jarno83, 2020-12-24:
Using Voodoo won't give anything. I'm using Sniki's EFi for T440S - https://github.com/Sniki/Lenovo-Thinkpad-T440S
Thanks

1Revenger1, 2020-12-25:
Hrmm, can you just send the entire dmesg log? It might be that VoodooInput doesn't say Voodoo in it's logs (which is kind of annoying tbh)

jarno83, 2020-12-26:
Hi, I hope this helps - https://pastebin.com/GKQ8hHWG
Thank you :)

1Revenger1, 2020-12-26:
Hrmm, are you using the debug version of VoodooRMI? Not seeing the logs.

jarno83, 2020-12-26:
Hi, no worries. I changed VoodooRMI and VoodooSMBus kext to latest debug versions. Strange is that with debug version touchpad is not working but trackpoint does. Here is the log - https://pastebin.com/PRww1QAH

Thanks again :)

1Revenger1, 2020-12-26:
Weird, it appears VoodooInput isn't being started. Do you mind checking that it's loaded with `kextstat | grep -i Voodoo` in terminal? May try moving VoodooInput around in your config.plist as well

Could try this latest version of VoodooRMI: https://github.com/VoodooSMBus/VoodooRMI/runs/1607290389 (There is an artifact on the right hand side which you should download). If neither of those work, you could try pairing VoodooInput from the release version of VoodooRMI (under VoodooRMI.kext/Contents/PlugIns) with the debug version of VoodooRMI and seeing if that'll work

ryulove3, 2020-12-27:
https://user-images.githubusercontent.com/18746664/103166242-7cf00900-4863-11eb-8095-e212aaef6b9f.mov

HP 8570W (3th Ivybridge) Catalina 10.15.7
Clover 5118 & Opencore 0.6.4 +VoodooRMI 1.2& 1.3

accidentally click on items <-- accidentally tab to click(one finger) and stop scrolling (two finger).

may find out by watching the video. Finger data disappears instantly.
------
Lenovo x250 (5th broadwell), hp 850G1 (4th Haswell) 
Big Sur 11.1 & Opencore 0.6.4 +VoodooRMI 1.2& 1.3---> No problems

jarno83, 2020-12-27:
I'm on catalina right now, will update soon. 
With the 1.3 debug, the os is not booting and will restart after some minutes of being hanged.

ryulove3, 2020-12-27:
> I think same happens to me.
If with RMI 1.3 & Opencore,   
There is no 'VoodooTrackpoint.kext'.
check config.plist. 'Enabled --> No' or Delete line (VoodooTrackpoint.kext).

ryulove3, 2020-12-27:
> I'm on catalina right now, will update soon.
> With the 1.3 debug and release, the os is not booting and will restart after some minutes of being hanged.
> 
> > Screen.Recording.2020-12-27.at.4.41.55.PM.mov
> > HP 8570W (3th Ivybridge) Catalina 10.15.7
> > Clover 5118 & Opencore 0.6.4 +VoodooRMI 1.2& 1.3
> > accidentally click on items <-- accidentally tab to click(one finger) and stop scrolling (two finger).
> > ## may find out by watching the video. Finger data disappears instantly.
> > Lenovo x250 (5th broadwell), hp 850G1 (4th Haswell)
> > Big Sur 11.1 & Opencore 0.6.4 +VoodooRMI 1.2& 1.3---> No problems
> 
> I think same happens to me.

If with RMI 1.3 & Opencore,   
There is no 'VoodooTrackpoint.kext'.
check config.plist. 'Enabled --> No' or Delete line (VoodooTrackpoint.kext).

ryulove3, 2020-12-27:
If with RMI 1.3 & Opencore,   
There is no 'VoodooTrackpoint.kext'.
check config.plist. 'Enabled --> No' or Delete line (VoodooTrackpoint.kext).

ryulove3, 2020-12-27:
Only one of the multi fingers disappears. One of four, one of three, one of two. If Only one finger, it stops responding very very very shortly and then recognizes it again.

shaizu, 2020-12-27:
> accidentally click on items <-- accidentally tab to click(one finger) and stop scrolling (two finger).

same thing happens to me even tried to debug but VoodooInput.kext doesn't load
contacted @1Revenger1 in discord for the same issue on discord month ago

ryulove3, 2021-01-05:
1.3 is really better compared to 1.2. Miss clicks have been noticeably reduced on the hp 8570w, and scrolling is also possible (of course it's not that smooth yet). On the hp 850 g1, lenovo x250, there was a problem with cursor popping when using multi-gesture (ex. 3finger moving), but it is improved. so impressive!!!
(8570w - catalina + opencore 0.6.4
850 g1, x250 - big sur + opencore 0.6.4)

jarno83, 2021-01-09:
Hi again. I'm now using bigsur and 1.3 works a lot better than 1.2 on catalina. I'm happy with that user experience :)

singhalrishi27, 2021-01-10:
> disappears and stops responding.

I have also same problem and was fine with VoodooRMI 1.2

jarno83, 2021-01-12:
It looks like the problem mainly appears after wake from sleep for some minutes when CPU is working hard (i5 4300u). Maybe with faster systems it's not so noticeable.

singhalrishi27, 2021-01-12:
> It looks like the problem mainly appears after wake from sleep for some minutes when CPU is working hard (i5 4300u). Maybe with faster systems it's not so noticeable.

anyways its a HP issue I guess @1Revenger1 Thinks that it is trackpad issue 
I don't think because v1.2 detects gestures perfectly as it should....
what do you think

1Revenger1, 2021-01-12:
> I don't think because v1.2 detects gestures perfectly as it should....

That's weird, the only code that changed how packets are recieved are changes to palm rejection. You could check logging data to see if fingers are being invalidated but last I checked, it looked fine. If it's not that, I have no clue why it's having issues on version 1.3.

singhalrishi27, 2021-01-13:
> That's weird, the only code that changed how packets are recieved are changes to palm rejection. You could check logging data to see if fingers are being invalidated but last I checked, it looked [...]

I saw that when using 4 finger 1 finger gets lost when 2 or 3 fingers same issue happens again

Sniki, 2021-03-28:
This may have been already fixed on Master recently.

closed 2021-09-11 by 1Revenger1