VoodooSMBus/VoodooRMI #92: VoodooRMI loading and finger detection

singhalrishi27, 2021-01-03:
It is a big story so let's start
Folio 9470m 
Ivy Bridge i3
HD 4000

I install voodoo SMBUS And RMI properly and rebooted I saw that trackpad was not working so i copied same EFI to my USB and Then Rebooted using USB surprisingly trackpad working and gestures were working...

I don't know why it didn't worked before so I tried choosing EFI file manually in BIOS from system disk then booted from EFI and Trackpad was working then rebooted(it automatically selected EFI) again no trackpad then I got to know that kext was only sticking when I select EFI file manually ...

new commits were added and I updated my kexts with latest commits and then that issue was fixed and another popped this kext is crashing every second one finger was working fine but when I used scrolling it was crashed like it stops detecting my finger at all even when I was scrolling and opening launchpad was a pain please fix these issues ASAP

Also my trackpoint and force touch is not working but I don't care

1Revenger1, 2021-01-04:
[VoodooRMI-1.3-Debug.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5763547/VoodooRMI-1.3-Debug.zip)
Do you mind trying this? This is the debug version of 1.3. If it doesn't work very well, can you please provide logs? 
[VoodooPS2Controller.kext.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5763559/VoodooPS2Controller.kext.zip)
Use the above version of VoodooPS2Controller as well, and *enable injecting* VoodooPS2Trackpad. This should make it work whether you boot from USB or not.

For getting logs, use `sudo dmesg > ~/Desktop/RMI.txt` after moving around the trackpad a little and having it stop working.

With 1.3, disable injecting VoodooTrackpoint as well.

singhalrishi27, 2021-01-04:
@1Revenger1 
Thanks for Response 
so, what I did is removed RMII2C From VoodooRMI../Plugins and Removed everything expect keyboard kext from Voodoops2.kext 
Logs 
[RMI.txt](https://github.com/VoodooSMBus/VoodooRMI/files/5763622/RMI.txt)

there was no problems while attaching kext while finger problem still existed..
here is a short video which shows my problem..
https://www.icloud.com/iclouddrive/0hPQ5yfphn0gc-W5qzldzifCw#PXL_20210104_081252048

notice how 2 finger gestures stops working and its a lot harder to open launchpad and close it 

I have increased cursor size to show you properly

1Revenger1, 2021-01-04:
Are you using VoodooInput from VoodooRMI/Content/PlugIns/ ?
I forgot to include this last time, but can you please add `msgbuf=1048576` to your boot arguments? It'll allow more log data to appear when running `sudo dmesg`.

I noticed that some of the packets were showing no fingers which was a bit odd - hopefully a bigger log shows a bit more information.

singhalrishi27, 2021-01-04:
yes I was using VoodooInput from VoodooRMI/Content/PlugIns

Here is the new log....
[RMI.txt](https://github.com/VoodooSMBus/VoodooRMI/files/5763750/RMI.txt) 

force touch still doesn't works

1Revenger1, 2021-01-04:
Force touch is not expected to work on trackpads that don't click down. I've only programmed force touch emulation to work with trackpads that  have the whole surface click down as it seems to work best there. Could try to add a version which works on trackpads which don't move at all, but I don't have one to test one.

For 4 finger gestures, it seems to be grabbing the thumb somewhat incorrectly. If the fingers are close together, it will use finger size/pressure (on the trackpad) to guess which finger is the thumb. If you want to make it just based off of which thumb is lowest (used when there is an obvious low finger), you could try turning down MinYDiffThumbDetection under VoodooRMI.kext/Contents/Info.plist. This value is the minimum difference in y values of the two lowest fingers on the trackpad in which it'll set the "lowest" finger as the thumb vs using pressure. (200 means that a difference of less than 200 will use pressure rather than location to find the thumb. 0 means that it'll always use the "lowest" finger as the thumb). Another option would be to make sure that your thumb is flatter on the trackpad when using the "Spread" gesture.

For 2 finger gestures, it seems like it may possibly be seeing only 1 finger at a time? This would be an issue with the trackpad itself, though you could try moving your fingers a part slightly to make it more obvious that there are two fingers rather than one. This really isn't something that I can fix, as the trackpad itself only reports 1 finger - and I can't change how it detects different fingers. When it does detect 2 fingers, they are both valid and seem to behave fine. Could see if scrolling works any better with VoodooPS2 maybe to see if it can detect the two fingers better.

For trackpoint, I'm not really sure to be honest. If you open ioregistryexplorer and look under `RMIBus`, does it show F03? If not, it may be that the trackpoint works over PS2 and doesn't have any passthrough over SMBus. I don't have any device like this to work with, though adding VoodooPS2Mouse *may* fix it (or completely break the trackpad).

singhalrishi27, 2021-01-04:
using 1.2 release it doesn't create any of these problems 
[image]

I don't care about trackpoint 
everything just worked fine before

singhalrishi27, 2021-01-05:
Hello @1Revenger1 

I rolled back to voodoormi 1.2 and older version of voodoo ps2 now every key works fine expect the loading issue also I gave this kext to my friend he was facing same finger issue and he also downgraded but he never had kext stick issue he was on Haskell maybe that's why I don't know here is a one more log for you
[RMI.txt](https://github.com/VoodooSMBus/VoodooRMI/files/5768719/RMI.txt)

singhalrishi27, 2021-01-10:
Hello @1Revenger1 

I rolled back to VoodooRMI 1.2 now every key works fine expect the loading issue also I gave this kext (VoodooRMI 3.0) to my friend he was facing same finger issue and he also downgraded but he never had kext stick issue he was on Haswell maybe that's why I don't know here is a one more log for you..catch we both had HP laptops and macOS Big Sur
[RMI.txt](https://github.com/VoodooSMBus/VoodooRMI/files/5768719/RMI.txt)

singhalrishi27, 2021-01-10:
Hello @1Revenger1 

I rolled back to VoodooRMI 1.2 now every key works fine expect the loading issue also I gave this kext (VoodooRMI 1.3.0) to my friend he was facing same finger issue and he also downgraded but he never had kext stick issue, he was on Haswell maybe that's why I don't know here is a one more log for you..catch we both had HP laptops and macOS Big Sur
[RMI.txt](https://github.com/VoodooSMBus/VoodooRMI/files/5768719/RMI.txt)

1Revenger1, 2021-02-04:
Someone else had an issue similar to this as well - what happens if you use 1.3.1? It should have VoodooSMBus downgraded to the release from 1.2. If 1.3.1 isn't working well, do you mind using VoodooSMBus from 1.2 and see if that behaves better? Leave VoodooRMI itself at 1.3.1

singhalrishi27, 2021-02-05:
> Someone else had an issue similar to this as well - what happens if you use 1.3.1? It should have VoodooSMBus downgraded to the release from 1.2. If 1.3.1 isn't working well, do you mind using VoodooSMBus from 1.2 and see if that behaves better? Leave VoodooRMI itself at 1.3.1

I've already did it 
it is better but still not good as voodooRMI 1.2 so I'm gonna stick to it

closed 2021-02-06 by singhalrishi27