VoodooSMBus/VoodooRMI #95: Buttons do not work on F3A

Someone52, 2021-01-17:
https://github.com/VoodooSMBus/VoodooRMI/blob/master/VoodooRMI/Functions/F3A.cpp

Seems like the bool F3A::start(IOService *provider) 
is missing a voodooTrackpointInstance = rmiBus->getVoodooInput() on the latest commit

Someone52, 2021-01-17:
Also there seems to be an issue here:

key_down = !(BIT(i) & data_regs[0]); // should be data_regs[i] instead??
key_code = gpioled_key_map[i];

which makes all 3 of my buttons register as left clicks, instead of left middle right

1Revenger1, 2021-01-18:
Thanks for catching the mistake in the first post - does seem that it is missing.

Do the buttons above the trackpad work? If so (that's generally done through F03 w/ PS2 passthrough), then that's normal. The entire clickpad is reported as one button, and it's up to the OS to figure out how to interpret it. You can set the bottom right to be right click if you want through the trackpad preferences

1Revenger1, 2021-01-18:
It's a slight bit misleading - button count is really gpio count which the trackpad says it has hooked up - it may be that the trackpad button is hooked up to pin 2 rather than pin 0. It's not guaranteed that every gpio slot is actually present. The best way to tell would be to get a log from VoodooRMI starting up in debug mode, as it'll announce whatever buttons it does find (readme should say how to get early boot logs using `sudo dmesg`).

In regards to emulating the behaviour of right and middle click buttons on the bottom of the trackpad - I have no clue how the synaptics driver decides to do that or not, and I'd rather not change the behaviour for everyone. I could put it behind an option in the plist I guess, though I guess I'd need to find a better way to do options. Any reason why setting "secondary click" in the trackpad preference to be in the bottom right of the trackpad isn't enough?

1Revenger1, 2021-02-04:
Closing this since F3A is working now
Can reopen if you want to add settings to try and emulate right/left click on the bottom but this isn't a priority.

closed 2021-02-04 by 1Revenger1