VoodooSMBus/VoodooRMI #97: [Lenovo T440] TouchPad being stuck after clicking

vaczi001, 2021-01-21:
Lenovo T440's original touchpad (not the T450) get stuck after clicking. I can move the cursor, but can't do anything even with an external mouse. Gestures are working in this time, but can't click anywhere. It's stuck in left click, and I have to reboot.
It's working if I just tap the touchpad.

**Steps to reproduce the behavior:**
1. Use Lenovo T440's original touchpad (not a T450 replaced one)
2. Click on touchpad (not tap)
3. Stuck on "left click mode"

**Expected behavior**
Like a normal mouse click, release after clicking. 

**Log**
Attaching verbose log file with DEBUG versions of VoodooPS2 and VoodooRMI. All of them are up-to-date (PS2: 2.2.0, RMI: 1.3)
[bootlog.log](https://github.com/VoodooSMBus/VoodooRMI/files/5851191/bootlog.log)

1Revenger1, 2021-01-23:
Hrmm, for some reason there is a ton of <compose failure [UUID]>. The log isn't useful unfortunately. I do notice a little weirdness with failing to write some stuff, though I'm not sure if that's related.

Are you able to get more logs? May have better luck using `sudo dmesg`, though you may need to use the `msgbuf=1048576` boot arg to expand it's buffer.

Are you injecting VoodooPS2Trackpad? You may have better luck injecting it if you aren't already.

vaczi001, 2021-01-23:
You're absolutely right, sorry. here is the right dmesg log.

Yes, I did what is suggested at release page:
Disabling VoodooPS2/Plugins/VoodooPS2Mouse, VoodooInput
Disabling VoodooRMI/Plugins/I2C
VoodooPS2/Plugins/VoodooPS2Trackpad is injected

[log.txt](https://github.com/VoodooSMBus/VoodooRMI/files/5859983/log.txt)

Here is a video about what exactly happening, for better explanation:
https://streamable.com/vuri4y

itsmevjnk, 2021-02-04:
I'm having the exact same problem with my X240.

1Revenger1, 2021-02-04:
That seems to be behaving really weirdly - can you both please send an ioreg with ioregistryexplorer? It seems to think that there are multiple buttons, which are all down all the time.

kushwavez, 2021-02-05:
@1Revenger1 Of course, here you are:
New log: [rmi_log.log](https://github.com/VoodooSMBus/VoodooRMI/files/5931793/rmi_log.log)
IOReg: [t440_ioreg.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5931798/t440_ioreg.zip)

1Revenger1, 2021-02-05:
Can you send a log from where you press the trackpoint buttons in the top left/right of the trackpad? It does appears that the buttons pull high when pressed rather than pull low found on most trackpads which I think it causing issues.

kushwavez, 2021-02-07:
Yes, here, top-left click
Video: https://streamable.com/jhsjey
Log:  
[logfile.txt](https://github.com/VoodooSMBus/VoodooRMI/files/5939037/logfile.txt)

nlabrad, 2021-02-26:
Having the same problem with an X1 Yoga. 
I can't even use a USB mouse to click. The only thing that seems to be working is clicking with the center button of the trackpad, but it's behaving oddly. I suspect this may be related with the way we make the three buttons work on Thinkpads in the first place, but it's just speculation.

nlabrad, 2021-02-27:
Update:

Everything works well unless I click on the touchpad buttons (the ones embedded the touchpad, not the ones between the keyboard and the pad).

If I click on the left touchpad button, I can only proceed to click with the middle three-button thing, and even that works so weird that it's hard to describe.

The SSDTs I have are SSDT-ALS0, SSDT-PNLF, SSDT-PTSWAK, SSDT-Sleep-PRW, SSDT-UIAC-AL, SSDT-XCPM. Not sure if any of those are in conflict, but I may boot from a Clover without any of these and see what happens.

1Revenger1, 2021-03-07:
https://files.gitter.im/5e938b02d73408ce4fe02f4f/1Hq7/VoodooRMI-1.3.2-DEBUG.zip
Do you guys mind trying the above version?
This requires having Acidanthera's VoodooPS2, on either the 2021 February or March release

kushwavez, 2021-03-08:
Okay, here you go. Same thing happening. Clicking on the pad, the cursor stuck on left-click no matter what (no matter where I press it).
[vrmilog_1.3.2.log](https://github.com/VoodooSMBus/VoodooRMI/files/6100455/vrmilog_1.3.2.log)
Also here a closer look of the T440 original trackpad. 

[image]

I don't know if it will help or not, but it doesn't have three physical buttons like the T450 or X1 does, I don't really know how it works so correct me if I'm wrong, but I think on Windows it's deciding whether it's left click or right click based on where you press it (?), top-left, or top-right. 
Unfortunately I don't know detailed informations about the trackpad, if you tell me how, I'll collect more info about it.

I don't know if it's related to the stuck thing or not.

1Revenger1, 2021-03-08:
@kushwavez Are you using VoodooPS2 from Acidanthera and have VoodooPS2Trackpad enabled? It's not reporting GPIO data that comes from VoodooPS2. It's a really odd interaction but there is some data I can only get from the PS2 side.

kushwavez, 2021-03-08:
Oh sorry, yes I didn't use VoodooPS2Trackpad. Now it's there.
Here is the new log.
[vrmi_1.3.2.log](https://github.com/VoodooSMBus/VoodooRMI/files/6100948/vrmi_1.3.2.log)

1Revenger1, 2021-03-08:
You're log confirms my suspicion at least, that the PS2 side reports the correct data. Should be fixed with the below kext:
https://files.gitter.im/5e938b02d73408ce4fe02f4f/IibT/VoodooRMI-1.3.2-DEBUG.zip

kushwavez, 2021-03-08:
Fixed.
It's working with that, all good. Finally the touchpad is ACTUALLY good now. The movements are much better compared to Windows, and VoodooPS2 too. 😊

Great work, and thank you!

closed 2021-03-08 by kushwavez

Sniki, 2021-03-08:
Nice, finally VoodooRMI is the all in one solution for the ThinkPad Touchpads, i felt really bad for the standard touchpad users that they couldn't enjoy the amazing TouchPad experience that VoodooRMI offers compared to any other existing solution. (way better than windows with the appropriate drivers).

Thanks @1Revenger1 , you are amazing !.

kushwavez, 2021-03-08:
Indeed. 
And the fact that it is working better on Mac than on Windows is just an icing on the cake. 
Marvellous!

1Revenger1, 2021-03-10:
@anhtutran37 please create a new issue or ask in the gitter.
This is unrelated to the issue (which is now closed)

PR #106 (2021-03-08): (Hopefully) fix stuck buttons
Fixes: #97 

Buttons have two sources: PS2 Passthrough (F03), or GPIO functions (F30/F3A).
The GPIO functions can have buttons which say they are there but are misconfigured or are not there. This can be seen as two additional buttons which are reported within the indexes for trackpoint buttons, but are always pulled low. VoodooPS2 correctly reports whether trackpoint buttons through GPIO functions should exist or not, so we use this data now when mapping to just ignore the trackpoint button indexes.