VoodooSMBus/VoodooRMI #98: Trackpad buttons not working

sortedcord, 2021-01-22:
I am on a HP 15bs 661tx Laptop with Synaptics SMBus touchpad. I used to have Mojave 1.14.6 with clover and I had VoodooRMI installed on that and it worked without any such issues. However, I then switched over to OpenCore and Catalina 1.15.7 and the touchpad buttons would not work anymore.  [My battery is already patched just in case]
All of the gestures work as normal just that the touchpad buttons don't work. I tried using both VoodooPS2Controller's `voodoo input.kext` and voodooRMI's but none of them work.
However one thing that I did notice was that when I was booting in, the logs did register the button clicks.

`Plugins in VoodooRMI:-`
-> RMISMBus.kext
->VoodooInput.kext

`Plugins in VoodooPS2`
->VoodooPS2Keyboard.kext
->VoodooPS2Trackpad.kext

This configuration would work in Mojave but not in Catalina along with OC. Any kind of assistance would be appreciated.

1Revenger1, 2021-01-23:
Do you have any logs from VoodooRMI?

Is VoodooRMI up to date?

My guess is an issue with VoodooInput but I wouldn't know why

sortedcord, 2021-01-23:
I have all of my kexts updated and I keep checking for Updates on a regular basis that includes voodooRMI. Also I dont think it is a hardware issue as I have linux dualbooted alongside. And Also I noticed that `ApplePS2SmartTouchPad.kext` work well in the sense that my trackpad buttons work but the gestures are all weird.

sortedcord, 2021-01-23:
# Logs from RMI

2021-01-23 21:23:33.257740+0530 0x12f3     Default     0x0                  443    0    useractivityd: [com.apple.useractivity:main] DETERMINE: F83805D0-0220-4CD9-AF4F-4156A23A9F0C/<private> as the current item-to-advertise.
2021-01-23 21:24:33.119854+0530 0x156b     Default     0x0                  443    0    useractivityd: [com.apple.useractivity:main] DETERMINE: F83805D0-0220-4CD9-AF4F-4156A23A9F0C/<private> as the current item-to-advertise.
2021-01-23 21:24:33.596839+0530 0x1688     Default     0x0                  443    0    useractivityd: [com.apple.useractivity:main] DETERMINE: F83805D0-0220-4CD9-AF4F-4156A23A9F0C/<private> as the current item-to-advertise.
2021-01-23 21:24:42.146996+0530 0x1706     Default     0x0                  443    0    useractivityd: [com.apple.useractivity:main] DETERMINE: F83805D0-0220-4CD9-AF4F-4156A23A9F0C/<private> as the current item-to-advertise.
2021-01-23 21:26:17.139261+0530 0x199b     Default     0x0                  443    0    useractivityd: [com.apple.useractivity:main] DETERMINE: No item is being advertised, so removing any previously set of advertised items.
2021-01-23 21:27:17.110823+0530 0x1ac0     Default     0x0                  443    0    useractivityd: [com.apple.useractivity:main] DETERMINE: F83805D0-0220-4CD9-AF4F-4156A23A9F0C/<private> as the current item-to-advertise.
2021-01-23 21:27:29.145239+0530 0x1af6     Default     0x0                  443    0    useractivityd: [com.apple.useractivity:main] DETERMINE: F83805D0-0220-4CD9-AF4F-4156A23A9F0C/<private> as the current item-to-advertise.
2021-01-23 21:27:29.192517+0530 0x19cf     Default     0x0                  443    0    useractivityd: [com.apple.useractivity:main] DETERMINE: F83805D0-0220-4CD9-AF4F-4156A23A9F0C/<private> as the current item-to-advertise.
2021-01-23 21:27:29.367005+0530 0x1af6     Default     0x0                  443    0    useractivityd: [com.apple.useractivity:main] DETERMINE: F83805D0-0220-4CD9-AF4F-4156A23A9F0C/<private> as the current item-to-advertise.
2021-01-23 21:27:29.423368+0530 0x1af6     Default     0x0                  443    0    useractivityd: [com.apple.useractivity:main] DETERMINE: F83805D0-0220-4CD9-AF4F-4156A23A9F0C/<private> as the current item-to-advertise.
2021-01-23 21:27:29.583863+0530 0x1af7     Default     0x0                  443    0    useractivityd: [com.apple.useractivity:main] DETERMINE: F83805D0-0220-4CD9-AF4F-4156A23A9F0C/<private> as the current item-to-advertise.
2021-01-23 21:27:29.640294+0530 0x1af7     Default     0x0                  443    0    useractivityd: [com.apple.useractivity:main] DETERMINE: F83805D0-0220-4CD9-AF4F-4156A23A9F0C/<private> as the current item-to-advertise.
2021-01-23 21:27:29.779230+0530 0x1af6     Default     0x0                  443    0    useractivityd: [com.apple.useractivity:main] DETERMINE: F83805D0-0220-4CD9-AF4F-4156A23A9F0C/<private> as the current item-to-advertise.
2021-01-23 21:27:29.835603+0530 0x1af6     Default     0x0                  443    0    useractivityd: [com.apple.useractivity:main] DETERMINE: F83805D0-0220-4CD9-AF4F-4156A23A9F0C/<private> as the current item-to-advertise.

sortedcord, 2021-01-23:
Here is a screenshot from IORegistryExplorer
[image]
[image]

sortedcord, 2021-01-25:
# Update
Alright so I'm using clover right now and VoodooRMI works. This is not solving the problem, just avoiding it. I'll provide further updates if I'm able to detect anything. But it seems like my OC install does not load VoodooRMI.

sortedcord, 2021-02-01:
Looks like my kexts were in the wrong order. I corrected that and now the VoodooRMI loads properly.

closed 2021-02-01 by sortedcord