VoodooSMBus/VoodooRMI #100: [Lenovo ThinkPad X1 Carbon 6th] v1.3's movement quality is worse than v1.2's

vaczi001, 2021-01-29:
**Describe the bug**
Touch behaviour, sensitivity, gestures quality and the mouse cursor movement quality are worse on v1.3 than on v1.2
Sometimes it doesn't register my movements and the mouse cursor movements are also feels bad.

Tested 3 times just to make sure that there is significant quality difference between v1.2 and v1.3

I downgraded back to VoodooRMI v1.2

**To Reproduce**
Steps to reproduce the behavior:
1. Use VoodooRMI v1.3

**Expected behavior**
Good movements, and overall quality like was on VoodooRMI v1.2 (v1.2 is significantly better, actually it's better than on Windows)

**System**
Lenovo ThinkPad X1 Carbon 6th
Synaptics Touchpad (SMBus)
Clover v5129
VoodooRMI v1.3 (I2C removed)
VoodooPS2Controller v2.2.0 (VoodooPS2Trackpad, VoodooInput, VoodooPS2Mouse removed)

**Log**
[logv13.txt](https://github.com/VoodooSMBus/VoodooRMI/files/5895760/logv13.txt)

1Revenger1, 2021-01-29:
Can you try using VoodooSMBus from v1.2? (Leave VoodooRMI alone)
How the trackpad behaves shouldn't really change, only thing I can think of is maybe changes I made in VoodooSMBus.

vaczi001, 2021-01-29:
Hmm. Maybe it's my imagination but the mouse cursor movements feels better now.

But sometimes it's still doesn't recognise my touches.

singhalrishi27, 2021-01-31:
#92

1Revenger1, 2021-02-04:
Looking at the original log you gave, are you trying to do gestures right after using the keyboard?
How palm rejection works is that it checks a couple different things
* Very tiny touches (this doesn't seem to be the issue)
* Very tall touches in the bottom half of the trackpad (think like laying your finger down vertically)
* Rejects any fingers in boxes on the left, right, and top 1/4 of the trackpad 2 seconds after typing
  * This timeout is configurable in the Info.plist for VoodooRMI
If it's just not detecting fingers though at all, I can't see why it wouldn't - everything else looks fine looking through the log.

kushwavez, 2021-02-05:
No, palm rejection works as intended, but the problem is the sensitivity (I suppose?). Sometimes when I try to swipe for example two finger swipe to scroll it's just doesn't seems to recognise one of my finger and just stop scrolling. 
With v1.2 I have none of the mentioned issues at all, working perfectly fine. 

Now I'm waiting for Clover's dev to (hopefully) hardcode the kext order for VoodooRMI, after that I'll try with v1.3.1 (right now it won't load) and will use for some time again. 
If somethings feels off I'll definitely notice because I'm using v1.2 for months now and it's perfectly fine.

kushwavez, 2021-02-05:
I opened an issue on Clover's git about the order issue, for now I added VoodooSMBus.kext to the order manually:

[CloverX64-RELEASE-20210204204543-6bc80b8-dirty-kushwavez.efi.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5931774/CloverX64-RELEASE-20210204204543-6bc80b8-dirty-kushwavez.efi.zip)
With this, v1.3.1 (and other versions) can load successfully. 
It's working fine, this time I don't recognise any issue, I have to modify palm rejection configuration indeed, but other than that v1.3.1 seems fine.

kushwavez, 2021-02-05:
Okay new things about swiping: If I swipe from the very bottom, the gesture doesn't recognised. Also the same if I swipe from the very top. 
An explanation video: https://streamable.com/nw7s36

Another about palm rejection. Something's off for me. Sometimes after writing the mouse is only rejected for 0.5 seconds after the 2nd movement, not the 1st one.

1Revenger1, 2021-02-05:
[VoodooRMI-1.3.2-NO_MIN_Z.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5934940/VoodooRMI-1.3.2-DEBUG.zip)
[VoodooRMI-1.3.2-NO_Z_NO_SHAPE.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5934944/VoodooRMI-1.3.2-DEBUG.zip)
Do you mind trying out the two versions above? These would be most likely the changes that would've made it worse.

singhalrishi27, 2021-02-06:
> [VoodooRMI-1.3.2-NO_MIN_Z.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5934940/VoodooRMI-1.3.2-DEBUG.zip) [...]

idk I've only Tried No MIN Z and everything is working as expected do you want me try NO SHAPE version

1Revenger1, 2021-02-06:
No, if NO_MIN_Z works fine, then no need to test NO_Z_NO_SHAPE. If kushwavez says that NO_MIN_Z is all good as well, then I'll just add those edits to the repo. It's a bigger issue if only NO_Z_NO_SHAPE works well though as I would need to make a lot of changes to palm rejection likely 😅

kushwavez, 2021-02-06:
I'm really sorry @1Revenger1 but looks like NO_Z_NO_SHAPE works better than NO_MIN_Z 😞

VoodooRMI-1.3.2-NO_MIN_Z.zip -> Scrolling is good now, but palm rejection behaves the same. After typing, on first movement it's not immediately rejecting, but on the second movement. (or maybe it's not related to first and second movement, but time? Like it's only rejecting after ~0.5 seconds)

VoodooRMI-1.3.2-NO_Z_NO_SHAPE.zip -> Scrolling is good, palm rejection also good, I have no problems with this.

1Revenger1, 2021-02-06:
[VoodooRMI-1.3.2-NO_Z_NO_TOP_THUMB.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5937769/VoodooRMI-1.3.2-DEBUG.zip)
Mind trying this version? If this behaves similarly to NO_MIN_Z, then it's something wrong with my finger width/height thing. If this behaves correctly though, likely something to do with trying to detect a thumb while using tracepoint buttons.

kushwavez, 2021-02-06:
VoodooRMI-1.3.2-NO_Z_NO_TOP_THUMB.zip -> Unfortunately still experiencing the same issue.

1Revenger1, 2021-02-07:
[VoodooRMI-1.3.2-Shuffle.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5937894/VoodooRMI-1.3.2-DEBUG.zip)
[VoodooRMI-1.3.2-NO_WIDTH_HEIGHT.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5937896/VoodooRMI-1.3.2-DEBUG.zip)

Do you mind trying the two above? I reordered some stuff.

kushwavez, 2021-02-07:
VoodooRMI-1.3.2-Shuffle.zip -> same as before
VoodooRMI-1.3.2-NO_WIDTH_HEIGHT.zip -> Basic and other movements are glitched out, doesn't really work at all.

Sniki, 2021-02-11:
I can confirm the same problem on My Lenovo ThinkPad T440S and X240 which both have the upgraded trackpads with physical buttons from T450S and X250.

The movements are annoying, on the corners it somehow doesn’t get input.
Previous version was fine and even better the one before palm rejection, is there a way to disable that ?,
For me the standard rejection time implemented in macOS was perfect, honestly for people with slow typing and switching between touch and keyboard it may be useful but for power users it’s a PITA having to wait for the rejection time.

i would be really happy to have an option to disable it if it’s possible.

1Revenger1, 2021-02-19:
The rejection for typing only works on the left, right, and top of the trackpad. It shouldn't prevent movements in the lower 80% of the trackpad and 80% width of the trackpad.

macOS doesn't have any sort of rejection time built in, it's built into VoodooPS2/VoodooI2C/VoodooRMI

With this version below, I've reduced the amount of time the palm rejection works for (2s -> 250ms) though this is configurable in the info.plist within the kext. I've also reduced the box size on the top a little for palm rejection, and fix some other things. Do you guys mind trying this out?
[VoodooRMI-1.3.2-DEBUG.zip](https://github.com/VoodooSMBus/VoodooRMI/files/6007085/VoodooRMI-1.3.2-DEBUG.zip)

kushwavez, 2021-02-19:
> VoodooRMI-1.3.2-DEBUG.zip

Movements, scrolling, gestures are good with this. 
As far as I can see palm rejection is also good. The possibility to configure or disable palm rejection is a really good move.

It's been just 4 hours since I started using it, but I can't see any problems with it so far. 
I'll use it for 1 more day, if I can't see any problems I'll close this issue.

Thank you very much for your hard work.👏🏻

closed 2021-02-20 by kushwavez

singhalrishi27, 2021-02-21:
> The rejection for typing only works on the left, right, and top of the trackpad. It shouldn't prevent movements in the lower 80% of the trackpad and 80% width of the trackpad.
>
> macOS doesn't have [...]

Hii this also worked for me can you commit this so I can get it in release version

kushwavez, 2021-02-21:
Also some minor thing on this 1.3.2 DEBUG: 
When I just move the cursor with one finger, sometimes it's losing the movement and start clicking. Something has to do with sensitivity. On v1.2 it was fine.

PR #104 (2021-02-27): Simplify check for palm rejection
Fixes #100 

This is reverting some changes I made with a previous PR, while tuning some numbers. This specifically:
* Always compares wx/wy of a finger (Conditionally checking this seemed to cause weird issues)
* Reduce timeout while typing
* Reduce size of top box for trackpoint/top buttons