VoodooSMBus/VoodooRMI #102: Thinkpad X1 Carbon 6th Trackpoint middle button clicks while on scroll

kskim7, 2021-02-04:
Problem : 
The trackpoint scrolling behaves as 'clicks' while it's on scroll mode.

First of all, thanks for your work! With your VoodooRMI's, almost everything works as it should!
I'm using your VoodooRMI's on my X1 Carbon 6th on Big Sur, and have upgraded to your newly updated v1.3.1 in hope it solves the problem.  
Everything seems fine, but I can't figure out why the trackpoint scrolling behaves as 'clicks' while it's on scroll mode.
It opens up 20-30 tabs in a second as the cursor passes on any clickable link when scrolling on a web browser. 
The touchpad doesn't show such behavior. 

**Expected behavior**
Trackpoint's middle button is used for scrolling. And when it's pressed and begin scrolling, it should continue working as scroll button, not the middle click button while it's pressed on.

Any idea on how to solve the problem? Might be some other issue other than voodoormi's??

1Revenger1, 2021-02-04:
Can you send an ioreg using IORegistryExplorer? My X1 doesn't have this issue, and I wonder if it's something related to trackpoint buttons not being sent over the PS2 passthrough.

kskim7, 2021-02-05:
Thanks, oh so you have an X1 too. 
It’s strange that no one is experiencing what I have.
Actually, I’ve tested several EFI’s made by different users, and this is the only EFI(by Kushwavez) that shows such behavior. But I need to stick using this EFI in order to have everything work(except the scroll) because other EFI’s didn’t boot at all.  

So.. yeah, since I am not familiar with IOregistry, I just saved the whole thing on the first screen on IORegistryExplorer.
Would this be okay?
Or should I just screenshot it??

Thanks a lot!!
[kskim7's x1 ioreg.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5929586/kskim7.s.x1.ioreg.zip)

1Revenger1, 2021-02-05:
That was exactly what I was looking for.

The scrolling thing looks like a bug with VoodooPS2, as VoodooRMI isn't actually loading up at all. Looking under SBUS, VoodooRMI isn't seen there at all. I'd check that you have the latest version of VoodooRMI and that the load order is correct (VoodooRMI/VoodooSMBus before RMISMBus in your config.plist)

kskim7, 2021-02-05:
Oh, indeed, I don't see VoodooRMI on the IOReg trees. Perhaps I must add some lines on the config.plist?
But the thing is, I'm using Clover, not OpenCore, so the config.plist doesn't contain any voodoo related lines. 

Btw, I asked the same issue to the original publisher(kushwavez) and he said no scrolling thing on his X1.
So perhaps VoodooRMI's working fine on his X1. A hardware problem maybe??

1Revenger1, 2021-02-05:
What version of Clover? The versions which boot Big Sur I think has issues injecting VoodooRMI due to the order of kexts - but I don't use Clover on my machine and don't really have the time to set it up and file a report to them

kskim7, 2021-02-05:
It's Clover v5129 - the latest version. 
In fact, that #103 issue is the same author that I referred to. 
I see.. maybe a clover issue then.
I had once tried another OC efi files, and when it booted up, it didn't have the scroll issue.

closed 2021-02-05 by 1Revenger1

kushwavez, 2021-02-05:
@kskim7 I created an issue on Clover's git about the order issue. The fix is really quick and the devs will fix this real soon hopefully. In the meantime you can use my modified CLOVERX64.efi, I fixed the order issue.
[CloverX64-RELEASE-20210204204543-6bc80b8-dirty-kushwavez.efi.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5931823/CloverX64-RELEASE-20210204204543-6bc80b8-dirty-kushwavez.efi.zip)