VoodooSMBus/VoodooRMI #103: VoodooRMI v1.3.1 won't load - Clover v5129

kushwavez, 2021-02-04:
**Describe the bug**
VoodooRMI v1.3.1 won't load, VoodooPS2Trackpad being used instead. Kextstat shows that VoodooSMbus is loaded, but not RMI.

**Expected behavior**
VoodooRMI loaded instead of VoodooPS2

**Log**
[2021-2-4_21-27-28_CLOVERX64.efi.log](https://github.com/VoodooSMBus/VoodooRMI/files/5928639/2021-2-4_21-27-28_CLOVERX64.efi.log)
[rmi1.3.1_log.log](https://github.com/VoodooSMBus/VoodooRMI/files/5928640/rmi1.3.1_log.log)

**System**
Lenovo X1 Carbon 6th 
Clover v5129
VoodooPS2Controller v2.2.1 - DEBUG 
VoodooRMI v1.3.1 - DEBUG

1Revenger1, 2021-02-04:
VoodooSMBus needs to be loaded before RMISMBus - this is a Clover bug
If you look at the log, it attempts to inject RMISMBus before VoodooSMBus. Order should be as followed:
SMBus: VoodooSMBus/VoodooRMI -> RMISMBus
I2C: VoodooI2C/VoodooRMI -> RMII2C
Afaik, where VoodooInput goes doesn't really matter that much

closed 2021-02-04 by 1Revenger1

kushwavez, 2021-02-05:
I see, thanks. I'll ask the devs to hardcode VoodooRMI's orders

kushwavez, 2021-02-05:
In the meantime I added VoodooSMBus to the order array: 
[CloverX64-RELEASE-20210204204543-6bc80b8-dirty-kushwavez.efi.zip](https://github.com/VoodooSMBus/VoodooRMI/files/5931809/CloverX64-RELEASE-20210204204543-6bc80b8-dirty-kushwavez.efi.zip)