VoodooSMBus/VoodooRMI #107: Doc Does not specify where is VoodooTrackpoint...

amitchaudhari9121, 2021-03-20:


amitchaudhari9121, 2021-03-20:
I solved my issue!

It seems that trackpoint kext was removed in 1.3. so I rolledback to kexts from version 1.2 release and it works flawlessly now.

could you please update the docs?

Either mention version specifics (opencore users use 1.2, or something like that), or add an extra note about trackpoint?

1Revenger1, 2021-03-20:
I guess I forgot to remove it from the docs - I'll update it at some point in the next couple days.

It got merged into VoodooInput, so it is not required anymore. The trackpoint should still work normally.

amitchaudhari9121, 2021-03-20:
It did not work though,
Trackpoint was not working at all , and the physical buttons on top of tracpad were not working as well. rolled back and both started working just fine. 

I am fairly new to this hackintosh (this is my first time 😊 ) 
But I once opened my specific laptop (ThinkPad X1C3 2015) down to every part. and the trackpoint is a separate module completely. So maybe, it needed a separate entry in the config.plist. Complete Speculation on my point though. I don't even know how would i should go about confirming it though...

I'd love to investigate further !!!

1Revenger1, 2021-03-20:
It requires you to use the VoodooInput from VoodooRMI, or atleast an up to date version of VoodooInput. Make sure it isn't one from a really old version of VoodooPS2 or an old version that you put in yourself.

amitchaudhari9121, 2021-03-21:
Pull request #108 addresses this now . 

Thankyou!

closed 2021-03-21 by amitchaudhari9121