VoodooSMBus/VoodooRMI #109: ThinkPad T430 Trackpad and Trackpoint not working

ALeRobbies, 2021-04-06:
Decription:
I tried installing VoodooRMI on a Thinkpad T430 running Big Sur 11.2.3. Trackpad is running over SMBus according to Windows. However, VoodooRMI won't load and the trackpad is left nonfunctional. All necessary kexts (VoodooPS2 with VoodooPS2Mouse disabled, VoodooSMBus, VoodooRMI, RMISMBus and VoodooInput) are installed and loading in the correct order.

Log:
[    3.770115]: VRMI - Info: Found PS2 Trackpad driver! Waiting for registerService()
[    3.770464]: VRMI - Debug: Too early notification - No VoodooInput on PS2 yet
[    4.163671]: VRMI - Info: VoodooPS2Mouse finished init, starting...
[    9.176930]: VRMI - Error: Error: Failed to read SMBus version. Code: 0xFFFFFFFA

According to IOReg, everything is loaded except for VoodooRMI (because it fails to load).
[image]

1Revenger1, 2021-04-06:
Can you send an ioreg file?
I'd also make sure that you are using Acidanthera's VoodooPS2 2.2.0 or later.

ALeRobbies, 2021-04-06:
Here is the IOReg
[Robert's MacBook Pro.zip](https://github.com/VoodooSMBus/VoodooRMI/files/6267503/Robert.s.MacBook.Pro.zip)

Also, I am running Acidanthera's latest VoodooPS2 2.2.2

1Revenger1, 2021-04-06:
Huh
So, the error code means that the trackpad isn't found/responding over SMBus
I looked through your ioregistry as well, and VoodooPS2 doesn't report that the trackpad supports SMBus.

I know that there were issues in the past where the trackpad needed to be reset over PS2 before it was recognized, though I don't think that this is similar. It did give a similar errors though.

Unless you find anymore information (could look within synaptics.c or rmi_driver.c within the linux kernel), I don't think there is anything I can do right now.

closed 2021-09-11 by 1Revenger1