VoodooSMBus/VoodooRMI #115: Thinkpad L14 Gen 1 - Tracking issues when swiping fast and tapping (maybe a rejection issue?)

Caipe97, 2021-05-09:
**Describe the bug**
Using VoodooRMI in SMBUS mode on a Thinkpad L14 Gen 1, the touchpad tracking is innacurate when swiping fast, and taps are usually rejected.

macOS Big Sur 11.2.2

**To Reproduce**
Steps to reproduce the behavior:
1. Install VoodooPS2Controller
2. Install VoodooRMI 1.3.3 with included VoodooSMBUS (disabling I2C kexts as in guide)
3. Try to use the touchpad while swiping fast with one or many fingers

**Expected behavior**
Tracking should be accurate and should not be rejected

**Log**
[log.txt](https://github.com/VoodooSMBus/VoodooRMI/files/6448369/log.txt)
I used the debug version of VoodooRMI to capture the log while doing the following gestures, aided by FingerMgmt:

Swiping to any direction with one finger:
[image]

It works fine if the gesture isn't quick:
[image]

Tapping is also troublesome:
[image]

Caipe97, 2021-05-12:
Update: Testing the Debug version VoodooRMI-1.3.2-NO_Z_NO_SHAPE from issue #100 seems to make the touchpad work normally. However, while using it I get huge power spikes (possibly because of the constant debugging messages).

Any insights?

1Revenger1, 2021-09-02:
Sorry that I haven't replied
Is the newest version here any better?
https://dortania.github.io/builds/?product=VoodooRMI&viewall=true
The palm rejection should be reworked, with the conditions I removed in NO_Z_NO_SHAPE relaxed.

closed 2021-09-11 by 1Revenger1

Caipe97, 2021-09-18:
I find that it works pretty good, though the touchpad tracking has become quite stuttery, still bearable tho... Could it be related to battery patching?

1Revenger1, 2021-09-18:
hrmm, it depends. If your using ACPIBatteryManager, and the _STA method for your battery includes a wait, it could be. If not, I honestly don't have any clues. Is it any better in the center vs the edges?

1Revenger1, 2021-09-18:
If you want to try another version of VoodooSMBus, you could try this
[VoodooSMBus.kext.zip](https://github.com/VoodooSMBus/VoodooRMI/files/7190738/VoodooSMBus.kext.zip)
This changes how interrupts work a bit, though I haven't really noticed any difference on my machine.

Caipe97, 2021-10-02:
I've tried the latest version and the cursor isn't still quite as smooth as with VoodooPS2. Tested version 1.3.5 from your link + your VoodooSMBus kext. I'm using SMCBatteryManager and my DSDT patch, I'll get back with details when I have the time. Gestures work perfectly. Definitely an improvement over what 1.3.3 was.

1Revenger1, 2021-10-02:
Hrmm, I know that someone else at some point had run into a similar thing in Linux. Something to do with the reporting rate of the trackpad. I'm not sure if there's a way I can change reporting rate though, not super familiar with the F12 function.

1Revenger1, 2021-10-02:
[VoodooRMI-1.3.4-DEBUG-Read-reportrate-bit.zip](https://github.com/VoodooSMBus/VoodooRMI/files/7272370/VoodooRMI-1.3.4-DEBUG-Read-reportrate-bit.zip
[VoodooRMI-1.3.4-DEBUG-Set-Bit.zip](https://github.com/VoodooSMBus/VoodooRMI/files/7272373/VoodooRMI-1.3.4-DEBUG-Set-Bit.zip)
)

I'd be interested to see what happens if you try both of those versions. For the first version, look under the properties of F01->PowerProperties in ioreg and see what the report rate bit is set to. The second version sets the bit to 1. No clue if that will change anything though.

Caipe97, 2021-10-04:
The first version gives this info in ioreg:
[image]

These are all the available functions my touchpad supports, if you need a bit more context..:
[image]

Going to test the second version...

Caipe97, 2021-10-04:
I've found no difference using any of the versions regarding improvements... What I found is that the tracking feels very much like using the Thinkpad nub/trackpoint (which I've always found "jumpier" than the main touchpad)