VoodooSMBus/VoodooRMI #116: Dell 5558 Broadwell VoodooRMI Trackpad not working

gelomon, 2021-06-12:
**Describe the bug**
Successfully loaded all kexts, checked that my trackpad supports SMBUS but still its not working
Trackpad Details: Synaptics F12
Bootloader: OC

**VRMI LOG**
```
[    4.344392]: VRMI - Info: Found PS2 Trackpad driver! Waiting for registerService()
[    4.344702]: VRMI - Debug: Too early notification - No VoodooInput on PS2 yet
[   22.507334]: VRMI - Debug: Found GPIO data!
[   22.519434]: VRMI - Info: VoodooPS2Trackpad finished init, starting...
[   27.194328]: macl_read_xattr_contents: failed to size xattr: 2macl_read_xattr_contents: failed to size xattr: 2macl_get_vnode: failed to resolve vnode: 2macl_get_vnode: failed to resolve vnode: 2Sandbox: systemstats(72) deny(1) file-write-data /dev/io8logSandbox: systemstats(72) deny(1) file-write-data /dev/io8logtempSandbox: systemstats(72) deny(1) mach-lookup com.apple.tccd.systemVRMI - Error: Error: Failed to read SMBus version. Code: 0xFFFFFFFA
```

**VSMB LOG**
```
[    2.228595]: VSMB - Debug: Publishing nub for slave device at address 0x2c
[    2.228617]: VSMB - Debug: Publishing nub for slave device at address 0x15
[   24.292388]: Sandbox: distnoted(322) deny(1) file-read-metadata /usr/sbin/distnotedSandbox: distnoted(322) deny(1) file-read-metadata /usrSandbox: fud(83) allow mach-lookup com.apple.commcenter.coretelephony.xpcVSMB - Debug: No response
[   25.016653]: Sandbox: 2 duplicate reports for fud allow mach-lookup com.apple.commcenter.coretelephony.xpcSandbox: distnoted(326) deny(1) file-read-metadata /usr/sbin/distnotedSandbox: distnoted(326) deny(1) file-read-metadata /usrVSMB - Debug: No response
[   25.638717]: VSMB - Debug: No response
[   26.152288]: VSMB - Debug: No response
[   26.667140]: VSMB - Debug: No response
[   27.182090]: VSMB - Debug: No response
```

**VoodooPS2Trackpad Log**
```
[    3.041734]: VoodooPS2Trackpad: sending final init sequence...
[    3.203971]: VoodooPS2Trackpad: Identify bytes = { 0x1, 0x47, 0x18 }
[    4.294996]: VoodooPS2Trackpad starting: Synaptics TouchPad reports type 0x47, version 8.1
[    4.344349]: VoodooPS2Trackpad: nExtendedQueries=5
[    5.345360]: VoodooPS2Trackpad: supports EW=1
[    6.394386]: VoodooPS2Trackpad: passthru1=0, passthru2=0, passthru=0
[    7.444845]: VoodooPS2Trackpad: ledpresent=0
[    8.545648]: VoodooPS2Trackpad: Mode/model($01) bytes = { 0x2c, 0xc6, 0xc1 }
[    9.608317]: VoodooPS2Trackpad: Capabilities($02) bytes = { 0xd0, 0x3, 0x23 }
[   10.540174]: Sandbox: calculated role flags for 0x1000016 is 0x4VoodooPS2Trackpad: Model ID($03) bytes = { 0x1, 0xe2, 0xb1 }
[   11.730742]: VoodooPS2Trackpad: SN Prefix($06) bytes = { 0x0, 0x0, 0x0 }
[   12.648170]: System Policy: diskarbitrationd(94) deny(1) file-read-data /dev/rdisk4s5s1VoodooPS2Trackpad: SN Suffix($07) bytes = { 0x0, 0x0, 0x0 }
[   13.829501]: VoodooPS2Trackpad: Resolutions($08) bytes = { 0x0, 0x80, 0x2c }
[   14.637166]: Sandbox: analyticsd(137) deny(1) file-write-owner /Library/CoreAnalyticsSandbox: analyticsd(137) deny(1) file-write-mode /Library/CoreAnalyticsGot new blocked hash data: 1 hashesSetting 1 new blocked hashesGot new blocked team data: 1 teamsSetting 1 new blocked teamsGot new active rule version: 17112441767188538581Waking up reference: 2Thread waiting on reference 2 woke upevaluation result: 2, allowed, cache,  1623422757Waking up reference: 1Thread waiting on reference 1 woke upevaluation result: 1, allowed, cache,  1623422757VoodooPS2Trackpad: Extended Model($09) bytes = { 0x84, 0x3, 0x0 }
[   16.175464]: VoodooPS2Trackpad: Continued Capabilities($0C) bytes = { 0x12, 0xe8, 0x0 }
[   17.192367]: VoodooPS2Trackpad: clickpadtype=1
[   18.207672]: VoodooPS2Trackpad: _reportsv=0
[   19.207804]: VoodooPS2Trackpad: Trackpad supports extendedW mode
[   20.208360]: VoodooPS2Trackpad: Trackpad supports SMBus operationVoodooPS2Trackpad: Maximum coords($0D) bytes = { 0xb1, 0x40, 0x91 }
[   21.268758]: Sandbox: 11 duplicate reports for com.apple.AppleU deny(1) file-read-metadata /SystemSandbox: distnoted(317) deny(1) file-read-metadata /usr/sbin/distnotedSandbox: distnoted(317) deny(1) file-read-metadata /usrVoodooPS2Trackpad: Minimum coords bytes($0F) = { 0x27, 0xbf, 0x25 }
[   21.332497]: VoodooPS2Trackpad: logical 1528x1456-5414x4398 physical_max 7774x5886 upmm 50x50AppleLMUController::smcReadKey Error: received error 0x84 when reading key 'MSLD'
[   22.336107]: VoodooPS2Trackpad: sending final init sequence...
[   22.519434]: VRMI - Info: VoodooPS2Trackpad finished init, starting...
[   22.588040]: VoodooPS2Trackpad: sending kDP_Reset $FF
```

**IOREG**
[ioreg.zip](https://github.com/VoodooSMBus/VoodooRMI/files/6642810/ioreg.zip)

**Screenshots**

_kext loading_
[image]

_boot args_
[image]

1Revenger1, 2021-06-12:
Huh, this is the first dell device I've seen use an SMBus trackpad. Sure it's SMBus and not I2C? No response generally means that there isn't actually a device there.

gelomon, 2021-06-12:
Hmm I’m not quite sure but looking on guide and cheking ps2trackpad on ioreg and log shows smbus support. 😅

1Revenger1, 2021-06-12:
Might be best to check in Linux or Windows.
If Linux, run these in terminal:
```
sudo dmesg | grep -i input
xinput --list
```
If windows, I'd go into device manager and see if you can find an I2C HID device. If not, check for the SMBus driver controller. Should say something like Synaptics SMBus Device instead of Intel SMBus Controller. It won't show up as a HID/mouse device, you'll have to search around for it in another sections of device manager.

gelomon, 2021-06-12:
Damn, You’re right it’s an i2c. I’ll try using i2c again

closed 2021-06-12 by gelomon