VoodooSMBus/VoodooRMI #117: Lenovo Thinkbook 14-IIL SYNA2B60 detected on system preferences but cannot be used.

azisnaufal, 2021-06-22:
As per title. SYNA2B60 on Lenovo Thinkbook 14-IIL is detected. The system preferences does show the trackpad settings. But the touchpad cannot be used, the cursor is not moving at all. Here is the log:

```
[    3.826661]: VRMI - Debug: RMII2C::TPD0 probing
[    3.880179]: VRMI - Debug: RMII2C::TPD0 Trying to set mode, attempt 0
[    3.893512]: VRMI - Debug: RMII2C::TPD0 mode set
[    4.437065]: VRMI - Debug: RMII2C::TPD0 powerState 1 : on
[    4.437096]: VRMI - Debug: Config TrackpointMultiplier loaded: a -> a
[    4.437098]: VRMI - Debug: Config TrackpointScrollMultiplierX loaded: a -> 1e
[    4.437100]: VRMI - Debug: Config TrackpointScrollMultiplierY loaded: a -> 1e
[    4.437102]: VRMI - Debug: Config TrackpointDeadzone loaded: 1 -> 1
[    4.437103]: VRMI - Debug: Config DisableWhileTypingTimeout loaded: 7d0 -> fa
[    4.437105]: VRMI - Debug: Config DisableWhileTrackpointTimeout loaded: 7d0 -> fa
[    4.437107]: VRMI - Debug: Config ForceTouchMinPressure loaded: 50 -> 5a
[    4.437109]: VRMI - Debug: Config ForceTouchEmulation loaded: 1 -> 1
[    4.437110]: VRMI - Debug: Config MinYDiffThumbDetection loaded: c8 -> c8
[    4.437112]: VRMI - Debug: Config FingerMajorMinorDiffMax loaded: 3 -> 3
[    4.437113]: VRMI - Debug: Updating Configuration
[    4.437119]: VRMI - Info: RMI Bus (DEBUG) Starting up!
[    4.438595]: VRMI - Debug: RMII2C::TPD0 mode set
[    4.438612]: VRMI - Info: RMII2C::TPD0 reset completed
[    4.438613]: VRMI - Error: Unable to reset
[    4.438614]: VRMI - Error: RMI initial reset failed! Continuing in spite of this
[    4.439059]: VRMI - Debug: rmi_probe_interrupts: Counting IRQs
[    4.446619]: VRMI - Debug: IRQ Count: 8
[    4.446652]: VRMI - Debug: rmi_init_functions: Creating functions
[    4.447479]: VRMI - Info: Initializing F34.
[    4.447503]: VRMI - Debug: Function F34 - IRQs: 1 CMD Base: 0 CTRL Base: 63 DATA Base: 0 QRY Base: 118 VER: 0
[    4.447506]: VRMI - Info: F34 not implemented
[    4.448061]: VRMI - Info: Initializing F01.
[    4.448077]: VRMI - Debug: Function F1 - IRQs: 2 CMD Base: 116 CTRL Base: 67 DATA Base: 19 QRY Base: 127 VER: 0
[    4.452703]: VRMI - Info: Found RMI4 device, manufacturer: Synaptics, product: TM3336-003, fw id: 2716135
[    4.455099]: VRMI - Info: Initializing F11.
[    4.455120]: VRMI - Debug: Function F11 - IRQs: 2 CMD Base: 117 CTRL Base: 73 DATA Base: 23 QRY Base: 156 VER: 1
[    4.463325]: VRMI - Info: Initializing F30.
[    4.463343]: VRMI - Debug: Function F30 - IRQs: 1 CMD Base: 256 CTRL Base: 269 DATA Base: 256 QRY Base: 348 VER: 0
[    4.464516]: VRMI - Debug: F30: Found Button 1
[    4.465064]: VRMI - Info: Initializing F54.
[    4.465083]: VRMI - Debug: Function F54 - IRQs: 1 CMD Base: 347 CTRL Base: 280 DATA Base: 257 QRY Base: 350 VER: 0
[    4.465085]: VRMI - Info: F54 not implemented
[    4.466443]: VRMI - Info: Initializing F55.
[    4.466461]: VRMI - Debug: Function F55 - IRQs: 1 CMD Base: 515 CTRL Base: 512 DATA Base: 512 QRY Base: 516 VER: 0
[    4.466463]: VRMI - Info: F55 not implemented
[    5.973839]: VRMI - Debug: RMII2C::TPD0 mode set
[    5.990971]: VRMI - Info: RMII2C::TPD0 reset completed
[    6.002613]: VRMI - Debug: Wakeup
```

OpenCore 0.7.0
VoodooI2C 2.6.5
VoodooRMI 1.3.3

What should I do next? Do I need to provide my dsdt too?

1Revenger1, 2021-06-22:
No DSDT needed.
Can you make sure in ioregistryexplorer that VoodooInput is attaching to F11?
Seems weird that setPowerState is being ran *after* initializing the trackpad, I wonder if that's where some of our issues are coming from related to power management...

Let me see if I can create a build of some sort which tries to fix that.

azisnaufal, 2021-06-22:
[image]

I see F11 and VoodooInput on the left side. Is it attached? 

here is full ioreg file https://drive.google.com/file/d/1xbEW-_Li2R1UyNElbO9mOw8sHC1EQmmb/view?usp=sharing

1Revenger1, 2021-06-22:
Looks fine, do you mind trying the version below and sending a log?
[VoodooRMI-1.3.4-DEBUG.zip](https://github.com/VoodooSMBus/VoodooRMI/files/6692325/VoodooRMI-1.3.4-DEBUG.zip)

azisnaufal, 2021-06-22:
Here is the log of the second 1.3.4 debug zip

```
[    3.839895]: VRMI - Debug: RMII2C::TPD0 probing
[    3.852602]: VRMI - Debug: RMII2C::TPD0 Trying to set mode, attempt 0
[    3.865768]: VRMI - Debug: RMII2C::TPD0 mode set
[    4.409246]: VRMI - Debug: Config TrackpointMultiplier loaded: a -> a
[    4.422001]: VRMI - Debug: Config TrackpointScrollMultiplierX loaded: a -> 1e
[    4.435127]: VRMI - Debug: Config TrackpointScrollMultiplierY loaded: a -> 1e
[    4.448174]: VRMI - Debug: Config TrackpointDeadzone loaded: 1 -> 1
[    4.460894]: VRMI - Debug: Config DisableWhileTypingTimeout loaded: 7d0 -> fa
[    4.474110]: VRMI - Debug: Config DisableWhileTrackpointTimeout loaded: 7d0 -> fa
[    4.487563]: VRMI - Debug: Config ForceTouchMinPressure loaded: 50 -> 5a
[    4.500769]: VRMI - Debug: Config ForceTouchEmulation loaded: 1 -> 1
[    4.513840]: VRMI - Debug: Config MinYDiffThumbDetection loaded: c8 -> c8
[    4.527127]: VRMI - Debug: Config FingerMajorMinorDiffMax loaded: 3 -> 3
[    4.540445]: VRMI - Debug: Updating Configuration
[    4.552967]: VRMI - Info: RMI Bus (DEBUG) Starting up!
[    4.567148]: VRMI - Debug: RMII2C::TPD0 mode set
[    4.580164]: VRMI - Info: RMII2C::TPD0 reset completed
[    4.593506]: VRMI - Error: Unable to reset
[    4.606482]: VRMI - Error: RMI initial reset failed! Continuing in spite of this
[    4.621148]: VRMI - Debug: rmi_probe_interrupts: Counting IRQs
[    4.641393]: VRMI - Debug: IRQ Count: 8
[    4.653953]: VRMI - Debug: rmi_init_functions: Creating functions
[    4.667953]: VRMI - Info: Initializing F34.
[    4.680195]: VRMI - Debug: Function F34 - IRQs: 1 CMD Base: 0 CTRL Base: 63 DATA Base: 0 QRY Base: 118 VER: 0
[    4.695317]: VRMI - Info: F34 not implemented
[    4.708444]: VRMI - Info: Initializing F01.
[    4.720878]: VRMI - Debug: Function F1 - IRQs: 2 CMD Base: 116 CTRL Base: 67 DATA Base: 19 QRY Base: 127 VER: 0
[    4.740404]: VRMI - Info: Found RMI4 device, manufacturer: Synaptics, product: TM3336-003, fw id: 2716135
[    4.757776]: VRMI - Info: Initializing F11.
[    4.770440]: VRMI - Debug: Function F11 - IRQs: 2 CMD Base: 117 CTRL Base: 73 DATA Base: 23 QRY Base: 156 VER: 1
[    4.793845]: VRMI - Info: Initializing F30.
[    4.806709]: VRMI - Debug: Function F30 - IRQs: 1 CMD Base: 256 CTRL Base: 269 DATA Base: 256 QRY Base: 348 VER: 0
[    4.823833]: VRMI - Debug: F30: Found Button 1
[    4.837723]: VRMI - Info: Initializing F54.
[    4.850852]: VRMI - Debug: Function F54 - IRQs: 1 CMD Base: 347 CTRL Base: 280 DATA Base: 257 QRY Base: 350 VER: 0
[    4.866761]: VRMI - Info: F54 not implemented
[    4.880878]: VRMI - Info: Initializing F55.
[    4.893473]: VRMI - Debug: Function F55 - IRQs: 1 CMD Base: 515 CTRL Base: 512 DATA Base: 512 QRY Base: 516 VER: 0
[    4.909041]: VRMI - Info: F55 not implemented
```

I still have a feeling of this is my fault at DSDT patching when gpio pinning but I'm not really sure why and where. Do you mind to check my DSDT ?

[dsdt-azis.zip](https://github.com/VoodooSMBus/VoodooRMI/files/6692506/dsdt-azis.zip)

1Revenger1, 2021-06-22:
I honestly am not that great with I2C pinning - I don't have any I2C trackpad. If you want to try and narrow it down though, you could try using `-vi2c-force-polling` to disable interrupts and use polling mode. This should work even if your pinning is messed up, though cause a lot more interrupts when your trackpad isn't doing anything.

azisnaufal, 2021-06-22:
OwO
with `-vi2c-force-polling`,  your last 1.3.4 debug kext, and my pathced dsdt.  My touchpad is usable right now. But still don't know what to patch to enable this touchpad without `-vi2c-force-polling` bootarg. Thanks anyway

azisnaufal, 2021-06-22:
Now I downgrade the version to 1.3.3 release kext, still with force polling bootargs. Thank you very much for that suggestion. closing this issue right now.

closed 2021-06-22 by azisnaufal