VoodooSMBus/VoodooRMI #125: Panic on Acer Nitro 5 an515-52 using VoodooRMI-1.3.4-RELEASE

TheLifeIsYours, 2021-09-08:
**Describe the bug**
On booting for the first time after having finished making an EFI using Dortania's guide, I get an error saying `Prelinked injection VoodooRMI.kext\Contents\PlugIns\RMII2C.kext () - Invalid Parameter`

**To Reproduce**
Steps to reproduce the behavior:
1. add VoodooRMI.kext and VoodooSMBus.kext from [VoodooRMI-1.3.4-RELEASE.zip](https://github.com/VoodooSMBus/VoodooRMI/releases/)
2. Rebuild config.plist
3. Insert USB into laptop (Acer Nitro 5 an515-52)
4. Pick EFI boot device
5. Wait for panic. 

**Expected behavior**
Expect to be able to boot into recoverymode after a bit.

**Log**
[opencore-2021-09-08-170834.txt](https://github.com/VoodooSMBus/VoodooRMI/files/7130728/opencore-2021-09-08-170834.txt)

1Revenger1, 2021-09-08:
RMII2C needs VoodooI2C. Neither are needed though if using it for an SMBus trackpad. So it's safe to disable RMII2C.

Do you have a picture of the panic? The log doesn't show it.

TheLifeIsYours, 2021-09-08:
[image]

TheLifeIsYours, 2021-09-08:
I'm thinking I might have configured something wrong, as I'm getting the same panic with the previous release as well.

1Revenger1, 2021-09-08:
That's a separate issue with configuring Booter->Quirks or CFG lock. Look at the troubleshooting section of the Dortania guide for more help. This isn't an issue with VoodooRMI. If you disable RMII2C.kext in your config.plost, you'd have the same issue

closed 2021-09-08 by 1Revenger1