VoodooSMBus/VoodooRMI #130: Support for SYNA7DB5.

afkniladri, 2021-09-19:
Going through the readme I came across that my device wasn't supported. Hence I raised an issue. Thanks.

1Revenger1, 2021-09-19:
The list isn't complete. Does SYNA7DB5 work with VoodooRMI? Could add it to the list if you want.

afkniladri, 2021-09-19:
Yes please do. Btw mine is I2C based working with (SSDT-XOSI)[https://dortania.github.io/Getting-Started-With-ACPI/Laptops/trackpad.html]. VoodooI2CHID works for me, synaptics.kext doesn't. Providing the details if it helps.

(MY REPOSITORY)[https://github.com/afkniladri/Acer-Nitro5-Hackintosh-OC]

1Revenger1, 2021-09-19:
Do you know which Trackpad function it uses? It'll either be F11/F12 and it'll be attached to RMIBus in IORegistryExplorer.

afkniladri, 2021-09-19:
Sorry but how to check it ? This is all I hav
[image]
e.

1Revenger1, 2021-09-19:
You'll only see it when using VoodooRMI

afkniladri, 2021-09-19:
Ok. I'll report to you soon.

afkniladri, 2021-09-19:
Should I disable VoodooI2CHID.kext ?

1Revenger1, 2021-09-19:
No you don't need to. VoodooRMI has a higher probe score than it.

afkniladri, 2021-09-19:
Sorry but what about VoodooI2CServices.kext and VoodooInput.kext (VoodooI2C) ? should I disable

1Revenger1, 2021-09-19:
VoodooI2C depends on VoodooI2CServices, so keep that no matter what.
Use the VoodooInput from VoodooRMI, disable the one in VoodooI2C

afkniladri, 2021-09-19:
ok.

afkniladri, 2021-09-19:
Still no RMIBUS in IORegistry , trackpad works. But when I disable I2CHID trackpad won't work.

1Revenger1, 2021-09-19:
May not work with it then, could look to see what the logging says. Add `msgbuf=1048576` to your boot-args, then use `sudo dmesg | grep -i "VRMI" > ~/Desktop/vrmi_log.txt` once you've rebooted and are in the macOS terminal

afkniladri, 2021-09-19:
can I share my config.plist with you ? You can lead me where I'm going wrong
[config.plist.zip](https://github.com/VoodooSMBus/VoodooRMI/files/7192452/config.plist.zip)

1Revenger1, 2021-09-19:
VoodooRMI should be before RMII2C.kext

afkniladri, 2021-09-22:
[image]

Excuse me but where to look for F11/F12 functionality ?

afkniladri, 2021-09-22:
[image]

afkniladri, 2021-09-22:
Working perfectly fine except the fact that when I test out the rejection on palm detection. It breaks the system. The trackpad won't anymore, ofc till next boot.

afkniladri, 2021-09-22:
Main Function is F12 . Just to confirm.
[image]

1Revenger1, 2021-09-22:
hrmm, that's kinda weird
can you provide a log?

afkniladri, 2021-09-22:
Nevermind .. after a few tweaks with the configuration (palmRejectionWidth specifically). A bit of tweaks here and there it would be flawless, just like windows. Your support much appreciated.

closed 2021-09-22 by afkniladri