VoodooSMBus/VoodooRMI #132: Disable All DeadZones in the Edges of Synaptics touchpad

hoseinrez, 2021-10-01:
Hey guys ...

1Revenger1, 2021-10-01:
Is the deadzone there in windows and other OSes? It might be a deadzone in the hardware. There isn't any deadzones in VoodooRMI for the bottom.

hoseinrez, 2021-10-01:
there is no disabled or deadzones for touchpad in other OSes
also before VoodooRMI and VoodooPS2Trackpad i was using Old ApplePS2SmartTouchpad.kext until i saw "Disable...EdgeActiveArea" i think ... in plist and i set them all for top, bottom, right and left to 0 then my problem solved !
Config -> Lenovo Z510 Ideapad with PS2 Synaptics touchpad on SMBus

1Revenger1, 2021-10-01:
Can you send an ioreg from ioregistryexplorer and a log using the `sudo dmesg` instructions in the README? As far as I know, deadzone isn't something I can set, unless for some reason some of the values that get sent are negative? Instead of using the msgbuf=1048576 boot-arg, use DebugEnhancer.kext as well.

hoseinrez, 2021-10-01:
Log and IOReg copy attached ...
Hosein's-MBP11,5.ioreg
log.txt

hoseinrez, 2021-10-02:
I have figured out clicking is working in the edges but there is no cursor movement.

1Revenger1, 2021-10-02:
I think it might be an issue with Apple's MT2 stack (or maybe a value VoodooInput reports but I doubt this taking a quick lance at the code, don't see anything defining a deadzone).

I do have an extremely tiny deadzone on the edges of my trackpad even with setting deadzones to zero. If the finger starts outside that zone and then goes back into it, it seems to work fine. It's only if the finger starts in that deadzone. I'm testing this on the bottom.

I don't think this is something that I can fix at least, unless maybe we added 10% to the min/max coordinates but I'm not sure that's a good decision?

closed 2021-10-02 by 1Revenger1

hoseinrez, 2021-10-02:
@1Revenger1, thanks for your response.
I think that's not a good idea for all touchpads but I can test it and report back to you.
Process will be faster if you tell me which part of the code should exactly change ...
so I don't have to read big part of the codes :)

1Revenger1, 2021-10-02:
https://github.com/VoodooSMBus/VoodooRMI/blob/master/VoodooRMI/Functions/Input/RMITrackpadFunction.cpp#L65-L66
Probably want to change the max values here. You'll need to add an offset to the x/y [here](https://github.com/VoodooSMBus/VoodooRMI/blob/master/VoodooRMI/Functions/Input/RMITrackpadFunction.cpp#L203-L204) as well. Maybe like 10% more, and then offset by 1/2 of whatever you add to the max x/y?

hoseinrez, 2021-10-02:
Thank you so much @1Revenger1 .

1Revenger1, 2021-10-03:
[VoodooRMI-1.3.5-DEBUG.zip](https://github.com/VoodooSMBus/VoodooRMI/files/7272449/VoodooRMI-1.3.5-DEBUG.zip)
This should have a 5% increase on all sides.

hoseinrez, 2021-10-03:
Yeah its working !
Now disabled areas on the sides are narrower.
And ... can I have the source ?

1Revenger1, 2021-10-03:
```
diff --git a/VoodooRMI/Functions/Input/RMITrackpadFunction.cpp b/VoodooRMI/Functions/Input/RMITrackpadFunction.cpp
index a594524..3b6bc0f 100644
--- a/VoodooRMI/Functions/Input/RMITrackpadFunction.cpp
+++ b/VoodooRMI/Functions/Input/RMITrackpadFunction.cpp
@@ -62,8 +62,10 @@ bool RMITrackpadFunction::start(IOService *provider)
              0, 0,
              max_x, trackpointRejectHeight);
     
-    setProperty(VOODOO_INPUT_LOGICAL_MAX_X_KEY, max_x, 16);
-    setProperty(VOODOO_INPUT_LOGICAL_MAX_Y_KEY, max_y, 16);
+    deadzone_mitigation = double(max_x) * 0.1;
+    deadzone_half = deadzone_mitigation / 2;
+    setProperty(VOODOO_INPUT_LOGICAL_MAX_X_KEY, max_x + deadzone_mitigation, 16);
+    setProperty(VOODOO_INPUT_LOGICAL_MAX_Y_KEY, max_y + deadzone_mitigation, 16);
     // Need to be in 0.01mm units
     setProperty(VOODOO_INPUT_PHYSICAL_MAX_X_KEY, x_mm * 100, 16);
     setProperty(VOODOO_INPUT_PHYSICAL_MAX_Y_KEY, y_mm * 100, 16);
@@ -200,8 +202,8 @@ void RMITrackpadFunction::handleReport(RMI2DSensorReport *report)
         transducer.currentCoordinates.width = obj.z / 2.0;
         transducer.timestamp = report->timestamp;
         
-        transducer.currentCoordinates.x = obj.x;
-        transducer.currentCoordinates.y = max_y - obj.y;
+        transducer.currentCoordinates.x = obj.x + deadzone_half;
+        transducer.currentCoordinates.y = max_y - obj.y + deadzone_half;
         
         int deltaWidth = abs(obj.wx - obj.wy);
         
diff --git a/VoodooRMI/Functions/Input/RMITrackpadFunction.hpp b/VoodooRMI/Functions/Input/RMITrackpadFunction.hpp
index 789c1bc..8d17ee0 100644
--- a/VoodooRMI/Functions/Input/RMITrackpadFunction.hpp
+++ b/VoodooRMI/Functions/Input/RMITrackpadFunction.hpp
@@ -77,6 +77,8 @@ public:
 protected:
     u16 min_x{0};
     u16 min_y{0};
+    u16 deadzone_mitigation{0};
+    u16 deadzone_half{0};
     u16 max_x;
     u16 max_y;
     u8 x_mm;
``
[vrmi_patch.txt](https://github.com/VoodooSMBus/VoodooRMI/files/7272547/vrmi_patch.txt)
`