VoodooSMBus/VoodooRMI #139: trackpad have lag and double tab not good 3finger for change desktop

Edwardwich, 2021-11-17:
**trackpad have lag and double tab not good 3finger for change desktop**
trackpad have lag and double tab not good 3finger for change desktop
this problem only on Mac OS Monterey

**To Reproduce**
voodooRMI.kext-v13.5
VoodooPS2Controller.kext-v2.2.7

**IORegistryExplorer**

[ioreg.zip](https://github.com/VoodooSMBus/VoodooRMI/files/7556957/ioreg.zip)

**Log**
log from Hackintool

[log .zip](https://github.com/VoodooSMBus/VoodooRMI/files/7556949/log.zip)

1Revenger1, 2021-11-17:
ACPIBatteryManager known to cause issues due to sleep delays in the _STA method for your batteries. Try using VirtualSMC

Edwardwich, 2021-11-17:
Its my laptop git 
https://github.com/OpenIntelWireless/itlwm

1Revenger1, 2021-11-18:
Sorry, I don't really understand what you are saying. It sounds like possibly an issue with your BAT device not getting notifications when plugging/unplugging the device. I'd try with ACPIBatteryManager anyways just to see if the issue goes away. If it does, then you can spend more time making it work better.

Edwardwich, 2021-11-18:
> Sorry, I don't really understand what you are saying. It sounds like possibly an issue with your BAT device not getting notifications when plugging/unplugging the device. I'd try with [...]

plz see this picture 

laptop adapor 60w and show 300w:
[image]

and after show using adapter :

[image]

[image]

and after back to charging : 

[image]

[image]

1Revenger1, 2021-11-19:
Oops, meant SMCBatteryManager in my previous post.

Does the lag go away when using SMCBatteryManager? Not worth trying to fix power adapter status if it doesn't fix the issue.

Edwardwich, 2021-11-19:
> Oops, meant SMCBatteryManager in my previous post.
> 
> Does the lag go away when using SMCBatteryManager? Not worth trying to fix power adapter status if it doesn't fix the issue.

Trackpad work better 
But have some lag

This problem I have smc battery manger kext Is it non-fixable? 
Iuse hackintosh Forums in telegram but no one find for fix this problem
Can you help?

closed 2021-11-21 by Edwardwich