VoodooSMBus/VoodooRMI #140: Synaptics trackpad issue with T440P using T450 3 button trackpad

VinylNerd, 2021-12-03:
Hello, please forgive my lack of logs and IO reg data per my first post, as it seems the way this should be setup is literally explained different in 3-4 places, and i have multiple guides telling me what to do with nothing working whatsoever.

basically i seen some posts where people said it only worked for them with voodoo RMI 1.2 and older, and im assuming the guide is updated for the most recent version, since the way the kexts are included in builds have been changing it seems impossible for me to verify how i should be attemtping this.

trackpad works, but 3 buttons dont

they also dont work in windows, but that also seems to be a problem, someone on reddid has said its working for them with voodooRMI and voodoosmbus kext aswell as voodoops2 and disable voodoo input, but it is not working for me.

this situation is extra confusing because at this point since its not working im feeling the 3 buttons are broken
Last login: Fri Dec  3 13:45:36 on ttys000
Connor@Connors-MBP-2 ~ % kextstat | grep Voodoo
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
   78    1 0                  0x18000    0x18000    de.leo-labs.VoodooSMBus (3.0) C7EA611A-7A6E-3210-800F-79585B09468C <14 6 5 3>
   88    1 0                  0x29000    0x29000    com.1Revenger1.VoodooRMI (1.3.4) E101F26E-9935-3092-AFF4-7796D1B84130 <6 5 3>
  118    0 0                  0x19000    0x19000    me.kishorprins.VoodooInput (1.1.3) 374A88BF-5D62-3ED2-8E75-6A331AF58A2E <61 6 5 3>
Connor@Connors-MBP-2 ~ 
[SynapticsTrackpadTroubleshoot.zip](https://github.com/VoodooSMBus/VoodooRMI/files/7650099/SynapticsTrackpadTroubleshoot.zip)
%

1Revenger1, 2021-12-03:
are you able to provide a log using DebugEnhancer.kext and `sudo dmesg | grep -i VRMI > ~/Desktop/log.txt`? This requires the debug version of VoodooRMI, and you'll want to try and not use the trackpad until you run the command after booting. There is a lot of spam in the log when using the trackpad which can overwrite the logs from bootup.

VinylNerd, 2021-12-03:
@1Revenger1 thanks, ill do that now, just to be clear the debug version is VoodooRMI.kext.dSYM right?

1Revenger1, 2021-12-03:
No, the dSym are debug symbols not useful to most people. Download the debug version from the release page.

VinylNerd, 2021-12-03:
thanks, here you go

[    3.816503]: VRMI - Info: Found PS2 Trackpad driver! Waiting for registerService()
[    3.816773]: VRMI - Debug: Too early notification - No VoodooInput on PS2 yet
[    4.190018]: VRMI - Debug: Found GPIO data!
[    4.190024]: VRMI - Info: VoodooPS2Trackpad finished init, starting...
[    7.131649]: ALF, old data swfs_pid_entry <ptr>, updaterules_msg <ptr>, updaterules_state <ptr>VRMI - Info: SMBus version 2
[    7.203047]: VRMI - Debug: Config TrackpointMultiplier loaded: a -> a
[    7.203054]: VRMI - Debug: Config TrackpointScrollMultiplierX loaded: a -> 1e
[    7.203058]: VRMI - Debug: Config TrackpointScrollMultiplierY loaded: a -> 1e
[    7.203062]: VRMI - Debug: Config TrackpointDeadzone loaded: 1 -> 1
[    7.203067]: VRMI - Debug: Config DisableWhileTypingTimeout loaded: 7d0 -> fa
[    7.203071]: VRMI - Debug: Config DisableWhileTrackpointTimeout loaded: 7d0 -> fa
[    7.203076]: VRMI - Debug: Config ForceTouchMinPressure loaded: 50 -> 5a
[    7.203081]: VRMI - Debug: Config ForceTouchEmulation loaded: 1 -> 1
[    7.203084]: VRMI - Debug: Config MinYDiffThumbDetection loaded: c8 -> c8
[    7.203088]: VRMI - Debug: Config FingerMajorMinorDiffMax loaded: a -> a
[    7.203093]: VRMI - Debug: Config PalmRejectionWidth loaded: f -> a
[    7.203097]: VRMI - Debug: Config PalmRejectionHeight loaded: 50 -> 3c
[    7.203105]: VRMI - Debug: Config PalmRejectionTrackpointHeight loaded: 14 -> 14
[    7.203109]: VRMI - Debug: Updating Configuration
[    7.203121]: VRMI - Info: RMI Bus (DEBUG) Starting up!
[    7.203125]: VRMI - Debug: Config Clickpad loaded: 0 -> 1
[    7.203128]: VRMI - Debug: Config TrackstickButtons loaded: 1 -> 1
[    7.203134]: VRMI - Info: Recieved GPIO Data
[    7.208722]: VRMI - Debug: rmi_probe_interrupts: Counting IRQs
[    7.228878]: VRMI - Debug: IRQ Count: 7
[    7.228915]: VRMI - Debug: rmi_init_functions: Creating functions
[    7.230784]: VRMI - Info: Initializing F34.
[    7.230792]: VRMI - Debug: Function F34 - IRQs: 1 CMD Base: 0 CTRL Base: 77 DATA Base: 0 QRY Base: 128 VER: 0
[    7.230798]: VRMI - Info: F34 not implemented
[    7.232673]: VRMI - Info: Initializing F01.
[    7.232682]: VRMI - Debug: Function F1 - IRQs: 1 CMD Base: 126 CTRL Base: 81 DATA Base: 19 QRY Base: 137 VER: 0
[    7.233055]: Sandbox: coreduetd(101) deny(1) mach-lookup com.apple.tccdSandbox: coreduetd(101) deny(1) mach-lookup com.apple.tccd.systemVRMI - Info: Found RMI4 device, manufacturer: Synaptics, product: TM3157-005, fw id: 2403788
[    7.254962]: VRMI - Info: Initializing F03.
[    7.254969]: VRMI - Debug: Function F3 - IRQs: 1 CMD Base: 0 CTRL Base: 0 DATA Base: 21 QRY Base: 164 VER: 0
[    7.259536]: VRMI - Info: Initializing F11.
[    7.259543]: VRMI - Debug: Function F11 - IRQs: 2 CMD Base: 127 CTRL Base: 87 DATA Base: 37 QRY Base: 166 VER: 1
[    7.285361]: VRMI - Info: Initializing F30.
[    7.285370]: VRMI - Debug: Function F30 - IRQs: 1 CMD Base: 256 CTRL Base: 269 DATA Base: 256 QRY Base: 348 VER: 0
[    7.289248]: VRMI - Debug: F30: Found Button 1
[    7.289257]: VRMI - Debug: F30: Found Trackpoint button 1
[    7.289261]: VRMI - Debug: F30: Found Trackpoint button 2
[    7.289264]: VRMI - Debug: F30: Found Trackpoint button 3
[    7.291150]: VRMI - Info: Initializing F54.
[    7.291159]: VRMI - Debug: Function F54 - IRQs: 1 CMD Base: 347 CTRL Base: 280 DATA Base: 257 QRY Base: 350 VER: 0
[    7.291165]: VRMI - Info: F54 not implemented
[    7.297864]: Got new blocked hash data: 1 hashesSetting 1 new blocked hashesGot new blocked team data: 1 teamsSetting 1 new blocked teamsGot new active rule version: 10415043430518216312VRMI - Debug: F03 - Consumed <ptr>h (14) from PS2 guest
[    7.354983]: Waking up reference: 3Waking up reference: 8Thread waiting on reference 8 woke upevaluation result: 8, allowed, cache,  1638281954Waking up reference: 1Waking up reference: 7Waking up reference: 4Thread waiting on reference 3 woke upevaluation result: 3, allowed, cache,  1638284451Waking up reference: 5Thread waiting on reference 1 woke upevaluation result: 1, allowed, cache,  1638281954Waking up reference: 2Thread waiting on reference 2 woke upevaluation result: 2, allowed, cache,  1638540950Waking up reference: 6Thread waiting on reference 7 woke upevaluation result: 7, allowed, cache,  1638281953Thread waiting on reference 4 woke upThread waiting on reference 5 woke upevaluation result: 4, allowed, cache,  1638281954evaluation result: 5, allowed, cache,  1638281954Sandbox: dasd(122) allow ipc-posix-shm-write-create com.apple.dasd.budgetsThread waiting on reference 6 woke upevaluation result: 6, allowed, cache,  1638281954VRMI - Debug: F03 - PS2 Command [Send: 0 Receive: 2 cmd: ff]
[    7.409908]: Sandbox: dasd(122) allow file-read-data /private/var/db/DuetActivityScheduler/DuetActivitySchedulerClassC.dbVRMI - Debug: F03 - Recieved data over PS2: fa
[    7.444903]: Sandbox: dasd(122) allow file-read-data /private/var/db/DetachedSignaturesVRMI - Debug: F03 - Recieved data over PS2: aa
[    7.473156]: VRMI - Debug: F03 - Recieved data over PS2: 0
[    7.473170]: VRMI - Debug: F03 - PS2 Command [Send: 0 Receive: 2 cmd: e1]
[    7.487886]: VRMI - Debug: F03 - Recieved data over PS2: fa
[    7.487897]: VRMI - Debug: F03 - Recieved data over PS2: 1
[    7.487901]: VRMI - Debug: F03 - Recieved data over PS2: e
[... cut ...] g: F03 - PS2 Command [Send: 0 Receive: 0 cmd: e7]
[    8.793256]: VRMI - Debug: F03 - Recieved data over PS2: fa
[    9.293684]: VRMI - Debug: F03 - PS2 Command [Send: 1 Receive: 0 cmd: f3]
[    9.301678]: VRMI - Debug: F03 - Recieved data over PS2: fa
[    9.308964]: VRMI - Debug: F03 - Recieved data over PS2: fa
[    9.809039]: VRMI - Debug: F03 - PS2 Command [Send: 0 Receive: 0 cmd: f4]
[    9.817165]: VRMI - Debug: F03 - Recieved data over PS2: fa
[   10.318190]: VRMI - Info: Finish PS2 init

1Revenger1, 2021-12-03:
Looks like the buttons are found. Mind sending a log after trying to use the track point and buttons?

VinylNerd, 2021-12-03:
yes, i rebooted and did that for you

first track point (small red dot on keyboard) just to be clear then all 3 buttons a few times each and then trackpoint again

[    3.768667]: VRMI - Info: Found PS2 Trackpad driver! Waiting for registerService()
[    3.768928]: VRMI - Debug: Too early notification - No VoodooInput on PS2 yet
[    4.141999]: VRMI - Debug: Found GPIO data!
[    4.142006]: VRMI - Info: VoodooPS2Trackpad finished init, starting...
[    6.610483]: ALF, old data swfs_pid_entry <ptr>, updaterules_msg <ptr>, updaterules_state <ptr>Sandbox: locationd(116) deny(1) mach-lookupSandbox: locationd(116) deny(1) mach-lookup com.apple.wifi.sharekitVRMI - Info: SMBus version 2
[    6.661136]: VRMI - Debug: Config TrackpointMultiplier loaded: a -> a
[    6.661144]: VRMI - Debug: Config TrackpointScrollMultiplierX loaded: a -> 1e
[    6.661149]: VRMI - Debug: Config TrackpointScrollMultiplierY loaded: a -> 1e
[    6.661154]: VRMI - Debug: Config TrackpointDeadzone loaded: 1 -> 1
[    6.661158]: VRMI - Debug: Config DisableWhileTypingTimeout loaded: 7d0 -> fa
[    6.661164]: VRMI - Debug: Config DisableWhileTrackpointTimeout loaded: 7d0 -> fa
[    6.661169]: VRMI - Debug: Config ForceTouchMinPressure loaded: 50 -> 5a
[    6.661174]: VRMI - Debug: Config ForceTouchEmulation loaded: 1 -> 1
[    6.661178]: VRMI - Debug: Config MinYDiffThumbDetection loaded: c8 -> c8
[    6.661182]: VRMI - Debug: Config FingerMajorMinorDiffMax loaded: a -> a
[    6.661187]: VRMI - Debug: Config PalmRejectionWidth loaded: f -> a
[    6.661191]: VRMI - Debug: Config PalmRejectionHeight loaded: 50 -> 3c
[    6.661195]: VRMI - Debug: Config PalmRejectionTrackpointHeight loaded: 14 -> 14
[    6.661200]: VRMI - Debug: Updating Configuration
[    6.661213]: VRMI - Info: RMI Bus (DEBUG) Starting up!
[    6.661218]: VRMI - Debug: Config Clickpad loaded: 0 -> 1
[    6.661221]: VRMI - Debug: Config TrackstickButtons loaded: 1 -> 1
[    6.661227]: VRMI - Info: Recieved GPIO Data
[    6.666973]: VRMI - Debug: rmi_probe_interrupts: Counting IRQs
[    6.687202]: VRMI - Debug: IRQ Count: 7
[    6.687243]: VRMI - Debug: rmi_init_functions: Creating functions
[    6.689117]: VRMI - Info: Initializing F34.
[    6.689125]: VRMI - Debug: Function F34 - IRQs: 1 CMD Base: 0 CTRL Base: 77 DATA Base: 0 QRY Base: 128 VER: 0
[    6.689131]: VRMI - Info: F34 not implemented
[    6.689938]: Got new blocked hash data: 1 hashesSetting 1 new blocked hashesGot new blocked team data: 1 teamsSetting 1 new blocked teamsGot new active rule version: 10415043430518216312VRMI - Info: Initializing F01.
[    6.691007]: VRMI - Debug: Function F1 - IRQs: 1 CMD Base: 126 CTRL Base: 81 DATA Base: 19 QRY Base: 137 VER: 0
[    6.706149]: VRMI - Info: Found RMI4 device, manufacturer: Synaptics, product: TM3157-005, fw id: 2403788
[    6.713551]: VRMI - Info: Initializing F03.
[    6.713561]: VRMI - Debug: Function F3 - IRQs: 1 CMD Base: 0 CTRL Base: 0 DATA Base: 21 QRY Base: 164 VER: 0
[    6.718182]: VRMI - Info: Initializing F11.
[    6.718192]: VRMI - Debug: Function F11 - IRQs: 2 CMD Base: 127 CTRL Base: 87 DATA Base: 37 QRY Base: 166 VER: 1
[    6.744263]: VRMI - Info: Initializing F30.
[    6.744274]: VRMI - Debug: Function F30 - IRQs: 1 CMD Base: 256 CTRL Base: 269 DATA Base: 256 QRY Base: 348 VER: 0
[    6.748159]: VRMI - Debug: F30: Found Button 1
[    6.748168]: VRMI - Debug: F30: Found Trackpoint button 1
[    6.748171]: VRMI - Debug: F30: Found Trackpoint button 2
[    6.748174]: VRMI - Debug: F30: Found Trackpoint button 3
[    6.750158]: VRMI - Info: Initializing F54.
[    6.750167]: VRMI - Debug: Function F54 - IRQs: 1 CMD Base: 347 CTRL Base: 280 DATA Base: 257 QRY Base: 350 VER: 0
[    6.750174]: VRMI - Info: F54 not implemented
[    6.750195]: Sandbox: distnoted(246) deny(1) file-read-metadata /usr/sbin/distnotedSandbox: distnoted(246) deny(1) file-read-metadata /usrWaking up reference: 3Waking up reference: 2Thread waiting on reference 2 woke upevaluation result: 2, allowed, cache,  1638281954Waking up reference: 5VRMI - Debug: F03 - Consumed <ptr>h (14) from PS2 guest
[    6.778319]: Waking up reference: 6Thread waiting on reference 3 woke upevaluation result: 3, allowed, cache,  1638284451Thread waiting on reference 5 woke upevaluation result: 5, allowed, cache,  1638281953Thread waiting on reference 4 woke upevaluation result: 4, allowed, cache,  1638281954Thread waiting on reference 6 woke upevaluation result: 6, allowed, cache,  1638281954ALF, old data swfs_pid_entry <ptr>, updaterules_msg <ptr>, updaterules_state <ptr>VRMI - Debug: F03 - PS2 Command [Send: 0 Receive: 2 cmd: ff]
[    6.875185]: VRMI - Debug: F03 - Recieved data over PS2: fa
[    6.935585]: VRMI - Debug: F03 - Recieved data over PS2: aa
[    6.935594]: VRMI - Debug: F03 - Recieved data over PS2: 0
[    6.935607]: VRMI - Debug: F03 - PS2 Command [Send: 0 Receive: 2 cmd: e1]
[    6.949397]: VRMI - Debug: F03 - Recieved data over PS2: fa
[    6.949408]: VRMI - Debug: F03 - Recieved data over PS2: 1
[    6.949411]: VRMI - Debug: F03 - Recieved data over PS2: e
[    6.949427]: VRMI - Debug: F03 - PS2 Command [Send: 1 Receive: 2 cmd: e2]
[    6.955478]: Sandbox: contextstored(106) deny(1) mach-lookup com.apple.tccd.systemVRMI - Debug: F03 - Recieved data over PS2: fa
[    6.971656]: VRMI - Debug: F03 - Recieved data over PS2: fa
[    7.231144]: VRMI - Debug: F03 - Recieved data over PS2: aa
[    7.231151]: VRMI - Debug: F03 - Recieved data over PS2: 0
[... cut ...] ndbox: dasd(122) allow iokit-get-properties system-typeSandbox: dasd(122) allow file-read-data /private/var/db/DetachedSignaturesVRMI - Debug: F03 - PS2 Command [Send: 1 Receive: 0 cmd: f3]
[    8.271182]: VRMI - Debug: F03 - Recieved data over PS2: fa
[    8.278584]: VRMI - Debug: F03 - Recieved data over PS2: fa
[    8.779628]: VRMI - Debug: F03 - PS2 Command [Send: 0 Receive: 0 cmd: f4]
[    8.787469]: VRMI - Debug: F03 - Recieved data over PS2: fa
[    9.288519]: VRMI - Info: Finish PS2 init

1Revenger1, 2021-12-03:
Log doesn’t look any different. You did it after right? No need to reboot for these logs

VinylNerd, 2021-12-03:
yes i did the log after, is there anything else i can do to try to help? are these logs getting you anywhere?

1Revenger1, 2021-12-03:
The log you gave is helpful because it recognizes the buttons are there. I need a log which shows the track point and buttons being used though to see why it isn’t working. If you could try again, that’d be helpful.

VinylNerd, 2021-12-04:
[log.txt](https://github.com/VoodooSMBus/VoodooRMI/files/7653017/log.txt)
here you go i clicked the buttons on trackpad aswell

1Revenger1, 2021-12-04:
It looks like F30 isn't generating interrupts when clicking with the trackpoint buttons. What you could try is to hold them down and then click the clickpad at the same time to see if it registers (atleast in the logs).
You said they don't work in windows, and it may be some configuration with the trackpad itself that we can't change in VRMI. I'll look around a bit and see though.

VinylNerd, 2021-12-04:
holding down all 3 and clicking on trackpad

[log.txt](https://github.com/VoodooSMBus/VoodooRMI/files/7653086/log.txt)

1Revenger1, 2021-12-04:
hrmmm, they still read as "Up", so something is wrong with them. I'll look further I guess.

VinylNerd, 2021-12-04:
here is some more logs for you, happy to do anything else, i think it might be best to verify i have same hardware ID as someone else to verify my part is genuine, but not sure best way to do that as i feel the kexts may be spoofing the HWID or something like that, however my product feels quality and to be genuine. its my understanding that my trackpad should be supported by the kext yes?  
[log.txt](https://github.com/VoodooSMBus/VoodooRMI/files/7653763/log.txt)

VinylNerd, 2021-12-05:
it seems my trackpad is not genuine as another user tried my EFI with his synaptics and said it works, i have orderd a new one from amazon and will report back if i have luck, however the fact we can see the buttons being pressed is still confusing, only had to open my laptop about 5 times now to try to troubleshoot this lol, thought possibly i damaged the cable at first and when i went back to alps i would also loose the buttons but when i go back to alps it works as expected

amazon delivery comes tomorrow so will report back soon

VinylNerd, 2021-12-05:
[image]

1Revenger1, 2021-12-06:
Not really sure, probably something related to how the trackpad is configured internally or wired up to your device. It sees that the buttons are there, but can not figure out if they’re pressed down. Because it never sees that they are pressed down, it never sends an interrupt/notification to a driver. I’m looking at some things to see if there is a way to configure it though. Probably won’t have anything to test until later this week once finals are done.

VinylNerd, 2021-12-06:
i see, that makes sense. well im replacing it, but i have been fully refunded without return so can keep this to test, my new one arrives today, will report back later today after ive tried it out