VoodooSMBus/VoodooRMI #143: (Rarely) System won't boot with VoodooRMI

sawarek, 2022-01-05:
I'm using the latest versions of VoodooI2C and VoodooRMI (I2C kexts): 2.6.5. and 1.3.4. My touchpad works, but sometimes the system just won't boot and stuck on **busy timeout[0], (60s) IOHIDInterface**, then on apple logo. When it happens I can hear a short sound (maybe HID initialization?)

After hard reset I can boot again and can enter to the system.

sawarek, 2022-01-05:
Checked dmesg:

<pre>VRMI - Info: RMI Bus (RELEASE) Starting up!
VRMI - Info: RMII2C::TPD0 reset completed
VRMI - Error: Unable to reset
VRMI - Error: RMI initial reset failed! Continuing in spite of this
VRMI - Info: Initializing F34.
VRMI - Info: F34 not implemented
VRMI - Info: Initializing F01.
VRMI - Info: Found RMI4 device, manufacturer: Synaptics, product: TM3096-006, fw id: 2430125
VRMI - Info: Initializing F11.
VRMI - Info: Initializing F54.
VRMI - Info: F54 not implemented
VRMI - Info: Initializing F30.</pre>

1Revenger1, 2022-01-05:
Weird, I'm not really sure what would cause this. VoodooRMI doesn't use any of the HID services Apple provides. Can you confirm this only happens when using VoodooRMI? I'm wondering if it's some other device. VoodooInput is the driver more closely using Apple's HID services, and hasn't had any similar issues for SMBus/PS2 trackpads as far as I am aware.

sawarek, 2022-01-05:
Can confirm. I tried many times. This happens only when I’m using RMI (VoodooRMI, RMII2C and VoodooInput). The system won’t boot (from time to time), especially after restart. Sometimes it boots after 1 hard reset, sometimes - after second hard reset.

No problem with booting while using VoodooI2C + VoodoI2CSynaptics + VoodooPS2. 

I tried to remove  USB, Bluetooth kexts.

sawarek, 2022-01-05:
<pre>Users-MacBook-Pro:~ user$ kextstat | grep Voodoo
   40    1 0xffffff7f8357e000 0xb000     0xb000     com.alexandred.VoodooI2CServices (1) DBFA5EBE-2783-30ED-8E39-5829C42D6832 <5 3>
   52    1 0xffffff7f83558000 0x1f000    0x1f000    org.coolstar.VoodooGPIO (1.1) FB3BE4D0-E5F6-30B4-9797-DE6ADDEC2F90 <12 6 5 3>
   62    1 0xffffff7f83589000 0x2f000    0x2f000    com.alexandred.VoodooI2C (2.5.2) 46356335-D26B-38BF-ACE3-43565C70A2C8 <54 52 40 13 12 6 5 3 1>
   93    1 0xffffff7f835c3000 0x22000    0x22000    com.1Revenger1.VoodooRMI (1.3.4) E101F26E-9935-3092-AFF4-7796D1B84130 <6 5 3>
  101    0 0xffffff7f835fb000 0x6000     0x6000     me.kishorprins.VoodooInput (1.1.3) 374A88BF-5D62-3ED2-8E75-6A331AF58A2E <54 6 5 3>
</pre>

singhalrishi27, 2022-02-05:
Same happened with me

ethernetme, 2022-02-25:
Same happens to me, but it just happens with VoodooRMI 1.3.4. I use VoodooRMI 1.3.3, but with this version the touchpad stops working on wake. Please help.

FreeJHack, 2022-03-24:
I had the same boot issue and after some testing I found a solution (for laptop only).
What I discovered is that set 1 sec in the "FirstPollDelay" property of ACPIBatteryManager resolved the issue.
You could try with different values but with higher values may arise an issue with battery recognition during start-up: this depends on how fast your laptop is.
Even the kext loading sequence in OC's kernel configuration is important.
Please, also note that I'm running Mojave with FakeSMC, ACPIBatteryManager and OC 0.7.9, not tested on newer OSXs.
Hope this could be helpful, in any case, it could be a good starting point to evaluate this issue.

elmaleek03, 2022-05-19:
i had the same problem, it sometimes booting like 4-5 mins, and sometimes it boots normally. this issue persist in release v1.3.4. the v1.3.3 doesnt have this issue, but the touchpad doesnt work after wake from sleep.
 
i got my problem fixed by using with using v1.3.3 release and change the RMII2C from the v1.3.4 with the v1.3.3 one. its works flawlessly now.

TPD0, SYNA2b33, F11

ethernetme, 2022-08-13:
> I had the same boot issue and after some testing I found a solution (for laptop only). What I discovered is that set 1 sec in the "FirstPollDelay" property of ACPIBatteryManager resolved the issue. [...]

I am running MacOS Monterey with OC 0.8.3 and VirtualSMC. Did you find any way to resolve the issue with VirtualSMC ?

1Revenger1, 2024-10-14:
Want to see if this still happens with the latest release? May want to grab latest commit as well of both VoodooPS2 and this if not and test that out as well.

ethernetme, 2024-10-14:
> Want to see if this still happens with the latest release? May want to grab latest commit as well of both VoodooPS2 and this if not and test that out as well.

I really appreciate your work on updating the kext, but I'm not using macOS on that computer anymore. I've been planning to install it again though, so I'll let you know when I'm able to test it. Thanks in advance

1Revenger1, 2024-10-14:
No worries. It's been a little while (er uhh a couple years? Oops)

ethernetme, 2024-10-15:
> No worries. It's been a little while (er uhh a couple years? Oops)

Update: Today I was able to test it again. I tried with both the latest commits and the latest releases of the kexts, but unfortunately neither of them worked. The boot process still hangs and it never finishes booting. I can tell you more information about my computer, if that's of any help; my touchpad model is SYNA2B49 (I2C)

1Revenger1, 2024-10-15:
Any log statements that stand out? Does it help if you run it with cpus=1 in the boot-args?
Try setting that boot-arg and sending a picture of where it gets stuck, if you are willing.

PR #193 (2024-10-23): Pass I2C HID input report and timestamp to functions
Closes #189 
Closes #174 
Closes #143
closed 2024-10-23 by 1Revenger1