VoodooSMBus/VoodooRMI #145: Can't work on Samsung Notebook 900X3L

junehwan69, 2022-02-05:
It shows system path of device manager in the windows like HID\SYN2602&COL01\5&4054C9A&0&0000.
I think it can't be support in the current kext.
Please make kext to support this device.

Thanks in advance.

1Revenger1, 2022-02-05:
If the trackpad supports the RMI4 protocol, it should work. Does it work with VoodooI2CHID?

junehwan69, 2022-02-05:
Sadly. No, it can't work too.

Sent by mobile!

1Revenger1, 2022-02-06:
Then interrupts could be set up incorrectly. Can you try forcing polling mode using `-vi2c-force-polling` in your boot args?

junehwan69, 2022-02-06:
OK, I'll try it.

Sent by mobile!

junehwan69, 2022-02-07:
Hi,

I'm solving this problem as VoodooPS2/SSDT-Thinkpad_Trackpad.dsl at master · acidanthera/VoodooPS2 (github.com)<https://github.com/acidanthera/VoodooPS2/blob/master/Docs/ACPI/SSDT-Thinkpad_Trackpad.dsl>

Thanks your helping.
________________________________
보낸 사람: Kim June Hwan ***@***.***>
보낸 날짜: 2022년 2월 6일 일요일 오전 10:50
받는 사람: VoodooSMBus/VoodooRMI ***@***.***>; VoodooSMBus/VoodooRMI ***@***.***>
참조: Author ***@***.***>
제목: Re: [VoodooSMBus/VoodooRMI] Can't work on Samsung Notebook 900X3L (Issue #145)

OK, I'll try it.

Sent by mobile!

closed 2022-02-21 by 1Revenger1