VoodooSMBus/VoodooRMI #146: HP Omen 15 Synaptics touchpad only works properly after sleep

Darkspeed294, 2022-03-07:
I have a HP Omen 15 with a Synaptics trackpad with two physical buttons (SYN327C). I have been trying to figure out how I can enable the trackpad buttons and I have partially achieved this with Voodoops2, VoodooRMI and VoodooSMBUS. Gestures work too, But now I have some weird behaviour going on.

When booting from an off state (G5 or S0) and loading into MacOS, the trackpad has a ~0.5 second delay from the point I rest my finger on the pad. if I drag my finger it would start to drag the cursor around the screen after this delay, but tapping to click and two/three finger taps are just not registered Within MacOS. 

Furthermore, the trackpad buttons are only functional when I rest my finger on the trackpad. as soon as I lift my finger, they stop working again.

When the laptop goes into sleep mode and wakes again, the trackpad with its button starts to work again (although the cursor becomes laggy but who cares), including taps and gestures.

What could be the cause? What is a fix? do I need to enable a _STA method or power states somewhere in my DSDT? 

PS: Rehabmans voodoops2 works but doesn't support gestures, So I am willing to sleep and wake for this method until i find a fix.

[DSDT.aml.zip](https://github.com/VoodooSMBus/VoodooRMI/files/8200799/DSDT.aml.zip)

Darkspeed294, 2022-03-07:
Besides this, I cannot flick the trackpad. every time I lift my finger, the trackpad seems to go into some sort of "off" state. After this I need to keep my finger on the trackpad for it to work again. If you guys really want I can make a video to demonstrate what is going on.

This project already kinda works, so thank you all already for making this all possible. you can't have a Hackintosh without it being hacky at some point you know!

Darkspeed294, 2022-03-08:
Demonstration video:
https://www.youtube.com/watch?v=SV2jXuuJ0ac
Laptop Github:
https://github.com/Darkspeed294/HP-Omen-15-DC-1xxxx-Hackintosh-Opencore

Darkspeed294, 2022-03-08:
https://github.com/VoodooSMBus/VoodooRMI/issues/115

I believe I have the same problem as him

Darkspeed294, 2022-03-08:
Update: after waking from sleep, the trackpad is functional, but after a certain period of time, the laptop starts to hang/stutter and when shutting down causes a stall.

Darkspeed294, 2022-03-09:
I fixed it by using VooDooRMI 1.3

Darkspeed294, 2022-03-10:
Using VooDooRMI 1.3 seems to do the trick and the trackpad with its buttons seem to be functional. However, now I face the issue like before that MacOS starts to hang and stutter after 4-5 minutes of usage. Making this "fix" useless. 
Playing around with the Kexts, it seems that the problem lies within the VooDooSMBUS.Kext
How can I get log from the debug version so I can see what is going on behind the screens?

Darkspeed294, 2022-03-10:
without VooDooSMBUS.Kext the hanging disappears and the trackpad is functional, but the trackpad buttons are non-functional.

all I need is to drag and resize windows

Darkspeed294, 2022-03-10:
I finally got a useful log from terminal this time. has this something to do with gpio pinning or interups 
[image]

Darkspeed294, 2022-03-11:
[image]
voodoormi 1.3 release (debug doesn't seem to work properly)

Darkspeed294, 2022-03-11:
[image]
VooDooRMI release 1.3 Log (debug doesn't seem to work)

tlefko, 2022-03-16:
on my lenovo thinkpad x1 carbon X1C5, i have a similar issue. Gestures and scrolling do not work naturally, and physical buttons have issues as well. i have also explored 1.3, and found similar problems. the behaviour for me is consistent after sleep. in specific, 3/4 finger gestures are essentially not functional unless being mapped to KBM shortcuts — while scrolling with two fingers is unusable laggy, inconsistent. mouse also feels laggy ish. interestingly, GPI0 pinning seems to be initialized okay based on the verbose log.

Darkspeed294, 2022-05-19:
After 4 months now, I still have found no solution to this problem. I have tried anything to make it work normally, but I just can't. from dsdt patches to all kexts that make Synaptics work, nothing suffices.

I got it to work kinda, but every time I boot there is a 30-40% chance it functions properly. The other 60-70% it has a track delay, I can't tap or swipe to switch desktops and the buttons don't work unless I touch the trackpad whilst pressing the button. 

The debug log doesn't tell me shit, even with debug enhancer. Apple console log is also fkn dumb only saying <private>. I don't know what to look at in ioregistery next to differences when it does and doesn't work properly. Very frustrating!

Please revenger, Hear my plea! I will give you anything you ask for.

Darkspeed294, 2022-07-25:
I will reopen another issue with voodoormi 1.3.4 with more information regarding this problem. Closing this issue.

Darkspeed294, 2023-02-05:
great work Revenger!

build 1.4 DEBUG seems to show a 100% proper functionality rate from wake, hot- and cold boot. Well done! You cannot understand how happy you've made me! My warmest congratulations on this achievement!

sincerely, Darkspeed294

Darkspeed294, 2023-02-05:
I will leave this open for a while, since a lot of people have been messaging me personally asking if I found a fix already.

1Revenger1, 2023-02-05:
*visible confusion*

Some stuff related to power management did change, but I didn't expect it to fix this, lol. That's great though.

1Revenger1, 2023-02-06:
GPIO pinning/I2C patches shouldn't be needed with VoodooSMBus/VoodooRMI (unless you are using VoodooRMI with VoodooI2C instead of VoodooSMBus).
Can I get a copy of your IOReg from IORegistryExplorer? Looks like there is something weird going on that was causing notifications to trigger multiple times in VoodooSMBus.

Darkspeed294, 2023-02-12:
I can still drop it if you want, but I couldn't find any differences in the ioregs of before and after. Yes, I am using VoodooRMI and VoodooSMBUS. I too am clueless what fixed it. I tried builds of your code that I modified, but I can't seem to figure out what it was. even after looking into it, I still don't know. 

but hey, I have seen the DSDT of my HP laptop and they suck man ass when it comes to properly making one as they are very convoluted. The ACPI tables are "modular" so to say as I think they made one-for-all to dump on other laptops too, but that brings a lot of bloat too. Fucking cheapskates.

anyways, I am not seeing any problems anymore and I can't find out what caused it. Apple sure loves magic I guess 🤷‍♂️

Great job! keep up the good work!

Darkspeed294, 2023-02-12:
[MacBook Pro.zip](https://github.com/VoodooSMBus/VoodooRMI/files/10717443/MacBook.Pro.zip)

Darkspeed294, 2023-02-12:
I recon it is caused by the bios incorrectly handling the ASPM of the trackpad, because I thought after the wake from sleep, macOS is responsible to initialise these devices again. But no clue how macOS does it.
Even patching DSDT properties like _STA, _DSM, _INI and even pinning (LMAO) did not seem to do anything.

Darkspeed294, 2023-02-12:
[image]

Darkspeed294, 2023-02-12:
don't think this will help, but can't hurt.

cesfhiil, 2023-07-14:
@Darkspeed294 was wondering if you can share the 1.4 debug file as I am having the exact same issues as you (on a ThinkPad X1 Extreme Gen 2) as I do not know how to compile all the new fixes since 1.3.4.

1Revenger1, 2024-10-27:
If you are still around, do you want to give the latest commit a try?

closed 2024-11-03 by 1Revenger1