VoodooSMBus/VoodooRMI #150: Tap to click not working on HP Envy 13 2018

duongnguyen16, 2022-04-26:
**Describe the bug**
I have installed both VoodooRMI and VoodooI2C kexts and followed the instruction.
I can move my cursor easily and use 3,4 finger gestures well.
But I can't tap to click so I must press at left/right trackpad (maybe it's clickpad) to use.
I have togged it at settings, but it does not work.
Any idea?

**To Reproduce**
Steps to reproduce the behavior:
1. Start the laptop
2. Opencore boot successfully
4. Tap to click not work

Darkspeed294, 2022-07-27:
Hey there,

If you put your finger on the trackpad, does it have an input delay?
I have the same problem where the trackpad doesn't want to recognise the initial F12 packages(touchpad finger position packages), making tap to clicks packages unregistered. It does seem to register packages after some input, but tap to click and two finger taps just don't go trough!

If you can, Take the voodoops2 and voodoormi debug kexts and drop a log with you ioregistryexplorer file here. I am already working a long time on this problem and a lot of people seem to have this problem too.

1Revenger1, 2024-10-27:
I believe this should be solved with https://github.com/VoodooSMBus/VoodooRMI/commit/61b34d463d8bce9c4cbffa89273850a5af24babe and other commits.

closed 2024-10-27 by 1Revenger1